iiab/iiab #1196: Calibre-Web http://box/books failing (http://box:8083 sometimes works)

holta, 2018-10-09:
http://box/books used to work, thanks to:

https://github.com/iiab/iiab/blob/master/roles/calibre-web/templates/calibre-web.conf.j2

http://box:8083 works on both Ubuntu 18.04 and Raspbian Lite &mdash; and yet in both cases http://box/books now fails.  FWIW underlying files appearing correct?  On both OS's they appear as follows:
```
root@box:~# ls -l /etc/apache2/*/calibre-web.conf
-rw-r--r-- 1 root www-data 251 Oct  8 20:13 /etc/apache2/sites-available/calibre-web.conf
lrwxrwxrwx 1 root root      35 Oct  8 20:13 /etc/apache2/sites-enabled/calibre-web.conf -> ../sites-available/calibre-web.conf

root@box:~# more /etc/apache2/sites-available/calibre-web.conf
<VirtualHost *:80>
    <Location "/books" >
        RequestHeader set X-SCRIPT-NAME /books
        RequestHeader set X-SCHEME http
        ProxyPass http://localhost:8083/
        ProxyPassReverse http://localhost:8083/
    </Location>
</VirtualHost>
```

holta, 2018-10-09:
@arky & All:

The URL (http://box/books) used to work across different IIAB Linuxes/distros just a few short weeks ago.

It still works on one RPi 3 (not 3 B+) installed in August, even despite that OS being upgraded regularly with "apt update; apt dist-upgrade" and that same IIAB being upgraded regularly with "git pull; ./iiab-install --reinstall"

_Please all who can try to understand why Apache's calibre-web.conf file and symbolic link (for http://box/books) are suddenly now failing across most installations._

holta, 2018-10-09:
I tested 9 IIAB installations, and the only one that one that still works did not have "git pull; ./iiab-install --reinstall" applied in recent days/weeks.

Conclusion: some networking and/or web server change (outside of calibre-web playbook, but within repo iiab/iiab) appears to have accidentally broken http://box/books

arky, 2018-10-09:
@holta  It would be difficult to address this problem without any errors, logs or debug information. 

Try increasing apache2 logging level to find any problems with mod_rewrite's. 

`LogLevel  debug rewrite:trace8 proxy:trace8`

holta, 2018-10-11:
> Try increasing apache2 logging level to find any problems with mod_rewrite's.
>
> `LogLevel debug rewrite:trace8 proxy:trace8`

I tried that without luck, so the mystery remains :/

In any case the problem appears fixed by PR #1213.

(If however however this deserves more investigation, please re-open this tkt as nec!)

closed 2018-10-11 by holta