iiab/iiab #1198: Using plaintext passwords in shell scripts is not recommend

arky, 2018-10-09:
A security issue has been introduced in transmission playbook. It is recommended that plaintext passwords should not be used in shell scripts. 

https://github.com/iiab/iiab/blob/72ed1d6887245b8361d1b17f72e451d14ebf9d63/roles/transmission/tasks/main.yml#L43

Such authentication is not required if the "rpc-authentication-required" is disabled in settings.json 

https://github.com/iiab/iiab/blob/72ed1d6887245b8361d1b17f72e451d14ebf9d63/roles/transmission/templates/settings.json.j2#L46

holta, 2018-10-09:
IIAB implementers access to http://box:9091 (and most all IIAB services) protected using http username/password Admin/changeme by default, so these services can be accessed from different devices, almost essentially anywhere (VPN, etc).

I don't see any practical alternatives if Transmission doesn't offer other ways to set this http username/password ?

arky, 2018-10-09:
The simplest solution is to disable rpc authentication from localhost in setting.json. That way the following playbook command will be executed without errors.

holta, 2018-10-09:
Most implementers do not want http://box:9091 modifiable (or even visible) to all their students/users/etc.

(The password serves a critical function here... perhaps you want to implement an installation-time flag for those individual implementers who do want http://box:9091 to be password-protected?)