iiab/iiab #1200: Undocumented roles

aidan-fitz, 2018-10-09:
The following roles do not have `README.*` files:
```
sshd
authserver
debian_schooltool
postgresql
ejabberd_xs
idmgr
calibre
network
ajenti
moodle-1.9
mysql
awstats
homepage
httpd
mongodb
ejabberd
monit
moodle
wordpress
pathagar
schooltool
nodogsplash
sugarizer
munin
0-init
cups
osm
nextcloud
docker
phpmyadmin
sugar-stats
teamviewer
openvpn
mediawiki
vnstat
```

holta, 2018-10-09:
Thanks much @aidan-fitz for generating the above list of 35 IIAB roles, needing more of an explanation!

FYI the 12 roles below should probably be removed from your list, as they're rarely if ever used: (as of Oct 2018 anyway!)

1. sshd
2. authserver
3. debian_schooltool
4. ejabberd_xs
5. idmgr
6. ajenti
7. moodle-1.9
8. pathagar
9. schooltool
10. nodogsplash
11. docker
12. sugar-stats

holta, 2018-10-09:
My mistake...now corrected above: **sshd** is in fact an active role :)

holta, 2018-11-04:
> Those in favor should mention how many of the existing readmes they have read.

Some of these README's are absolutely top-notch, but they have been overlooked as they were too hard to find.

So I'm changing that now, with direct links to the best README's from http://FAQ.IIAB.IO which has a very high Google page ranking.

This is a big job, and I'm not done yet (to make our best documentation accessibl!)  But please for now see ["What can I do with E-books and Internet-in-a-Box?"](http://wiki.laptop.org/go/IIAB/FAQ#What_can_I_do_with_E-books_and_Internet-in-a-Box.3F) which is helping to make all IIAB roles (playbooks) much more understandable to new implementers and go-getter teachers/students around the world.  Folks who truly want to understand (and where possible help us all make more/better digital libraries in coming years!)