iiab/iiab #1205: Signal

aidan-fitz, 2018-10-09:
IIAB instances could run a local-only Signal server to enable end-to-end secure communication.

- [Signal Server](https://github.com/signalapp/Signal-Server)
- [Contact Discovery Service](https://github.com/signalapp/ContactDiscoveryService) (beta)
- [Android app](https://github.com/signalapp/Signal-Android)
- [Desktop (Electron) app](https://github.com/signalapp/Signal-Desktop)

holta, 2018-10-09:
@floydianslips this is somewhat related to your use of Nextcloud for secure communications?

floydianslips, 2018-10-10:
I don't think Nextcloud uses the signal protocol, however it is end to end encryption. It can be used secure communication for sure.

samuk, 2018-10-11:
Briar might be a good choice for direct messaging (Signal but designed for offgrid) https://play.google.com/store/apps/details?id=org.briarproject.briar.android

Also Manyverse would be great for more social public chats: https://play.google.com/store/apps/details?id=se.manyver

samuk, 2018-10-11:
Also while I think of it https://play.google.com/store/apps/details?id=org.servalproject Is good for phone to phone filesharing. Which means it could be used to spread itself. It could also be used to share the Briar/ Manyverse apk's or indeed any other content from the LibraryBox. 

Sharing all three of these should come in at <100mb I think and would be a good way to spend a bit of space IMHO.

aidan-fitz, 2018-10-18:
Yes! That's actually a great idea! Is there currently an APK server on IIAB?

holta, 2018-10-18:
> Is there currently an APK server on IIAB?

@m-anish considered including F-Droid (community-maintained repo of Android software / APKs) as an IIAB option, but as far as I know that never actually happened (is that what you mean?)

>@holta, what mechanism do we currently use to distribute APKs to clients?

@tim-moody (and Kiwix, and others) helped generate several APKs that were then distributed from individual IIAB's at http://box/local_content typically.

In those cases, the APK files were just copied to /library/www/html/local_content (e.g. for medical students to download the most needed medical-content-APK's from IIAB to their own phones).

aidan-fitz, 2018-10-18:
Yes, that's what I mean. I remember reading that F-Droid has rather hefty system requirements; is that true?

tim-moody, 2018-10-18:
@aidan-fitz I used F-droid on my S6 and didn't notice anything out of the ordinary. I'm also not sure it is necessary to side-load an apk. The user has to be willing to go through the manual steps in spite of scary messages. But we have side-loaded the medical apk a number of times.

look at the top link at http://medbox.iiab.me/s-asia/ (which on the public server points to archive.org)

Also the medical version has apks from HealthPhone
http://medbox.iiab.me/healthphone/en/ at the bottom

aidan-fitz, 2018-10-18:
Ah, I meant the server. We should have a Google Play–like interface for installing and managing offline APKs so they don't have to be sideloaded. Is F-Droid like that?

tim-moody, 2018-10-18:
I'm not aware of anything. By side-load I mean from the server.

tim-moody, 2018-10-18:
F-Droid is client side

samuk, 2018-10-18:
@tim-moody 

I think you can run a server: https://f-droid.org/en/docs/Installing_the_Server_and_Repo_Tools/

But there is also an Android client: https://f-droid.org/en/packages/org.fdroid.fdroid/

It seems like it might be overkill to me, maybe start with a simple list of apk's and see how many there are that would be useful in an offline context?

tim-moody, 2018-10-18:
@samuk we already have such a list of apks on the server.

The user of an android phone clicks on one of the links on the home page and the apk downloads from the server and installs on the phone (if given permission by the user).

ericnitschke, 2018-11-22:
> The user of an android phone clicks on one of the links on the home page and the apk downloads from the server and installs on the phone (if given permission by the user).

The F-droid mobile app updates the associated apps, and itself.  One can pull the F-droid app update (and app updates) from one mobile phone to another using a wifi hotspot/bluetooth. Might be nice to have this feature within IIAB, similar to the way the the Kiwix catalog can be refreshed/updated.  The 'nearby' function (https://github.com/iiab/iiab/issues/1243#issuecomment-441017825) could also be useful.

ericnitschke, 2018-11-22:
> Briar might be a good choice for direct messaging (Signal but designed for offgrid) https://play.google.com/store/apps/details?id=org.briarproject.briar.android
> 
> Also Manyverse would be great for more social public chats: https://play.google.com/store/apps/details?id=se.manyver

We're using [RocketChat](https://f-droid.org/en/packages/chat.rocket.android/) for offline messaging in the [iNethi](https://www.inethi.org.za/about/) community network. Looking forward to testing others.

m-anish, 2019-01-20:
Love this conversation! I would love to see such apps on IIAB+Mesh networks!

m-anish, 2019-01-20:
Will add this to the mix

https://github.com/CableIM

Seems to be a signal fork dedicated to something like what we're doing :)

m-anish, 2019-01-21:
Also this.. 
https://riot.im

This seems to be the equivalent of slack, gitter etc. 

I had checked riot.im a while ago and it.. sucked, but it seems to have made lots of progress.

holta, 2019-03-03:
Tangentially related:
#1540 DESIGN, PROPOSAL | Rethink network, avahi, zeroconf, link-local for decentralized and multiple IIAB's on a network

holta, 2019-03-03:
Internet-in-a-Box should also consider endorsing "the best" Android filesharing apps among things like:

SHAREit, Xender, Zapya, AirDroid, Superbeam, Portal/PushBullet, Send Anywhere (etc!)

Sneakernet-of-Alexandria refs:
#828 #857 #1002 #1530 #1538

ericnitschke, 2019-03-04:
> Also Manyverse would be great for more social public chats: https://play.google.com/store/apps/details?id=se.manyver

Now has bluetooth sync: https://www.manyver.se/blog/bluetooth-sync

holta, 2019-03-04:
> > Also Manyverse would be great for more social public chats: https://play.google.com/store/apps/details?id=se.manyver
> 
> Now has bluetooth sync: https://www.manyver.se/blog/bluetooth-sync

@ericnitschke do you use it?  Can you say more if this makes offline/grassroots learning exchanges substantially more practical?

holta, 2019-03-04:
See Also:

#1542 What do you think about Mastodon? ["self-hosted, globally interconnected microblogging community"]

floydianslips, 2019-03-05:
I apologise that I don't have time to follow this thread very closely, but
I wanted to find out if anyone has tried out the chat function on
Nextcloud. I have had pretty good luck with it.

Josh