iiab/iiab #1206: Consolidate dependencies

aidan-fitz, 2018-10-09:
The person creating the IIAB instance should be able to configure all apps to use the same dependencies (such as databases) so that fewer dependencies are installed overall. Installing fewer dependencies will not only free up disk space for more apps but also reduce the average memory footprint of installed apps.

For example, if all of the apps that the user wants to install support MySQL and 2 of them also support Postgres, then only MySQL will be installed. Moreover, all of the apps will connect to the same MySQL daemon.

This is a [set cover problem](https://en.wikipedia.org/wiki/Set_cover_problem): we want to find the smallest number of dependencies that can simultaneously satisfy all of the apps' requirements. There is a polynomial-time algorithm for finding a set of dependencies that is at most _H(n)_ = 1 + 1/2 + ... + 1/_n_ times as large as the minimum needed to support _n_ apps.

Additionally, the user could declare that they prefer specific dependencies by listing them in priority order in `local_vars.yml`. For example:
```
preferred_dependencies:
- sqlite
- mysql
```

holta, 2018-10-09:
@aidan-fitz there's a longstanding debate on exactly this topic during our weekly community/team calls (http://minutes.iiab.io) as briefly summarized here: https://github.com/iiab/iiab/issues/1191

As a practical matter some core IIAB services are indeed essential-for-essentially-everybody, and pragmatically, for installation sanity.  Many believe SQLite and MySQL both belong in that category (anyway that's the consensus I'm hearing so far!)

However the road forward might await further usability "fieldback" from other school/medical systems around the world, before any "final" decision prevails &mdash; just in case "ultralight" IIAB's (handheld / knowledge hotspots of various kinds, that folks carry around in their purse etc) keep growing in popularity...

tim-moody, 2018-10-09:
the problem is that this means automating configurations for multiple databases, whereas we have generally only chosen one, even when the service supports multiple (usually the database).