iiab/iiab #1210: Mental health software

aidan-fitz, 2018-10-10:
We should add some mental health–related software to the IIAB project. I know this would be beneficial for schools because students experience a lot of stress and many students have mental illnesses. We should also look into other segments of IIAB users to see what would be most beneficial.

Some mental health apps we could use are:

- [if me](https://github.com/ifmeorg/ifme), anonymous forum software for sharing mental health experiences and giving P2P support
- A directory of local mental health resources (could use an existing app like MediaWiki or WordPress)
- A screening app (see [Mental Health America](https://screening.mentalhealthamerica.net/screening-tools) for examples)

arky, 2018-10-10:
@aidan-fitz In the country where I live, there isn't even a single mental health center or qualified professional to reach out to. I have been kicking few ideas around but nothing stuck. I will check out if me. 

Meanwhile @kananigit Can you please look at these suggestions and give me your feedback.

aidan-fitz, 2018-10-10:
I'm going to [Big Red Hacks](https://bigredhacks.com/) in less than two weeks; if there's an application in this space that you want to have for the project but doesn't exist yet, I can make it at the hackathon.