iiab/iiab #1212: [Travis CI] Build failing: Permission denied when running Ansible

aidan-fitz, 2018-10-11:
### Expected behavior and actual behavior.

The Travis CI pipeline for our branch #894 failed when trying to launch Ansible. The console output is reproduced below:

```
$ ansible --version
Unhandled error:
...
AnsibleError: Unable to create local directories(/root/.ansible/tmp): [Errno 13] Permission denied: '/root/.ansible'

Traceback (most recent call last):
...
ansible.errors.AnsibleError: Invalid settings supplied for DEFAULT_LOCAL_TMP: Unable to create local directories(/root/.ansible/tmp): [Errno 13] Permission denied: '/root/.ansible'
...
AnsibleError: Unable to create local directories(/root/.ansible/tmp): [Errno 13] Permission denied: '/root/.ansible'

The command "ansible --version" failed and exited with 1 during .

Your build has been stopped.
```

View the [full pipeline results](https://travis-ci.org/rgangela99/iiab/builds/439425832).

### Steps to reproduce the problem.

N/A

### Specifications like the version of the IIAB, operating system version, or hardware details.

We're getting this error on our branch. Could someone please confirm

holta, 2018-10-11:
Did you install Ansible using [/opt/iiab/iiab/scripts/ansible](https://github.com/iiab/iiab/blob/master/scripts/ansible) as recommended by our [traditional instructions](https://github.com/iiab/iiab/wiki/IIAB-Installation#do-everything-from-scratch) and/or our [1-line installer](http://download.iiab.io/6.7) ?

Aside/FWIW: most all IIAB volunteers do not use Travis-CI as of 2018, even while @arky put in a very brave effort trying to get this working in 2017 &mdash; there remain many challenges to making such a pipeline broadly effective that we have not solved at this time.

holta, 2018-10-11:
If you haven't seen it, see also http://FAQ.IIAB.IO which discusses:

_"What is Ansible and what version should I use?"_

aidan-fitz, 2018-10-11:
It's working on our staging box; it's only failing on Travis.

arky, 2018-10-11:
Since last few releases of Ansible, Ansible doesn't work out-of-box on Travis CI default container based environment (See https://github.com/iiab/iiab/issues/1066) . There are two action points here: file a bug with upstream ansible bug tracker and also to write a new travis file running ansible inside a docker or some other workaround.  

@aidan-fitz  Thanks for filing this bug. See if you could take a shot at fixing travis and send a PR. 

@holta  Please kindly make this bug as duplicate of https://github.com/iiab/iiab/issues/1066

tim-moody, 2018-10-11:
There are a number of things that we configure, most notably networking, that are not compatible with a container, though there are with vm. So travis is unlikely to work with our main playbook. If we were to split our roles between base server and other services, the latter would probably work at least once ansible works.

aidan-fitz, 2018-10-12:
@holta this is actually not a duplicate of #1066; the bugs are different.

holta, 2018-10-13:
> @holta this is actually not a duplicate of #1066; the bugs are different.

Thanks for clarifying!  Can you confirm this one (#1212) be closed now that PR #1217 is merged?

aidan-fitz, 2018-10-14:
Yes - though the build is still failing, this specific fault has been fixed. [log on Travis CI](https://travis-ci.org/rgangela99/iiab/builds/441155031)

arky, 2018-10-14:
@aidan-fitz @holta After some more digging, I found the real source of this problem was hardcoded paths in ansible.cfg 

https://github.com/iiab/iiab/blob/9269a15e84c495c95ea05ce6633c5cc34ec9f35c/ansible.cfg#L6-L7

Recommend re-opening this bug.

holta, 2018-10-17:
@aidan-fitz & @arky can you recheck / close this now?

arky, 2018-10-17:
@holta Looks good. Sending a PR with cleaned up and refactored travisCI file.

closed 2018-10-17 by holta