iiab/iiab #1214: dnsmasq service fails to start more than 50% of the time on Ubuntu 16.04

holta, 2018-10-11:
Arising from #1184: `./iiab-network` continues to fail more than 50% of the time on Ubuntu 16.04, with error messages:

> TASK [network : dhcp_server may be affected - starting - user choice] **********
fatal: [127.0.0.1]: FAILED! => {"changed": false, "msg": "Unable to start service dnsmasq: Job for dnsmasq.service failed because a timeout was exceeded. See \"systemctl status dnsmasq.service\" and \"journalctl -xe\" for details.\n"}
        to retry, use: --limit @/opt/iiab/iiab/iiab-stages.retry

jvonau, 2018-10-11:
I see this as 16.04 is not a primary target I would suggest not using dnsmasq with that distro until this could be further debugged. I'd be inclined to tag this as won't fix with a know bug warning to edit local_vars when using this distro.

jvonau, 2018-10-11:
https://bugs.launchpad.net/ubuntu/+source/dnsmasq/+bug/1761096

tim-moody, 2018-10-11:
agree with Jerry. Just use bind/dhcpd

holta, 2018-11-06:
@m_anish please see above, and help us migrate Node-RED and Mosquitto/MQTT (#854) to Ubuntu 18.04 if possible prior to IIAB 6.7's release in December!

closed 2018-11-06 by holta