iiab/iiab #1229: Incomplete LICENSE file

arky, 2018-10-15:
It is good practice to include a complete LICENSE file with a open source software github project. Please kindly complete the the Internet in a Box license file at your earliest convenience. 

https://github.com/iiab/iiab/blob/master/LICENSE.md

REFER:

https://help.github.com/articles/adding-a-license-to-a-repository/

https://github.com/iiab/iiab/blob/master/COPYING

tim-moody, 2018-10-15:
@arky The file you're looking for is https://github.com/iiab/iiab/blob/master/COPYING which is a complete license. I would be happy to see it renamed FULL-LICENSE or some such, but the text in the current LICENSE file needs to remain.

As an aggregator we are not responsible for the licenses of included software, which reside with their authors.

tim-moody, 2018-10-15:
how about if we rename LICENSE to LICENSING and COPYING to LICENSE (updating if you prefer)

aidan-fitz, 2018-10-16:
GitHub's license sniffer recognizes `COPYING` - see [here](https://github.com/TTimo/doom3.gpl) for an example.

arky, 2018-10-16:
Thanks for the links @aidan-fitz  I would love have something like this show up on IIAB Github (Screenshot).  

[image]

I generally prefer 'LICENSE', 'LICENSING' files over 'COPYING'.

tim-moody, 2018-10-16:
@arky I agree on name change. COPYING was legacy.

aidan-fitz, 2018-10-16:
Since IIAB is heavily based on Ansible, aren't we required to license it under GPLv3? This might only apply to software components, though. On the other hand, Google considers the Linux kernel license (GPLv2) not to apply to the Android project as a whole even though it is an essential component of Android, so maybe the GPLv3 share-alike condition doesn't apply to IIAB.

Personally, I'm a fan of the [ISC License](https://choosealicense.com/licenses/isc/), which is a slimmed-down MIT license or BSD 2-clause license, but I don't recommend it for this project because several countries, including Cambodia, have not adopted the Berne Convention.

tim-moody, 2018-10-16:
I think we need to be clear on what we license and what we don't. We don't offer an ansible license. We offer a license on our playbook, which does not incorporate any ansible code. Same for apache, mysql, and all the rest of the third party services. Except if we distribute 3rd party software we need to license our derivative code in a manner consistent with the 3rd party license.

We only license things for which we hold the copyright and this is even more complicated in that we are not a company but a collective and each hold copyright to the code we write individually to an extent that depends on the jurisdiction in which we develop. For example, in Canada everything I write is automatically copyrighted by me until I give the copyright away.

So my view is that we need to make clear that the act of putting code in this repository implies licensing under whatever license we agree upon.

PR #1234 (2018-10-17): Rename license files
Fixes #1229 per @arky's request.

License.md paginated to within 80 characters, as it was renamed to Licensing.md, along with basic corrections.
closed 2018-10-17 by holta