iiab/iiab #1232: Housekeeping: Remove deprecated files and unused scripts

arky, 2018-10-16:
While hunting for an elusive bug, I noticed how much cruft has accumulated in the top level IIAB directory. It's seems a little house keeping is in order. 

Perhaps we could remove .deprecated files and old unused scripts. Removing these files doesn't hurt since there is git history for future reference. 

```
ansible.cfg.deprecated  
iiab.yml.deprecated  
README.md.deprecated  
runansible.deprecated  
runtags.deprecated

```

holta, 2018-10-16:
1. Below are 10 other deprecated files (further down within /opt/iiab/iiab).  They don't do much harm...but might also be deleted in the coming year:
```
root@box:/opt/iiab/iiab# find */ -name '*deprecated' | sort

roles/network/tasks/create_ifcfg.yml.deprecated
roles/openvpn/templates/iiab-vpn.conf.in.deprecated
roles/openvpn/templates/iiab-vpn.j2.deprecated
roles/openvpn/templates/party-line.conf.j2.deprecated
roles/openvpn/templates/start.j2.deprecated
roles/openvpn/templates/up_wan.deprecated
roles/openvpn/templates/xsce-vpn.conf.in.deprecated
scripts/ansible-2.2.0-deprecated
scripts/ansible-2.4.x-deprecated
scripts/ansible-2.5.x-deprecated
```
2. Truly obsolete roles/playbooks should eventually be deleted too, as these confuse new implementers far more than empowering anybody :/

   So I've started relegating such to the bottom of [default_vars.yml](https://github.com/iiab/iiab/blob/master/vars/default_vars.yml) and [local_vars.yml](http://wiki.laptop.org/go/IIAB/local_vars.yml) &mdash; but eventually carrying this weight around should end.  Perhaps starting with the least used on this list:

   https://github.com/iiab/iiab/blob/master/unmaintained-roles.txt

arky, 2018-10-18:
Can someone audit all the playbooks, explain their purpose in a readme and remove if they are unused. Thanks! 

```
iiab-from-console.yml
iiab-network.yml
iiab-stages.yml
iiab-support.yml
run-one-role.yml

```

holta, 2018-10-18:
All 5 are currently used FYI:
- iiab-from-console.yml is used by http://box/admin -> Configure -> Install Configured Options
- iiab-network.yml is used by ./iiab-network - _runs the network role but no others_
- iiab-stages.yml is used by ./iiab-install - _similar to Install Configured Options above_
- iiab-support.yml is used by ./install-support - _installs & enables OpenVPN_
- run-one-role.yml is used by ./runrole - _to install/run a single role_

closed 2018-12-10 by holta