iiab/iiab #1241: Use Travis CI to check all PRs

aidan-fitz, 2018-10-18:
Travis CI is now working with flying colors! See my latest [build pipeline](https://travis-ci.org/rgangela99/iiab/builds/442962157) that confirms this.

We should now set up Travis to check every incoming PR and make sure it passes all checks before it can be merged.

holta, 2018-10-19:
Is a Travis CI quickie guide possible in [FAQ.IIAB.IO](http://wiki.laptop.org/go/IIAB/FAQ) and/or [IIAB Contributors Guide](https://github.com/iiab/iiab/wiki) within a couple sentences, so all IIAB developers can use this this?

(Without making excuses that configuring it is just too annoying ;)

arky, 2018-10-19:
@holta  Not sure about including it to the FAQ, TravisCI is primarily a developer tool. 

The contributors guide already has a section about it. You might have to remove the references to  ansible-lint and ansible-review tool which I remove to due to current state of the code. 

https://github.com/iiab/iiab/wiki/IIAB-Contributors-Guide#testing-your-code-with-travis-ci

holta, 2018-10-19:
> The contributors guide already has a section about it. You might have to remove the references to ansible-lint and ansible-review tool which I remove to due to current state of the code.
>
> https://github.com/iiab/iiab/wiki/IIAB-Contributors-Guide#testing-your-code-with-travis-ci

@aidan-fitz can you look this over, to make sure theabove is crystal clear and hassle-free to new/overseas developers?  (Who may not speak English as their native lingo!)

Feel free to edit the above directly if you can improve it!

tim-moody, 2018-10-19:
as @arky says, don't see any lint or review, just the syntax check.

aidan-fitz, 2018-10-20:
@holta The Travis stuff is written in the present tense, as if we're currently using Travis on the main repo. Is this a big deal? Are we re-enabling it?

holta, 2018-10-20:
> Are we re-enabling it?

Let me know what exact steps I should follow if the is wise.

aidan-fitz, 2018-10-20:
travis-ci.com now supports the new GitHub Checks API - here is a [recent blog post](https://blog.travis-ci.com/2018-05-07-announcing-support-for-github-checks-api-on-travis-ci-com) on how to enable it for your project. Here is a [blog post](https://blog.travis-ci.com/2018-05-02-open-source-projects-on-travis-ci-com-with-github-apps) on how to set up travis-ci.com for an open source project.

holta, 2018-10-22:
> travis-ci.com now supports the new GitHub Checks API - here is a [recent blog post](https://blog.travis-ci.com/2018-05-07-announcing-support-for-github-checks-api-on-travis-ci-com) on how to enable it for our project. Here is a [blog post](https://blog.travis-ci.com/2018-05-02-open-source-projects-on-travis-ci-com-with-github-apps) on how to set up travis-ci.com for an open source project.

Thanks: setting this up last year didn't quite work out, despite several tries, but hopefully this year it's more straightforward.  (Worst case if I still can't get this working by end of week, I might need to call you...)

holta, 2018-10-25:
> travis-ci.com now supports the new GitHub Checks API - here is a recent blog post on how to enable it for our project. Here is a blog post on how to set up travis-ci.com for an open source project.

Apologies I'm quite ignorant of Travis.

> @aidan-fitz & @arky let me know what exact steps I should follow if this is wise.

Can you provide precise steps/suggestions or perhaps better call?  As I'd rather not push buttons at random in places like the following...

OLD:
https://travis-ci.org/iiab/
https://travis-ci.org/iiab/iiab
https://travis-ci.org/organizations/iiab/repositories
https://travis-ci.org/holta/iiab/settings

PARTIALLY MIGRATED TO:
https://travis-ci.com/iiab/ "No active repositories could be found"
https://travis-ci.com/iiab/iiab "We couldn't find the repository iiab/iiab"
https://travis-ci.com/organizations/iiab/repositories "Activate" ? ("Internet-in-a-Box @iiab")
https://travis-ci.com/holta/iiab/settings (has very diff setting from the above https://travis-ci.org/holta/iiab/settings)
...
https://travis-ci.com/account/repositories

holta, 2018-10-25:
Related/Aside: external PR's (in my own GitHub acnt) like https://github.com/holta/iiab/pull/174 are apparently tested by Travis, showing...
```
Travis CI - Pull Request In progress — Build Started
```
...followed by:
```
Travis CI - Pull Request Successful in 1m — Build Passed
```
...with summary report at:

https://github.com/holta/iiab/pull/174/checks?check_run_id=26458832

aidan-fitz, 2018-10-26:
Unfortunately, I'm new to Travis too, so I'm not sure how helpful I can be.

What I know: For personal (non-org) projects, Travis will automatically build a PR if it is enabled for your account and the PR branch contains a `.travis.yml` file. Since I have migrated my account from [travis-ci.org](https://travis-ci.org) to [travis-ci.com](https://travis-ci.com), all of my builds since the migration appear on travis-ci.com whereas all of my previous builds are still on travis-ci.org.

I just tried authorizing Travis on an organization I control and creating a dummy pull request at socialhacks/socialhacks.github.io#2 (see the pull request for steps). So far I haven't gotten Travis to recognize the PR at all.

holta, 2018-10-27:
Thanks much Aidan.

FYI https://travis-ci.com/organizations/iiab/repositories has activated/integrated 7 of our 8 repos, but unfortunately not our main one (https://github.com/iiab/iiab) as explained by Travis here:

> The following repositories cannot be migrated to travis-ci.com at this time because they are currently active on our legacy platform travis-ci.org. This feature will be available shortly. Please read our docs on open source migration to learn more.
> https://docs.travis-ci.com/user/open-source-on-travis-ci-com/#existing-open-source-repositories-on-travis-ciorg

Sadly https://travis-ci.org/organizations/iiab and https://travis-ci.org/organizations/iiab/repositories both show "There was an error, please try again" so I cannot even wipe prior settings within travis-ci.org

So I've emailed support@travis-ci.com hoping for a bit more clarity, but as a non-paying organization I don't know if/when they will respond...keep in touch as we all try to sort this out together :>

aidan-fitz, 2018-10-27:
Alright, let's just wait then.

aidan-fitz, 2018-10-27:
> Repositories may also be migrated without their build history or build settings (including environment variables) immediately.

This is also an option, in case we don't care about scrapping that.

holta, 2018-10-27:
> > Repositories may also be migrated without their build history or build settings (including environment variables) immediately.
> 
> This is also an option, in case we don't care about scrapping that.

I wish this were an option!  Can't find such an option so far...in just case, I asked about that in my email to Travis...

holta, 2018-10-29:
Update: Travis responded to my email saying they have migrated our repository to travis-ci.com

Unfortunately it still doesn't work (e.g. https://travis-ci.com/organizations/iiab/repositories fails to load, with an error) so I've written them back hoping they can assist.

holta, 2018-10-29:
Better News: https://travis-ci.com/iiab/iiab/settings is now live at Travis-CI.com after another email with the company in Berlin.

PRs #1265 & #1267 appear to have been checked (for syntax issues) automatically here:
https://travis-ci.com/iiab/iiab/pull_requests

For example PR #1267 appears here:
https://travis-ci.com/iiab/iiab/builds/89478373

Anything more we should do before closing this ticket?

holta, 2018-11-02:
Travis-CI.com integration (e.g. https://travis-ci.com/iiab/iiab) appears reliable after half a week.

closed 2018-11-02 by holta