iiab/iiab #1243: Provide Android Media Distribution End on box.lan/apps

arky, 2018-10-18:
Provide 'box.lan/apps' self hosted app store for side loading Android app. This is a popular request in many countries like India and Cambodia. 

At the moment Torsten Grote of Guardian project is working on lightweight repomaker project https://gitlab.com/fdroid/repomaker

He has written a good account on the background of the project in 'Build your own app store' 
https://guardianproject.info/2017/02/22/build-your-own-app-store-android-media-distribution-for-everyone/

_Originally posted by @aidan-fitz in https://github.com/iiab/iiab/issues/1205#issuecomment-430883492_
_Originally posted by @samuk in https://github.com/iiab/iiab/issues/1205#issuecomment-431157444_

holta, 2018-11-06:
Tangentially related: #1069 (Add IIAB to Ansible Galaxy)

ericnitschke, 2018-11-22:
I was testing F-Droid's 'nearby' feature yesterday, transferring apps between mobile devices.

I wonder if it would it make sense/be possible to enable a "Find people near me" feature in IIAB?  Users running F-Droid on their mobile could see the IIAB F-Droid app store and download/install apps.

tim-moody, 2018-11-22:
Currently the medical apks are downloadable from iiab via a link on the home page. Does this constitute and 'app store' or is there more to it?

jvonau, 2022-04-05:
stale?

holta, 2022-04-05:
> stale?

Yes.  But it's not a bad long-term goal for the broader offline learning / library science community to consider?

samuk, 2022-04-05:
Here's some apps that are potentially useful offline: https://rosettaphone.org/apps/

holta, 2022-04-06:
> Here's some apps that are potentially useful offline: https://rosettaphone.org/apps/

Also potentially highlighting the best from...?

- https://f-droid.org
- https://digitallibrary.io