iiab/iiab #1257: Cannot reliably ssh or route to RPi's wlan0 192.168.0.* after WiFi-installing IIAB

holta, 2018-10-28:
Confoundingly, "ssh pi@10.8.0.x" and routing to RPi's OpenVPN address (10.8.0.*) always works reliably, even despite the fact it's routing over the very same WiFi connection that does not work when used directly ("ssh pi@192.168.0.x" often fails, at the very same time).

FYI "permanently" disabling promiscuous mode the RPi's wlan0 interface and rebooting does not help.

The symptom persists: specifically every 30min (or every hour or so) some of the other machines on this 192,168.0.* network cannot route to the RPi 3 B+ in question, showing the following error message:
```
root@box:/opt/iiab/iiab# ssh pi@192.168.0.102
ssh: connect to host 192.168.0.102 port 22: No route to host
```
Context:
- MIN-sized http://download.iiab.io/6.7 (master branch of IIAB) was installed onto Raspbian Lite 2018-10-09 on RPi 3 B+.
- `iiab-hotspot-on` has not yet been run, and the RPi's WiFi interface continues to work reliably as a WiFi client (for outgoing ssh connections anyway!) as seen by the existence of "wpa_supplicant" in the output of `systemctl status dhcpcd`

holta, 2018-12-10:
Tangentially related: #926

jvonau, 2020-04-29:
stale ticket flaky firmware

closed 2020-04-29 by holta