iiab/iiab #1287: Wordpress transfer/migration between boxes (with pages, posts, images, plugins, users etc.)

ericnitschke, 2018-11-07:
### Expected behavior and actual behavior.
### Steps to reproduce the problem.
### Specifications like the version of the IIAB, operating system version, or hardware details.

After testing the default WP export/import function and numerous plugins, we found one that actually works.

Specs:
RPi 3B+
IIAB 6.7 from today
Raspbian from Oct.

**Steps**

1. On Box 1 install "All-in-One WP Migration" plugin: https://wordpress.org/plugins/all-in-one-wp-migration/
2. Run it and export file to downloads.
3. Copy file from downloads to flash drive.
4. On Box 2 install WP.  Site name, and username/password can be anything.
5. Install "All-in-One WP Migration" on Box 2.
6. On Box 2, change both upload max and post max to 256MB in php.ini (or to whatever your file size is).  The plugin max is ~500MB.
7. Upload File from flash drive through the plugin.
8.  Log into WP with credentials from Box 1.
9. Profit

I can fill in more details when I replicate this tomorrow.

It would be great to increase upload and post max file size limits to 500MB in local_vars.yml,  (apache_high_php_limits: True) or give a space to modify the specific number according to peoples' requirements.  Otherwise users will have to modify php.ini (/etc/php/7.x/apache2/php.ini) to their needs for each box.

holta, 2018-11-07:
Ref #1147

holta, 2018-11-08:
@kanangit can you confirm the above WordPress "duplicate my entire site" scheme is a good one, so we can broadly recommend it in http://wiki.laptop.org/go/IIAB/FAQ#WordPress_.26_Moodle_Administration:_What_tips_.26_tricks_exist.3F if so?

holta, 2018-11-10:
> @kananigit can you confirm the above WordPress "duplicate my entire site" scheme is a good one, so we can broadly recommend it in http://wiki.laptop.org/go/IIAB/FAQ#WordPress_.26_Moodle_Administration:_What_tips_.26_tricks_exist.3F if so?

@kananigit I'd like to migrate @ericnitschke's above recommendations to http://wiki.laptop.org/go/IIAB/FAQ#WordPress_.26_Moodle_Administration:_What_tips_.26_tricks_exist.3F if you confirm this approach should be broadly recommended?

kananigit, 2018-11-12:
Hi Adam,
The WP All plugin is pretty useful for moving  the whole  word-press
installation to another server. It overwrites all the content in the
destination word-press installation, so that's something to check out for.
Especially those people wanting to only transfer a few posts maybe.

So it is useful for WHOLE website migrations. We might wanna put that
warning out there. I have not tested it extensively but it does have a very
big following and user base. like 1 million active installs.

ericnitschke, 2018-11-12:
Thanks, @kananigit.  The default WP export/import tool does allow one to transfer only pages/posts, but for some reason it doesn't bring over attachments (media), so in the past I've had to manually migrate the images then reconnect them.  Keen to hear if you have another method of doing this.

holta, 2018-12-10:
I'm declaring victory now that this is documented in http://FAQ.IIAB.IO ("30 WordPress & Moodle Administration: What tips & tricks exist?")

But please re-open as nec, if critical issues are outstanding!

closed 2018-12-10 by holta