iiab/iiab #1306: Failed installation on ubuntu 18.04 (mini) using 6.7 1 line installer script

mrdavidhaag, 2018-11-24:
Trying to install iiab on VM in Proxmox using 1 line installer script 6.7 and installation failed at TASK 4

TASK [4-server-options : Restart dnsmasq] ******************************************************
fatal: [127.0.0.1]: FAILED! => {"changed": false, "msg": "Unable to start service dnsmasq: Job for dnsmasq.service failed because the control process exited with error code.\nSee \"systemctl status dnsmasq.service\" and \"journalctl -xe\" for details.\n"}
        to retry, use: --limit @/opt/iiab/iiab/iiab-stages.retry

PLAY RECAP *************************************************************************************
127.0.0.1                  : ok=176  changed=122  unreachable=0    failed=1

Tried using Do everything from Scratch method - same result dnsmasq: failed to create listening socket for port 53: Address already in use

netstat tells me that bind9 and systemd-resolve are both listening on port 53

root@box:/home/awm# netstat -nlpt | grep 53
tcp        0      0 10.10.35.20:53          0.0.0.0:*               LISTEN      417/named
tcp        0      0 127.0.0.1:53            0.0.0.0:*               LISTEN      417/named
tcp        0      0 127.0.0.53:53           0.0.0.0:*               LISTEN      404/systemd-resolve
tcp        0      0 127.0.0.1:953           0.0.0.0:*               LISTEN      417/named

So I tried to stop bind9 service and run iiab install again - same error

root@box:/opt/iiab/iiab# systemctl status dnsmasq.service
● dnsmasq.service - dnsmasq - A lightweight DHCP and caching DNS server
   Loaded: loaded (/lib/systemd/system/dnsmasq.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Sat 2018-11-24 10:58:06 WIT; 24s ago
  Process: 5557 ExecStart=/etc/init.d/dnsmasq systemd-exec (code=exited, status=2)
  Process: 5556 ExecStartPre=/usr/sbin/dnsmasq --test (code=exited, status=0/SUCCESS)

Nov 24 10:58:06 box.lan systemd[1]: Starting dnsmasq - A lightweight DHCP and caching DNS server...
Nov 24 10:58:06 box.lan dnsmasq[5556]: dnsmasq: syntax check OK.
Nov 24 10:58:06 box.lan dnsmasq[5557]: dnsmasq: failed to create listening socket for port 53: Address already in use
Nov 24 10:58:06 box.lan systemd[1]: dnsmasq.service: Control process exited, code=exited status=2
Nov 24 10:58:06 box.lan systemd[1]: dnsmasq.service: Failed with result 'exit-code'.
Nov 24 10:58:06 box.lan systemd[1]: Failed to start dnsmasq - A lightweight DHCP and caching DNS server.

If I stop systemd-resolve.service then the tasks fail at check for internet.

Any ideas or suggestions greatly appreciated.  My Linux skills are thin.

tim-moody, 2018-11-24:
@mrdavidhaag 
Some settings in /etc/iiab/local_vars.yml may conflict as there has been confusion between default_vars.yml and various local_vars.yml files in the 1 line installer. Check /etc/iiab/local_vars.yml and see if you have the following settings. When both named and dnsmasq are installed and enabled the conflict you describe will occur. Then rerun the install.

dhcpd_install: False
dhcpd_enabled: False
named_install: False
named_enabled: False
block_DNS: False

dnsmasq_install: True
dnsmasq_enabled: True

mrdavidhaag, 2018-11-24:
Thank you so much for your quick response and good advice.
I will try this on Monday.

holta, 2018-11-25:
> @mrdavidhaag
> Some settings in /etc/iiab/local_vars.yml may conflict as there has been confusion between default_vars.yml and various local_vars.yml files in the 1 line installer. Check [...]

@jvonau can you confirm?  And suggest alternatives for IIAB 6.7/master if dnsmasq is suddenly conflicting on Ubuntu 18.04 as is claimed above?

(Personally I've never had any such problems installing IIAB 6.7/master on Ubuntu 18.04 on VirtualBox, so I'd also ask if @mrdavidhaag 's use of Ubuntu 18.04 on ProxMox is perhaps somehow different than others' use of Ubuntu 18.04 on VirtualBox ?)

mrdavidhaag, 2018-11-25:
After the failure on ProxMox VM, I tried Build from Scratch procedure on
barebones NUC DC32171YE machine with Ubuntu minimal desktop installation
loaded.  Result was same dnsmasq error port 53 already in use at the TASK 4
stage.  On this exact same NUC machine about 2 weeks ago I successfully
installed iiab with 1 line installer script, so I do not believe that it is
a hardware issue.  Perhaps something with Ubuntu 18.04 or something changed
in the iiab scripts.

I have already looked at the /etc/iiab/local_vars.yml on both of the failed
installation machines and they only have;
dnsmasq_install: True
dnsmasq_enabled: True
The other mentioned settings are not listed or commented out in either
files.
dhcpd_install: False
dhcpd_enabled: False
named_install: False
named_enabled: False
block_DNS: False

The same is true for the /etc/iiab/local_vars.yml in the successful machine
from two weeks ago no mention of named or dhcpd

The co-installation of named and dnsmasq was the first suspect I had and
have already tried searching in the /etc/iiab/local_vars.yml to find if
settings were conflicting, but found no setting for installation of named.
However, when I watch during the installation, I can see the named
installation taking place, but am unable so far to find out where this is
coming from.

I will try to add the suggested False lines to the /etc/iiab/local_vars.yml
and try installation again.  I will report results here.

mrdavidhaag, 2018-11-25:
I added the suggested "False" lines just above the dnsmasq True statements
in the ?etc/iiab/local_vars.yml file as below screenshot on a fresh ProxMox
VM with Ubuntu 18.04 mini installation

[image: image.png]

This time bind - named did not install, however, same results with failure
in task4 trying to start dnsmasq service failed due to port 53 already in
use.

TASK [4-server-options : Restart apache2]
***************************************************************************************************************************************************************************************************
changed: [127.0.0.1]

TASK [4-server-options : Restart dnsmasq]
***************************************************************************************************************************************************************************************************
fatal: [127.0.0.1]: FAILED! => {"changed": false, "msg": "Unable to start
service dnsmasq: Job for dnsmasq.service failed because the control process
exited with error code.\nSee \"systemctl status dnsmasq.service\" and
\"journalctl -xe\" for details.\n"}
        to retry, use: --limit @/opt/iiab/iiab/iiab-stages.retry

PLAY RECAP
**********************************************************************************************************************************************************************************************************************************
127.0.0.1                  : ok=119  changed=80   unreachable=0    failed=1

root@awm:/opt/iiab/iiab# systemctl status dnsmasq.service
● dnsmasq.service - dnsmasq - A lightweight DHCP and caching DNS server
   Loaded: loaded (/lib/systemd/system/dnsmasq.service; enabled; vendor
preset: enabled)
   Active: failed (Result: exit-code) since Sun 2018-11-25 16:34:42 WIT;
46s ago
  Process: 2509 ExecStart=/etc/init.d/dnsmasq systemd-exec (code=exited,
status=2)
  Process: 2508 ExecStartPre=/usr/sbin/dnsmasq --test (code=exited,
status=0/SUCCESS)

Nov 25 16:34:42 box.lan systemd[1]: Starting dnsmasq - A lightweight DHCP
and caching DNS server...
Nov 25 16:34:42 box.lan dnsmasq[2508]: dnsmasq: syntax check OK.
Nov 25 16:34:42 box.lan dnsmasq[2509]: dnsmasq: failed to create listening
socket for port 53: Address already in use
Nov 25 16:34:42 box.lan systemd[1]: dnsmasq.service: Control process
exited, code=exited status=2
Nov 25 16:34:42 box.lan systemd[1]: dnsmasq.service: Failed with result
'exit-code'.
Nov 25 16:34:42 box.lan systemd[1]: Failed to start dnsmasq - A lightweight
DHCP and caching DNS server.
root@awm:/opt/iiab/iiab# netstat -nlpt | grep 53
tcp        0      0 127.0.0.53:53           0.0.0.0:*               LISTEN
    320/systemd-resolve
root@awm:/opt/iiab/iiab#

So the conflict does not seem to be bind-named but perhaps has to do with
systemd-resolve which is the only service listening to port 53 on
127.0.0.53:53 tcp as shown by netstat.

On this VM I can stop systemd-resolve.service and then start
dnsmasq.service without error, but then I cannot resolve any DNS to outside
world

root@awm:/opt/iiab/iiab# systemctl stop systemd-resolved.service
root@awm:/opt/iiab/iiab# systemctl start dnsmasq.service
root@awm:/opt/iiab/iiab# ping google.com
ping: google.com: Temporary failure in name resolution
root@awm:/opt/iiab/iiab# systemctl status dnsmasq.service
● dnsmasq.service - dnsmasq - A lightweight DHCP and caching DNS server
   Loaded: loaded (/lib/systemd/system/dnsmasq.service; enabled; vendor
preset: enabled)
   Active: active (running) since Sun 2018-11-25 16:38:09 WIT; 43s ago
  Process: 2542 ExecStartPost=/etc/init.d/dnsmasq systemd-start-resolvconf
(code=exited, status=0/SUCCESS)
  Process: 2533 ExecStart=/etc/init.d/dnsmasq systemd-exec (code=exited,
status=0/SUCCESS)
  Process: 2532 ExecStartPre=/usr/sbin/dnsmasq --test (code=exited,
status=0/SUCCESS)
 Main PID: 2541 (dnsmasq)
    Tasks: 1 (limit: 4666)
   CGroup: /system.slice/dnsmasq.service
           └─2541 /usr/sbin/dnsmasq -x /run/dnsmasq/dnsmasq.pid -u dnsmasq
-7 /etc/dnsmasq.d,.dpkg-dist,.dpkg-old,.dpkg-new --local-service
--trust-anchor=.,19036,8,2,49aac11d7b6f6446702e54a1607371607a1a41855200fd2ce1cdde32f24e8fb5
--tru

Nov 25 16:38:09 box.lan systemd[1]: Starting dnsmasq - A lightweight DHCP
and caching DNS server...
Nov 25 16:38:09 box.lan dnsmasq[2532]: dnsmasq: syntax check OK.
Nov 25 16:38:09 box.lan systemd[1]: Started dnsmasq - A lightweight DHCP
and caching DNS server.

root@awm:/opt/iiab/iiab#

Question:  Can IIAB run without dnsmasq.service?  I also tried to make the
variables for dnsmasq to false, but then the script checks dnsmasq anyways
and fails too.

holta, 2018-11-25:
> Perhaps something with Ubuntu 18.04 or something changed
in the iiab scripts.

Likely 18.04 and/or its latest version of dnsmasq changed?

Let's try to investigate on Raspbian Lite on Raspberry Pi 3 too, to shed light & solve this. 

Profound thanks @mrdavidhaag for the detailed report.

holta, 2018-11-25:
Related suggestion: PR #1303 ("bind9 and dnsmasq fighting over port 53")

tim-moody, 2018-11-25:
I agree with Adam that I have not seen this problem on VirtualBox, so your environment should be considered. 

> Proxmox VE is a complete open-source platform for all-inclusive enterprise virtualization that tightly integrates KVM hypervisor and LXC containers, software-defined storage and networking functionality on a single platform, and easily manages high availability clusters and disaster recovery tools with the built-in web management interface.

My experience is that containers, certainly docker, come with some networking already configured that can conflict with IIAB installation.

holta, 2018-11-25:
@jvonau writes:

> Disable and stop is what appears to be needed but that may not be the complete solution..

mrdavidhaag, 2018-11-25:
Thank you.  I have tried this before, however the installation fails when
trying to access internet.
If I stop systemd-resolve.service then the VM cannot resolve DNS as below
showing before and after systemd-resolved.service is stopped (the long ping
times to google.com are result of our Vsat service here in Papua, Indonesia)

*BEFORE*
root@awm:/home/awm#
root@awm:/home/awm# ping google.com
PING google.com (216.58.223.14) 56(84) bytes of data.
64 bytes from jnb01s07-in-f14.1e100.net (216.58.223.14): icmp_seq=1 ttl=41
time=979 ms
64 bytes from jnb01s07-in-f14.1e100.net (216.58.223.14): icmp_seq=2 ttl=41
time=976 ms
64 bytes from jnb01s07-in-f14.1e100.net (216.58.223.14): icmp_seq=3 ttl=41
time=1052 ms
64 bytes from jnb01s07-in-f14.1e100.net (216.58.223.14): icmp_seq=4 ttl=41
time=1067 ms
64 bytes from jnb01s07-in-f14.1e100.net (216.58.223.14): icmp_seq=5 ttl=41
time=960 ms
64 bytes from jnb01s07-in-f14.1e100.net (216.58.223.14): icmp_seq=6 ttl=41
time=1236 ms
64 bytes from jnb01s07-in-f14.1e100.net (216.58.223.14): icmp_seq=7 ttl=41
time=1050 ms
^C
--- google.com ping statistics ---
8 packets transmitted, 7 received, 12% packet loss, time 11306ms
rtt min/avg/max/mdev = 960.186/1046.257/1236.852/87.406 ms, pipe 2
root@awm:/home/awm# systemctl status systemd-resolved.service
● systemd-resolved.service - Network Name Resolution
   Loaded: loaded (/lib/systemd/system/systemd-resolved.service; enabled;
vendor preset: enabled)
   Active: active (running) since Sat 2018-11-24 08:10:32 WIT; 1 day 7h ago
     Docs: man:systemd-resolved.service(8)
           https://www.freedesktop.org/wiki/Software/systemd/resolved

https://www.freedesktop.org/wiki/Software/systemd/writing-network-configuration-managers

https://www.freedesktop.org/wiki/Software/systemd/writing-resolver-clients
 Main PID: 310 (systemd-resolve)
   Status: "Processing requests..."
    Tasks: 1 (limit: 4666)
   CGroup: /system.slice/systemd-resolved.service
           └─310 /lib/systemd/systemd-resolved

Nov 24 08:10:32 awm systemd-resolved[310]: Negative trust anchors:
10.in-addr.arpa 16.172.in-addr.arpa 17.172.in-addr.arpa 18.172.in-addr.arpa
19.172.in-addr.arpa 20.172.in-addr.arpa 21.172.in-addr.arpa
22.172.in-addr.arpa 23.172.in-addr
Nov 24 08:10:32 awm systemd-resolved[310]: Using system hostname 'awm'.
Nov 24 08:10:32 awm systemd[1]: Started Network Name Resolution.
Nov 24 08:35:17 awm systemd-resolved[310]: Using degraded feature set (UDP)
for DNS server 10.10.35.1.
Nov 24 11:15:07 awm systemd-resolved[310]: Grace period over, resuming full
feature set (UDP+EDNS0) for DNS server 10.10.35.1.
Nov 24 11:15:07 awm systemd-resolved[310]: Using degraded feature set (UDP)
for DNS server 10.10.35.1.
Nov 25 15:37:55 awm systemd-resolved[310]: Server returned error NXDOMAIN,
mitigating potential DNS violation DVE-2018-0001, retrying transaction with
reduced feature level UDP.
Nov 25 15:37:55 awm systemd-resolved[310]: Using degraded feature set (UDP)
for DNS server 10.10.35.1.
Nov 25 15:44:35 awm systemd-resolved[310]: Grace period over, resuming full
feature set (UDP+EDNS0) for DNS server 10.10.35.1.
Nov 25 15:44:35 awm systemd-resolved[310]: Using degraded feature set (UDP)
for DNS server 10.10.35.1.

root@awm:/home/awm# netstat -lnpt | grep 53
tcp        0      0 127.0.0.53:53           0.0.0.0:*               LISTEN
    310/systemd-resolve

*AFTER*
root@awm:/home/awm# systemctl stop systemd-resolved.service
root@awm:/home/awm# netstat -lnpt | grep 53
root@awm:/home/awm# ping google.com
ping: google.com: Temporary failure in name resolution
root@awm:/home/awm# systemctl status systemd-resolved.service
● systemd-resolved.service - Network Name Resolution
   Loaded: loaded (/lib/systemd/system/systemd-resolved.service; enabled;
vendor preset: enabled)
   Active: inactive (dead) since Sun 2018-11-25 15:46:11 WIT; 32s ago
     Docs: man:systemd-resolved.service(8)
           https://www.freedesktop.org/wiki/Software/systemd/resolved

https://www.freedesktop.org/wiki/Software/systemd/writing-network-configuration-managers

https://www.freedesktop.org/wiki/Software/systemd/writing-resolver-clients
  Process: 310 ExecStart=/lib/systemd/systemd-resolved (code=exited,
status=0/SUCCESS)
 Main PID: 310 (code=exited, status=0/SUCCESS)
   Status: "Shutting down..."

Nov 24 08:10:32 awm systemd[1]: Started Network Name Resolution.
Nov 24 08:35:17 awm systemd-resolved[310]: Using degraded feature set (UDP)
for DNS server 10.10.35.1.
Nov 24 11:15:07 awm systemd-resolved[310]: Grace period over, resuming full
feature set (UDP+EDNS0) for DNS server 10.10.35.1.
Nov 24 11:15:07 awm systemd-resolved[310]: Using degraded feature set (UDP)
for DNS server 10.10.35.1.
Nov 25 15:37:55 awm systemd-resolved[310]: Server returned error NXDOMAIN,
mitigating potential DNS violation DVE-2018-0001, retrying transaction with
reduced feature level UDP.
Nov 25 15:37:55 awm systemd-resolved[310]: Using degraded feature set (UDP)
for DNS server 10.10.35.1.
Nov 25 15:44:35 awm systemd-resolved[310]: Grace period over, resuming full
feature set (UDP+EDNS0) for DNS server 10.10.35.1.
Nov 25 15:44:35 awm systemd-resolved[310]: Using degraded feature set (UDP)
for DNS server 10.10.35.1.
Nov 25 15:46:11 awm systemd[1]: Stopping Network Name Resolution...
Nov 25 15:46:11 awm systemd[1]: Stopped Network Name Resolution.
root@awm:/home/awm#

holta, 2018-11-26:
@jvonau writes:

> I'm away from a computer  ATM... Seems the subject VM/pre-canned hosting service is setup slightly different from a stock real hardware Ubuntu install. [?]

> Bigger questions I have if this is a single interface VM why is captive portal being enabled at that point in the install? Before the number of network interfaces is calulated? There would be no point in turning on captive portal with only a single interface involved in the same way squid is only activated when there is a gateway present... Think CP should only be enabled when there is a LAN interface present..

holta, 2018-11-27:
@georgejhunt & all, do have a moment to respond to the ideas/questions above prior to Thursday's http://minutes.iiab.io 10AM NYC Time call?

Can someone with access to RPi 3 and Ubuntu 18.04 validate the situation there, in a known environment on known hardware or on a known VM environment?

mrdavidhaag, 2018-11-27:
Thank you for the follow up. Much appreciate your time and efforts.

Our goal is to test iiab as educational content server on Community
Cellular LTE
<https://www.internetsociety.org/blog/2018/09/building-a-community-lte-network-in-bokondini-indonesia/>
deployment in Papua, Indonesia. The idea is for users of the 4G signal to
have unlimited, non-revenue access to iiab educational content to make use
of available bandwidth.  We have limited internet backhaul due to Vsat, but
have an abundance of on-site bandwidth.  Hopefully integration of iiab will
allow us to put the excess bandwidth to good use.

georgejhunt, 2018-11-27:
I agree with Jerry. We should be running in appliance mode (only one
adapter). This information seems important:

"Our goal is to test iiab as educational content server on Community
Cellular LTE"  -- on a preexisting network structure with ip addresses and
dns handled by someone else

You can verify the automatic config decisions our code made by reading
/etc/iiab/iiab.ini -- search for "iiab_network_mode"
At the same location, verify that named, dnsmasq_enabled, dhcpd are all
False. Jerry is more able than I to figure out the logic in
/opt/iiab/iiab/roles/network/tasks/computed_services.yml. It looks to me
that there may need to be some tweaking.

In the meantime,  I think the VM's name resolution on port 53 must already
be in place. So I'd try these in local-vars.yml:
captive_portal_install: True
captive_portal_enabled: False
dnsmasq_install: True
dnsmasq_enabled: False

holta, 2018-11-28:
IIAB 6.7/master is definitely broken on Ubuntu 18.04 as tested on multiple NUCs: (unlike on Raspbian 2018-11-13 where IIAB 6.7/master still works)
```
TASK [4-server-options : Restart dnsmasq] **************************************
fatal: [127.0.0.1]: FAILED! => {"changed": false, "msg": "Unable to start service dnsmasq: Job for dnsmasq.service failed because the control process exited with error code.\nSee \"systemctl status dnsmasq.service\" and \"journalctl -xe\" for details.\n"}
```
Further detail:
```
root@box:~# systemctl status dnsmasq
● dnsmasq.service - dnsmasq - A lightweight DHCP and caching DNS server
   Loaded: loaded (/lib/systemd/system/dnsmasq.service; enabled; vendor preset: 
   Active: failed (Result: exit-code) since Tue 2018-11-27 18:57:11 EST; 12min a
  Process: 14669 ExecStart=/etc/init.d/dnsmasq systemd-exec (code=exited, status
  Process: 14662 ExecStartPre=/usr/sbin/dnsmasq --test (code=exited, status=0/SU

Nov 27 18:57:11 box.lan systemd[1]: Starting dnsmasq - A lightweight DHCP and ca
Nov 27 18:57:11 box.lan dnsmasq[14662]: dnsmasq: syntax check OK.
Nov 27 18:57:11 box.lan dnsmasq[14669]: dnsmasq: failed to create listening sock
Nov 27 18:57:11 box.lan systemd[1]: dnsmasq.service: Control process exited, cod
Nov 27 18:57:11 box.lan systemd[1]: dnsmasq.service: Failed with result 'exit-co
Nov 27 18:57:11 box.lan systemd[1]: Failed to start dnsmasq - A lightweight DHCP
lines 1-12/12 (END)...skipping...
● dnsmasq.service - dnsmasq - A lightweight DHCP and caching DNS server
   Loaded: loaded (/lib/systemd/system/dnsmasq.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Tue 2018-11-27 18:57:11 EST; 12min ago
  Process: 14669 ExecStart=/etc/init.d/dnsmasq systemd-exec (code=exited, status=2)
  Process: 14662 ExecStartPre=/usr/sbin/dnsmasq --test (code=exited, status=0/SUCCESS)

Nov 27 18:57:11 box.lan systemd[1]: Starting dnsmasq - A lightweight DHCP and caching DNS server...
Nov 27 18:57:11 box.lan dnsmasq[14662]: dnsmasq: syntax check OK.
Nov 27 18:57:11 box.lan dnsmasq[14669]: dnsmasq: failed to create listening socket for 172.18.96.1: Address already in use
Nov 27 18:57:11 box.lan systemd[1]: dnsmasq.service: Control process exited, code=exited status=2
Nov 27 18:57:11 box.lan systemd[1]: dnsmasq.service: Failed with result 'exit-code'.
Nov 27 18:57:11 box.lan systemd[1]: Failed to start dnsmasq - A lightweight DHCP and caching DNS server.
```
Thanks all for helping to narrow down what broke IIAB 6.7/master &mdash; which worked 2 weeks ago.

mrdavidhaag, 2018-11-28:
> Thanks all for helping to narrow down what broke IIAB 6.7/master — which

Thank you
Made the following changes in local-vars.yml:
captive_portal_install: True
captive_portal_enabled: False
dnsmasq_install: True
dnsmasq_enabled: False

and happy to report successful installation of iiab-install

this is on VM running Ubuntu 18.04-mini in ProxMox VME using the Do
everything from Scratch method

I have not tried 1 line installer-script (with edits)

Thank you all for super support and clear concise good advice

holta, 2018-11-30:
Clean install of IIAB 6.7/master failed on a 100% fresh Ubuntu Server 18.04.1 VM:

> TASK [4-server-options : Restart dnsmasq] **************************************
fatal: [127.0.0.1]: FAILED! => {"changed": false, "msg": "Unable to start service dnsmasq: Job for dnsmasq.service failed because the control process exited with error code.\nSee \"systemctl status dnsmasq.service\" and \"journalctl -xe\" for details.\n"}

George / Tim indicated yesterday (during our voice call) that the above would work.  It did not.  Seems we have some kind of common/intermittent failure?  FYI, all updates were applied to the VM prior to beginning, as per standard install recommendations, using `apt update && apt -y dist-upgrade && reboot`

@jvonau writes:

> U-18 Server most likely has changed in that systemd-resolved is now enabled to provide a local DNS server.. ip a and the contents of /etc/resolve.conf would be helpful.. can't VPN in ATM heading to work... 

> If CP is to be stand-alone please clarify the use of code in iiab-gen-iptables that is part of the original PR.

holta, 2018-11-30:
@jvonau requested output of `ip a` and `more /etc/resolv.conf` from the above failed IIAB 6.7/master install on Ubuntu Server 18.04.1:
```
root@box:~# ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
2: enp0s3: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UP group default qlen 1000
    link/ether 08:00:27:d1:14:a4 brd ff:ff:ff:ff:ff:ff
    inet 192.168.0.161/24 brd 192.168.0.255 scope global enp0s3
       valid_lft forever preferred_lft forever
3: tun0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UNKNOWN group default qlen 100
    link/none
    inet 10.8.0.50 peer 10.8.0.49/32 scope global tun0
       valid_lft forever preferred_lft forever
```
```
root@box:~# more /etc/resolv.conf
# This file is managed by man:systemd-resolved(8). Do not edit.
#
# This is a dynamic resolv.conf file for connecting local clients to the
# internal DNS stub resolver of systemd-resolved. This file lists all
# configured search domains.
#
# Run "systemd-resolve --status" to see details about the uplink DNS servers
# currently in use.
#
# Third party programs must not access this file directly, but only through the
# symlink at /etc/resolv.conf. To manage man:resolv.conf(5) in a different way,
# replace this symlink by a static file or a different symlink.
#
# See man:systemd-resolved.service(8) for details about the supported modes of
# operation for /etc/resolv.conf.

nameserver 127.0.0.53
```

holta, 2018-11-30:
"dnsmasq & systemd Causing Intermittent CPU Spikes"
https://unix.stackexchange.com/questions/417645/dnsmasq-systemd-causing-intermittent-cpu-spikes

@jvonau wonders if the above might be related?

holta, 2018-11-30:
> Clean install of IIAB 6.7/master failed on a 100% fresh Ubuntu Server 18.04.1 VM:

FYI I get the exact same/above failure on 2 other Ubuntu Server 18.04.1 machines, that happen to be NUC PC's (10.8.0.6 & 10.8.0.34) rather than VM's.

FWIW these NUC PC's were updated using `apt update && apt -y dist-upgrade && reboot` prior to running `cd /opt/iiab/iiab && git pull && ./iiab-install --reinstall`

jvonau, 2018-12-01:
The primary issue is dnsmasq fails to start when being installed by the system package manager so this is an upstream problem, search Ubuntu's bug tracker for other who have run across this issue or file a bug. I'll bet this could be reproduced by installing dnsmasq on a new VM without using iiab with just apt. The above observation on real hardware points to dnsmasq breaking on currently configured out in the wild machines that most likely will receive an update and suddenly break. The above link points to a possible solution to deal with the issue with systemd-resolved and dnsmasq.

holta, 2018-12-01:
@jvonau & All, there are many concrete suggestions here:

How to avoid conflicts between dnsmasq and systemd-resolved?https://unix.stackexchange.com/questions/304050/how-to-avoid-conflicts-between-dnsmasq-and-systemd-resolved offers many suggestions

Which do you suggest we try?

holta, 2018-12-01:
@tim-moody asks:

> isn't the answer just to stop and disable systemd-resolved?

holta, 2018-12-01:
> The primary issue is dnsmasq fails to start when being installed by the system package manager so this is an upstream problem, search Ubuntu's bug tracker for other who have run across this issue or [...]

@jvonau's above theory is confirmed: dnsmasq can no longer be installed on (updated) Ubuntu 18.04.1 VM's &mdash; as of about mid-November 2018.

I've tested these 2 solutions on a clean/minimal (updated) Ubuntu 18.04.1 VM: (prior to installing IIAB, or anything)

- adding `DNSStubListener=no` to /etc/systemd/resolved.conf works, allowing both services to auto-launch during U18 boot, which is the consensus solution according to [StackExchange](https://unix.stackexchange.com/questions/304050/how-to-avoid-conflicts-between-dnsmasq-and-systemd-resolved)
- `systemctl disable systemd-resolved && reboot` also works, as @tim-moody's suggested (but raises questions about the risks of IIAB diverging from Ubuntu mainline forward path)

I'd like an carefully considered solution that keeps in line with Ubuntu 18.04's trajectory long-term &mdash; so weeks of timewasting hopefully doesn't happen again :-)  FYI Ubuntu 18.04.1 currently uses systemd 237, Raspbian Lite currently uses systemd 232, and the [latest release](https://github.com/systemd/systemd/releases) from 2018-06-22 is 239 &mdash; all of which properly support the above syntax `DNSStubListener=no` if we go this route.

holta, 2018-12-02:
@jvonau can you offer a dnsmasq + systemd-resolved recommendation above?

jvonau, 2018-12-03:
I'd be leaning towards the "bind-interfaces" solution.

holta, 2018-12-03:
Hmm, putting `DNSStubListener=no` into /etc/systemd/resolved.conf on a clean Ubuntu Server 18.04 VM (as recommended by [StackExchange](https://unix.stackexchange.com/questions/304050/how-to-avoid-conflicts-between-dnsmasq-and-systemd-resolved)) might not be enough &mdash; when looking more closely at this VM a day later:

1. DNS fails to resolve (any name on the public Internet)
2. "systemctl status dnsmasq" shows this disturbing output:

> root@u18-server:~# systemctl status dnsmasq
● dnsmasq.service - dnsmasq - A lightweight DHCP and caching DNS server
   Loaded: loaded (/lib/systemd/system/dnsmasq.service; enabled; vendor preset: enabled)
   Active: active (running) since Sat 2018-12-01 12:35:39 EST; 1 day 12h ago
  Process: 405 ExecStartPost=/etc/init.d/dnsmasq systemd-start-resolvconf (code=exited, status=0/SUCCESS)
  Process: 367 ExecStart=/etc/init.d/dnsmasq systemd-exec (code=exited, status=0/SUCCESS)
  Process: 319 ExecStartPre=/usr/sbin/dnsmasq --test (code=exited, status=0/SUCCESS)
 Main PID: 404 (dnsmasq)
    Tasks: 1 (limit: 2325)
   CGroup: /system.slice/dnsmasq.service
           └─404 /usr/sbin/dnsmasq -x /run/dnsmasq/dnsmasq.pid -u dnsmasq -7 /etc/dnsmasq.d,.dpkg-dist,.dpkg-old,.dpkg-new --local-service --trust-anchor=.,19036,8,2,49
>
> Dec 03 00:49:49 u18-server dnsmasq[404]: Maximum number of concurrent DNS queries reached (max: 150)
Dec 03 00:49:59 u18-server dnsmasq[404]: Maximum number of concurrent DNS queries reached (max: 150)
Dec 03 00:50:09 u18-server dnsmasq[404]: Maximum number of concurrent DNS queries reached (max: 150)
Dec 03 00:50:19 u18-server dnsmasq[404]: Maximum number of concurrent DNS queries reached (max: 150)
Dec 03 00:50:29 u18-server dnsmasq[404]: Maximum number of concurrent DNS queries reached (max: 150)
Dec 03 00:50:39 u18-server dnsmasq[404]: Maximum number of concurrent DNS queries reached (max: 150)
Dec 03 00:50:49 u18-server dnsmasq[404]: Maximum number of concurrent DNS queries reached (max: 150)
Dec 03 00:50:59 u18-server dnsmasq[404]: Maximum number of concurrent DNS queries reached (max: 150)
Dec 03 00:51:09 u18-server dnsmasq[404]: Maximum number of concurrent DNS queries reached (max: 150)
Dec 03 00:51:19 u18-server dnsmasq[404]: Maximum number of concurrent DNS queries reached (max: 150)

holta, 2018-12-03:
After uncommenting `bind-interfaces` from /etc/dnsmasq.conf (and removing `DNSStubListener=no` from /etc/systemd/resolved.conf) and rebooting:

- DNS works
- `systemctl status systemd-resolved` returns a couple errors at the bottom here: _(is this serious?)_

> root@u18-server:~# systemctl status systemd-resolved
● systemd-resolved.service - Network Name Resolution
   Loaded: loaded (/lib/systemd/system/systemd-resolved.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2018-12-03 01:03:32 EST; 5min ago
     Docs: man:systemd-resolved.service(8)
           https://www.freedesktop.org/wiki/Software/systemd/resolved
           https://www.freedesktop.org/wiki/Software/systemd/writing-network-configuration-managers
           https://www.freedesktop.org/wiki/Software/systemd/writing-resolver-clients
 Main PID: 299 (systemd-resolve)
   Status: "Processing requests..."
    Tasks: 1 (limit: 2325)
   CGroup: /system.slice/systemd-resolved.service
           └─299 /lib/systemd/systemd-resolved
>
> Dec 03 01:03:32 u18-server systemd[1]: Starting Network Name Resolution...
Dec 03 01:03:32 u18-server systemd-resolved[299]: Positive Trust Anchors:
Dec 03 01:03:32 u18-server systemd-resolved[299]: . IN DS 19036 8 2 49aac11d7b6f6446702e54a1607371607a1a41855200fd2ce1cdde32f24e8fb5
Dec 03 01:03:32 u18-server systemd-resolved[299]: . IN DS 20326 8 2 e06d44b80b8f1d39a95c0b0d7c65d08458e880409bbc683457104237c7f8ec8d
Dec 03 01:03:32 u18-server systemd-resolved[299]: Negative trust anchors: 10.in-addr.arpa 16.172.in-addr.arpa 17.172.in-addr.arpa 18.172.in-addr.arpa 19.172.in-addr.arp
Dec 03 01:03:32 u18-server systemd-resolved[299]: Using system hostname 'u18-server'.
Dec 03 01:03:32 u18-server systemd[1]: Started Network Name Resolution.
Dec 03 01:03:36 u18-server systemd-resolved[299]: Server returned error NXDOMAIN, mitigating potential DNS violation DVE-2018-0001, retrying transaction with reduced fe
Dec 03 01:03:36 u18-server systemd-resolved[299]: Server returned error NXDOMAIN, mitigating potential DNS violation DVE-2018-0001, retrying transaction with reduced fe

holta, 2018-12-03:
This should be resolved by @jvonau's PR #1318 &mdash; thanks all for continuing to test!

holta, 2018-12-03:
Confirmed resolved by PR #1318 with a clean/fresh install on an Ubuntu 18.04.1 VM (exactly what had been failing in recent weeks).

_(Thanks all for ongoing feedback here or on a new tkt as appropriate!)_

holta, 2018-12-03:
It's possible this might not be fully fixed by PR #1318 ?

In any case, please follow up at #1319, Thanks All!

AaronNBrock, 2019-03-07:
Hello,

I'm running into this issue again running on a fresh install of Ubuntu Server 18.04.2 on a Lenovo Thinkpad T440p.

holta, 2019-03-07:
@AaronNBrock

1) Are you hopefully installing the latest IIAB 7.0/master using http://d.iiab.io ?  Or are you using the older http://d.iiab.io/6.7 ?
2) What exact kernel are you running?  (Ubuntu 18.04.2 can mean 4.15 or 4.18)
3) Please post a complete screenshot (photo) or pastebin of the full error sequence.
4) Are you installing MIN-sized, MEDIUM-sized, or BIG-sized?  Have you modified /etc/iiab/local_vars.yml ?
5) Please post your /etc/iiab/iiab.env and /etc/iiab/iiab.ini as well as the results of running "ip a"
6) Have you re-run "sudo iiab" to see if that helps?
7) Have you tried "systemctl restart dnsmasq" to see if that helps?

Thanks!

holta, 2019-03-07:
@AaronNBrock please also confirm whether you're using Ubuntu 18.04.2 Server or Desktop, as there are huge differences in the underlying networking (Server uses `systemd-networkd` whereas Desktop uses `NetworkManager`).

Let us know if you see "renderer: networkd" or "renderer: NetworkManager" within `/etc/netplan/01-iiab-config.yaml`, Thanks!

AaronNBrock, 2019-03-07:
@holta 

I'm using Ubuntu **Server** 18.04.2, however, I tried Desktop but ran into different issues.

1. I'm installing with the one-line installer: `curl d.iiab.io/install.txt | sudo bash` (which I believe is 7.0?)

2. `uname -a` shows that the kernel is version `Linux box.lan 4.15.0-46-generic` 

3. The error is:
```TASK [network : dhcp_server may be affected - starting -user choice] ******************************************************
fatal: [127.0.0.1]: FAILED! => {"changed": false, "msg": "Unable to start service dnsmasq: Job for dnsmasq.service failed because the control process exited with error code.\nSee "systemctl status dnsmasq.service" and "journalctl -xe" for details.\n"}
to retry, use: --limit @/opt/iiab/iiab/iiab-stages.retry```

4. I receive the same error when using MIN-sized or BIG-sized (I've not tested MEDIUM).  And the only change I made was `default_language: es`.

5. https://gist.github.com/AaronNBrock/0019c0e67c74fb058227b1d1eabd6173

(To be continued, I'm still working on some other questions, but I wanted to share this asap...)

holta, 2019-03-07:
@AaronNBrock we also need the output of your running "ip a", please post that Thanks!

AaronNBrock, 2019-03-07:
@holta 

Oops, missed that!  Here you go: https://gist.github.com/AaronNBrock/9f98dbc363416fd66fbec7f015e68794

AaronNBrock, 2019-03-07:
@holta 

Okay, so:

6. re-running `sudo iiab` fails at the `sudo apt-key adv --keyserver ...` step with the following error:
```
gpg: keyserver receive failed: Server indicated a failure
```

7. After running `sudo systemctl restart dnsmasq`, dnsmasq seems to start happily (as seen via `systemctl status dnsmasq`.  However, rerunning `sudo iiab` after this still results in the error from 6.

holta, 2019-03-07:
> 6. re-running `sudo iiab` fails at the `sudo apt-key adv --keyserver ...` step with the following error:
> 
> ```
> gpg: keyserver receive failed: Server indicated a failure
> ```

@AaronNBrock need more context to know what exact stage this is arising from.  Please paste in the _full_ command and the the _full_ error message, Thanks!

AaronNBrock, 2019-03-07:
@holta 

Here's the complete `script` of the command: https://gist.github.com/AaronNBrock/b001a65fa15875836c1c2afe7408a453

holta, 2019-03-07:
@AaronNBrock our DNS failing creates symptoms like this `sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-keys 93C4A3FD7BB9C367` failure, got it!

We need to figure out what's blocking port 53 on your system &mdash; you're trying to install 2 different DNS systems by accident &mdash; and we need to figure out why.

Keep running `systemctl restart dnsmasq` and whenever that fails please tell us the output of `netstat -naup | grep :53` so we can try to figure this out, Thanks!

AaronNBrock, 2019-03-07:
I've been running `system restart dnsmasq` but it hasn't failed, I've also tried re-imagining the device and going through the install process and trying `netstat - naup | grep :53` but even that doesn't show up with anything.

holta, 2019-03-07:
You want "netstat -naup | grep :53" check carefully!!

_(Somehow an extra/errorneous space crept into what you have above...)_

AaronNBrock, 2019-03-07:
Ah, okay, yep, that was a typo, here's the output:

```
udp        0      0 127.0.0.1:53            0.0.0.0:*                           1681/dnsmasq        
udp        0      0 172.18.96.1:53          0.0.0.0:*                           1681/dnsmasq        
udp        0      0 127.0.0.53:53           0.0.0.0:*                           811/systemd-resolve 
udp        0      0 0.0.0.0:5353            0.0.0.0:*                           1022/avahi-daemon:  
udp6       0      0 :::5353                 :::*                                1022/avahi-daemon:  
```

holta, 2019-03-07:
@AaronNBrock thx: that's looks normal with dnsmasq on Port 53.

1) Plz run `netstat -naup | grep :53` immediately after the next failure(s) and let us know the results.
2) Perhaps first run `cd /opt/iiab/iiab` then `./iiab-network` to try to reproduce the failure more efficiently?
3) If nec should we connect live with a screenshare or similar later?

AaronNBrock, 2019-03-07:
@holta 

The only way I'm able to cause a failure has been to completely wipe and reinstall it, and, running the command immediatly after using: `sudo iiap; sudo netstat -naup | grep :53` resulted in:
```
udp        0      0 127.0.0.53:53           0.0.0.0:*                           841/systemd-resolve 
udp        0      0 0.0.0.0:5353            0.0.0.0:*                           14423/avahi-daemon:  
udp6       0      0 :::5353                 :::*                                14423/avahi-daemon:  
```

I'm not really sure how to screen share besides just setting it up as an ssh server or something along those lines, would that suffice?

holta, 2019-03-07:
You must mean `sudo iiab` but understood: very helpdul, thanks Aaron.

Let me ask @jvonau for suggestions to see if a call or otherwise might be best to resolve this.

holta, 2019-03-07:
@AaronNBrock can you please pastebin (i.e. post somewhere) the entire /opt/iiab/iiab/iiab-install.log ?

The last few pages are the most important, but best if we have the whole thing to diagnose &mdash; there's a chance this is an UBuntu timing quirk relating to 127.0.0.53

PS this file can be 3000+ lines, so it may be easiest to upload it as follows:

- systemctl restart dnsmasq
- apt install pastbinit
- cat /opt/iiab/iiab/iiab-install.log | pastebinit

Then send us the URL, Thanks!

MrSteve2, 2019-03-08:
@holta Im taking over for @AaronNBrock  

Here is the log requested: [http://paste.ubuntu.com/p/P69cWkhmRK/](http://paste.ubuntu.com/p/P69cWkhmRK/)

Also would it make sense to try Debain? Or will that just be other issues?

holta, 2019-03-08:
> Here is the log requested: http://paste.ubuntu.com/p/P69cWkhmRK/

Confusing as it seems to be a "Desktop OS" not "Server OS", in contrast to what Aaron said earlier?  Evidence being that it appears to be running NetworkManager, as can be confirmed in /etc/netplan

In any case, can you please pastebin /opt/iiab/iiab/iiab-network.log which should be the final one at long last one~

> Also would it make sense to try Debain? Or will that just be other issues?

Sure, give Debian a shot on another ThinkPad in parallel in that's easy!

holta, 2019-03-08:
@MrSteve2 when you look at the very last line of http://paste.ubuntu.com/p/P69cWkhmRK/ that you posted a few hours ago, it would seem to confirm that the core IIAB install in fact _did not fail!_

> 2019-03-07 15:40:58,426 p=6848 u=root |  127.0.0.1                  : ok=320  changed=216  unreachable=0    failed=0 

Can you confirm that?  Is this perhaps related to your attempting "18.04 Desktop" instead of Aaron's earlier attempts on "18.04 Server", as the logs (containing many references to NetworkManager) would seem to indicate?

MrSteve2, 2019-03-08:
@holta correct those logs were from Desktop version.

holta, 2019-03-11:
> @holta correct those logs were from Desktop version.

@MrSteve2 did you make progress using Ubuntu 18.04.2 Desktop?
(Or using Debian 9.8 Server...or Desktop possibly?)

If so, definitely check out http://FAQ.IIAB.IO #15 and let us know:

"How do I customize my Internet-in-a-Box home page?"

holta, 2019-03-14:
Seems unrelated to:

#1569 dnsmasq sometimes fails to start on Raspbian Desktop (possibly also Lite?)

MrSteve2, 2019-03-15:
Yes thanks. I was able to get it working with Debian.  I plan on building a
few more and shipping to Haiti.  Really appreciate all your help.

Stephen

m-anish, 2019-03-19:
Okay, this is failing for me too:
http://paste.ubuntu.com/p/4njWzCnTtW/

Setup as follows:
Hardware:
* Intel NUC

Network:
* Internet through USB tethering
* LAN port left empty
* IIAB should setup in Gateway mode. 

Latest commit hash: 23286e14b39957b8a9f702408725715e113fe862

OS:
Ubuntu Server 18.04.2

holta, 2019-03-19:
@m-anish Ubuntu 18.04.2 can mean kernel 4.15 or 4.18 — plz confirm your exact kernel?

The output of `ip a` and `cat /etc/iiab/iiab.env` also give critical context for @jvonau

m-anish, 2019-03-19:
Running `./iiab-install --reinstall` fixes this.

m-anish, 2019-03-19:
* [iiab.env](http://paste.ubuntu.com/p/VPBmrv3j2Z/)
* [iiab.ini](http://paste.ubuntu.com/p/5tk83xmMWh/)
* `Linux box.lan 4.15.0-46-generic #49-Ubuntu SMP Wed Feb 6 09:33:07 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux`

holta, 2019-03-19:
> Running `./iiab-install --reinstall` fixes this.

Did you happen to try repeated runs of `./iiab-network` prior to that?  What were the results if so?

(We need to publish a workaround if possible, as this error is proving to be quite common.)

[7 more replies]

closed 2019-07-09 by holta