iiab/iiab #1307: iiab-make-kiwix-lib won't run without test.zim in content folder

ericnitschke, 2018-11-26:
### Expected behavior and actual behavior.

### Steps to reproduce the problem.

### Specifications like the version of the IIAB, operating system version, or hardware details.

-Manually added ZIMs to content folder from USB.
-Deleted test.zim thinking it wasn't necessary.
-Ran iiab-make-kiwix-lib to refresh Kiwix library. Failed because test.zim was gone.
-Copied test.zim from a different box to the content folder.
-Re-ran iiab-make-kiwix-lib and it worked fine.

Would be good to add the following steps (or something similar) to https://github.com/iiab/iiab/wiki/IIAB-Installation#add-content-manually

To add ZIMs to Kiwix with USB stick
First, move the files to the Box:
-Add desired ZIMs to USB stick.
-Plug stick into Box
-Copy ZIM files to library/zims/content

Now you want to update the Kiwix library with the new ZIMs (at box.lan/kiwix):
-Open terminal
-Run sudo iiab-make-kiwix-lib

holta, 2018-11-26:
@tim-moody is this expected and/or too hard to fix?

tim-moody, 2018-11-26:
@ericnitschke Please verify you are using the latest iiab master and latest iiab-admin-console as kiwix changed how deletions work.  iiab-make-kiwix-lib will not work if /library/zims/library.xml does not exist and will not work if there is not at least one zim, but it does not have to be test.zim.

root@box:/library/zims/content# ll
total 1479632
drwxr-xr-x 2 root root       4096 Nov 26 17:27 ./
drwxr-xr-x 4 root root       4096 Oct 22 19:41 ../
-rw-r--r-- 1 root root      53357 Nov 26 17:27 test.zim
-rw-r--r-- 1 root root 1208191347 Sep 20 18:47 wikipedia_en_medicine_novid_2018-09.zim
-rw-r--r-- 1 root root  306867359 Sep 22 01:22 wikipedia_es_medicine_novid_2018-09.zim

iiab-make-kiwix-lib

library.xml contains  <book id="c2b8c685-6a6f-edf7-c8d8-a5005144c8bf" path="content/test.zim" title="test" description="test"
http://192.168.0.85/kiwix/ shows test

rm test.zim

rerun iiab-make-kiwix-lib
test.zim is not in library.xml
http://192.168.0.85/kiwix/ does not show test

This issue can not be reproduced on Ubuntu 18.04 VPN.

holta, 2018-11-26:
@tim-moody asks if these related questions can be answered:

https://github.com/iiab/iiab/issues/828#issuecomment-441089572

holta, 2018-11-30:
@tim-moody & @ericnitschke can I close this now, or are there remaining issues/questions?

holta, 2018-12-10:
@ericnitschke please re-open as nec!

closed 2018-12-10 by holta