iiab/iiab #1314: Refine Kolibri 0.11 to remove red/pink warnings during (re)install

holta, 2018-11-30:
@arky & all let's try to work together to remove this very final red/pink errors (warning) for Kolibri 0.11.0 in advance of IIAB 6.7's release in December, if possible!

> TASK [kolibri : Create kolibri default facility name, admin account and language] ***
fatal: [127.0.0.1]: FAILED! => {"changed": true, "cmd": "export KOLIBRI_HOME=\"/library/kolibri\" && \"/usr/local/kolibri/bin/kolibri\" manage provisiondevice --facility \"Kolibri-in-a-Box\" --superusername \"Admin\" --superuserpassword \"changeme\" --preset \"formal\" --language_id \"en\" --verbosity 0 --noinput", "delta": "0:00:01.557827", "end": "2018-10-28 16:21:18.735685", "msg": "non-zero return code", "rc": 1, "start": "2018-10-28 16:21:17.177858", "stderr": "\u001b[37mINFO     Running Kolibri with the following settings: kolibri.deployment.default.settings.base\u001b[0m\n\u001b[33mWARNING  Facility with name Kolibri-in-a-Box already exists, not modifying preset.\u001b[0m\nTraceback (most recent call last):\n  File \"/usr/local/kolibri/bin/kolibri\", line 11, in <module>\n    sys.exit(main())\n  File \"/usr/local/kolibri/local/lib/python2.7/site-packages/kolibri/utils/cli.py\", line 637, in main\n    manage(command, args=django_args)\n  File \"/usr/local/kolibri/local/lib/python2.7/site-packages/kolibri/utils/cli.py\", line 427, in manage\n    execute_from_command_line(argv=argv)\n  File \"/usr/local/kolibri/lib/python2.7/site-packages/kolibri/dist/django/core/management/__init__.py\", line 364, in execute_from_command_line\n    utility.execute()\n  File \"/usr/local/kolibri/lib/python2.7/site-packages/kolibri/dist/django/core/management/__init__.py\", line 356, in execute\n    self.fetch_command(subcommand).run_from_argv(self.argv)\n  File \"/usr/local/kolibri/lib/python2.7/site-packages/kolibri/dist/django/core/management/base.py\", line 283, in run_from_argv\n    self.execute(*args, **cmd_options)\n  File \"/usr/local/kolibri/lib/python2.7/site-packages/kolibri/dist/django/core/management/base.py\", line 330, in execute\n    output = self.handle(*args, **options)\n  File \"/usr/local/kolibri/local/lib/python2.7/site-packages/kolibri/core/device/management/commands/provisiondevice.py\", line 119, in handle\n    interactive=options['interactive'],\n  File \"/usr/local/kolibri/local/lib/python2.7/site-packages/kolibri/core/device/management/commands/provisiondevice.py\", line 72, in create_superuser\n    FacilityUser.objects.create_superuser(username, password)\n  File \"/usr/local/kolibri/local/lib/python2.7/site-packages/kolibri/auth/models.py\", line 496, in create_superuser\n    raise ValidationError(_('An account with that username already exists'))\ndjango.core.exceptions.ValidationError: [u'An account with that username already exists']", "stderr_lines": ["\u001b[37mINFO     Running Kolibri with the following settings: kolibri.deployment.default.settings.base\u001b[0m", "\u001b[33mWARNING  Facility with name Kolibri-in-a-Box already exists, not modifying preset.\u001b[0m", "Traceback (most recent call last):", "  File \"/usr/local/kolibri/bin/kolibri\", line 11, in <module>", "    sys.exit(main())", "  File \"/usr/local/kolibri/local/lib/python2.7/site-packages/kolibri/utils/cli.py\", line 637, in main", "    manage(command, args=django_args)", "  File \"/usr/local/kolibri/local/lib/python2.7/site-packages/kolibri/utils/cli.py\", line 427, in manage", "    execute_from_command_line(argv=argv)", "  File \"/usr/local/kolibri/lib/python2.7/site-packages/kolibri/dist/django/core/management/__init__.py\", line 364, in execute_from_command_line", "    utility.execute()", "  File \"/usr/local/kolibri/lib/python2.7/site-packages/kolibri/dist/django/core/management/__init__.py\", line 356, in execute", "    self.fetch_command(subcommand).run_from_argv(self.argv)", "  File \"/usr/local/kolibri/lib/python2.7/site-packages/kolibri/dist/django/core/management/base.py\", line 283, in run_from_argv", "    self.execute(*args, **cmd_options)", "  File \"/usr/local/kolibri/lib/python2.7/site-packages/kolibri/dist/django/core/management/base.py\", line 330, in execute", "    output = self.handle(*args, **options)", "  File \"/usr/local/kolibri/local/lib/python2.7/site-packages/kolibri/core/device/management/commands/provisiondevice.py\", line 119, in handle", "    interactive=options['interactive'],", "  File \"/usr/local/kolibri/local/lib/python2.7/site-packages/kolibri/core/device/management/commands/provisiondevice.py\", line 72, in create_superuser", "    FacilityUser.objects.create_superuser(username, password)", "  File \"/usr/local/kolibri/local/lib/python2.7/site-packages/kolibri/auth/models.py\", line 496, in create_superuser", "    raise ValidationError(_('An account with that username already exists'))", "django.core.exceptions.ValidationError: [u'An account with that username already exists']"], "stdout": "", "stdout_lines": []}
...ignoring

Ref: #1130

arky, 2019-01-25:
@holta I have patched latest Kolibri development branch to handle this ValidationError exception with a warning. These changes would be shipped with next release. 

Please close this bug as fixed whenever it is appropriate.

holta, 2019-01-25:
> @holta I have patched latest Kolibri development branch to handle this ValidationError exception with a warning.

Fantastic news.

Is a PR possible in January possibly?

arky, 2019-01-25:
@holta This is a upstream bug, there is no patch necessary here.

holta, 2019-01-25:
> @holta This is a upstream bug, there is no patch necessary here.

Got it.  Should we consider `failed_when: false` for IIAB 6.7, as was done with PR #1396 ?

(I'm fine either way!)

holta, 2019-06-17:
Is there any partial/substantial progress here 7 months later?

holta, 2019-11-15:
Done (certainly good enough!) as Kolibri playbook was cleaned up & refactored with Benjamin Bach in September/October.

closed 2019-11-15 by holta