iiab/iiab #1315: Etherpad on IIAB

ericnitschke, 2018-12-01:
Has this been attempted yet?

A few resources:
https://gitlab.constantvzw.org/aa/etherbox
https://networksofonesown.constantvzw.org/etherbox/manual.html
https://pad.riseup.net/
https://sudoroom.org/wiki/Mesh/SudoPad

holta, 2018-12-01:
Super interesting proposal.

If you or anyone is willing to give it a shot:!

Question: should individual URL's be of the form http://box/etherpad/<MYPAD> ?

tim-moody, 2018-12-01:
would install from https://github.com/ether/etherpad-lite?
we already have nodejs and mysql

ericnitschke, 2018-12-03:
https://github.com/ether/etherpad-lite/wiki/Sites-that-run-Etherpad-Lite

ericnitschke, 2018-12-29:
https://apps.nextcloud.com/apps/ownpad

mir123, 2019-03-21:
https://github.com/ether/etherpad-lite

holta, 2019-03-21:
> https://apps.nextcloud.com/apps/ownpad

@m-anish discussed this during today's IIAB call (http://minutes.iiab.io)

Has anybody tried it yet?  Some Etherpads can be more friendly (and more flaky) than others :)