iiab/iiab #1320: dnsmasq fails to start on IIAB 6.7/master on Ubuntu 18.04.1 PC

holta, 2018-12-04:
This is a physical machine (not VM) i.e. a NUC with internal WiFi, so dual interfaces.

@jvonau can you point to what needs investigating or ideally help me look at 10.8.0.38 directly?

dnsmasq fails on every reboot as follows: (`journalctl -u dnsmasq` shows the exact same info as below,)

> root@box:~# systemctl status dnsmasq
● dnsmasq.service - dnsmasq - A lightweight DHCP and caching DNS server
   Loaded: loaded (/lib/systemd/system/dnsmasq.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Tue 2018-12-04 14:18:31 EST; 3min 2s ago
  Process: 970 ExecStart=/etc/init.d/dnsmasq systemd-exec (code=exited, status=2)
  Process: 932 ExecStartPre=/usr/sbin/dnsmasq --test (code=exited, status=0/SUCCESS)
>
> Dec 04 14:18:30 box.lan systemd[1]: Starting dnsmasq - A lightweight DHCP and caching DNS server...
Dec 04 14:18:30 box.lan dnsmasq[932]: dnsmasq: syntax check OK.
Dec 04 14:18:31 box.lan dnsmasq[970]: dnsmasq: unknown interface br0
Dec 04 14:18:31 box.lan systemd[1]: dnsmasq.service: Control process exited, code=exited status=2
Dec 04 14:18:31 box.lan systemd[1]: dnsmasq.service: Failed with result 'exit-code'.
Dec 04 14:18:31 box.lan systemd[1]: Failed to start dnsmasq - A lightweight DHCP and caching DNS server.

Even despite br0 appearing here:

> root@box:~# ip a
1: lo: [...]
2: enp3s0: [...]
3: wlp2s0: [...]
4: br0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default qlen 1000
    link/ether fe:0c:e4:ba:24:83 brd ff:ff:ff:ff:ff:ff
    inet 172.18.96.1/19 brd 172.18.127.255 scope global br0
       valid_lft forever preferred_lft forever
5: tun0: [...]

FYI manually running "systemctl start dnsmasq" works a few minutes after boot, but a systematic (not manual) solution is what's needed here across the board.

ASIDE: `systemctl status captive-portal` also show "Active: failed (Result: exit-code)" but I assume that's unrelated.

holta, 2018-12-04:
@jvonau suspects "Timing issue on boot.. dnsmasq is being started before br0 is created"

(Hopefully he can uncover a way forward later tonight, looking into this machine / 10.8.0.38 directly.)

jvonau, 2018-12-05:
> iiab-admin@box:~$ sudo journalctl -b 0 | grep br0
Dec 04 14:18:27 box.lan systemd-networkd[695]: br0: netdev ready
Dec 04 14:18:27 box.lan systemd-networkd[695]: br0: IPv6 successfully disabled
Dec 04 14:18:28 box.lan cloud-init[673]: ci-info: |  br0   | False |     .     |     .     |   .   | fe:0c:e4:ba:24:83 |
Dec 04 14:18:31 box.lan dnsmasq[970]: dnsmasq: unknown interface br0
Dec 04 14:18:34 box.lan kernel: br0: port 1(wlp2s0) entered blocking state
Dec 04 14:18:34 box.lan kernel: br0: port 1(wlp2s0) entered disabled state
Dec 04 14:18:34 box.lan kernel: br0: port 1(wlp2s0) entered blocking state
Dec 04 14:18:34 box.lan kernel: br0: port 1(wlp2s0) entered forwarding state
Dec 04 14:18:34 box.lan kernel: br0: port 1(wlp2s0) entered disabled state
Dec 04 14:18:34 box.lan kernel: br0: port 1(wlp2s0) entered blocking state
Dec 04 14:18:34 box.lan kernel: br0: port 1(wlp2s0) entered forwarding state
Dec 04 14:18:34 box.lan avahi-daemon[821]: Joining mDNS multicast group on interface br0.IPv4 with address 172.18.96.1.
Dec 04 14:18:34 box.lan systemd-networkd[695]: br0: Gained carrier

It appears that br0 is configured by systemd-networkd but gets disabled by cloud-init causing dnsmasq to fail...  https://cloudinit.readthedocs.io/en/latest/topics/network-config.html

holta, 2018-12-05:
@jvonau given your many discoveries here this morning, what workaround(s) should we perhaps consider for Ubuntu's networking networking intricacies?  (which prevent br0 and dnsmasq from starting up cleanly on boot ;)

holta, 2018-12-09:
@jvonau if there's something I can do to push forward dnsmasq reliability, hit me today/tmrw before I jet west ?  e.g. on 192.168.0.124 = 10.8.0.30

At least "systemctl start dnsmasq" now works manually!  But sadly dnsmasq doesn't auto-start on bootup :-(

holta, 2019-01-02:
Fixed by PR #1336

closed 2019-01-02 by holta