iiab/iiab #1323: captive-portal fails with Android 8.1

holta, 2018-12-04:
There is no effect even when I connect an Android 8.1 phone to IIAB 6.7/master with a verified-running captive-portal service (except possibly many minutes later, when Android 8.1 started hounding me "Sign in to network" ?  Which certainly didn't do much anything at all...)

Caveat #1322 this new captive-portal service runs on 50% of my RPi 3's but not on 50% of the others.

Caveat #1321 sadly this new captive-portal service appears not to run at all across all Ubuntu IIAB's so far.

While i'd very much LIKE to test Android 8.1 against IIAB 6.7/master on Ubuntu, but so far the result there is even worse (Android 8.1 hangs forever with "Obtaining IP address..." &mdash; possibly related to dnsmasq failing to start on that machine #1320 ?)

holta, 2018-12-04:
@floydianslips wrote:

> I installed last night and captive portal worked. after clicking the two buttons. I think I had to restart my browser on Android

> Android 9

> Google Pixel XL

> just did it again

> it takes a bit of effort obviously, but it does work

holta, 2018-12-09:
1) “systemctl status captive-portal” is no longer relevant as PR #1327 (Captive Portal as part of Apache) is now a part of IIAB 6.7/master.

2) SO THE NEW WAY TO SEE IF captive-portal IS ENGAGED:
```
ls /etc/apache2/sites-enabled    # you would see 001-captive-portal.conf
```

holta, 2019-01-02:
Just To Reconfirm:

Captive Portal doesn't work (anyway does not show anything) on Android 8.1 &mdash; when testing an Android 8.1 device against IIAB 6.7/master on Ubuntu Server 18.04.1 or Raspbian.  Ref: PR #1336

holta, 2019-01-08:
@tim-moody reports:

> JH Not captive portal working on Android 9
On android 8 not working either

tim-moody, 2019-01-08:
JH: It says "captive.apple.com" on the android 8

holta, 2019-01-08:
[image]

WikiDocJames, 2019-01-08:
Worked really well on my iphone 5. Did not work on the android 4, 8, or 9.

holta, 2019-01-08:
> Worked really well on my iphone 5. Did not work on the android 4, 8, or 9.

@WikiDocJames we need details of the phone make/model and exact version of Android.  The reason is Android 9 can work fine on other phones (e.g. Google's Pixel XL phone works great) yet not on yours.  We have no way to move forward without details.

Thanks!

WikiDocJames, 2019-01-09:
So I am on a Pixel 2XL. And it is android version 9. I connect to IIAB wifi and nothing happens.

holta, 2019-01-09:
@georgejhunt had me run:
```
sed -i -e "s/.*elif system_version\.start.*/    elif system_version\[:1\] >= '7':/" /opt/iiab/captive-portal/capture-wsgi.py
```
Which instantly made a big difference, allowing a basic Captive Portal to work with Android 8.1, with the following notification appearing on top &mdash; immediately when Wi-Fi connects:
```
Sign in to Wi-Fi network
"Internet in a Box" [OR ACTUAL SSID]
```
If you click/touch the above (you may need to pull down from the top of your screen if you wait too long) you're then presented with an actual page showing http://box.lan content, with the following on top:
```
Sign in to Internet in a Box [OR ACTUAL SSID]
box.lan
```
If you then touch the 3 dots (vertical ellipsis) in the very top-right, Android 8.1 then gives options:
- Do not use this network
- Use this network as is

Presumably we need to train people to click/touch **"Use this network as is"** ?

I'll merge @georgejhunt's recommendation here (PR #1363) hoping this spurs others to test yet more different kinds of Android devices, running many different versions of Android.

holta, 2019-01-09:
I do not understand why @georgejhunt's patch (PR #1363) _works_ on Raspbian (e.g. 10.8.0.42, successfully enabling Captive Portal for Android 8.1) but _not at all_ on Ubuntu 18.04 (e.g. 10.8.0.50) :/

So I've asked him if he can help investigate.

holta, 2019-01-09:
If Android 7.x, 8.x and 9.x prove generally reliable in broader testing (ask your friends to please volunteer their phone for 5min!) then we should close this ticket to instead focus on:
- https://github.com/iiab/iiab/issues/1366 "Captive Portal fails when IIAB installed on Ubuntu?"
- https://github.com/iiab/iiab/issues/1351 "Captive Portal fails with Windows 8.1"
- #1182 "Captive Portal improvements checklist"

holta, 2019-01-09:
@georgejhunt & @jvonau: non-https sites like http://mit.edu unfortunately do load in the above testing &mdash; when they're definitely supposed to be blocked by `iiab_gateway_enabled: False` &mdash; can you help explain ?

Context: this is the same IIAB 6.7/master on Raspbian (10.8.0.42) tested with an Android 8.1 client.

This is a problem for impoverished schools/clinics that literally cannot afford to give away Internet.  (At least TLS/SSL sites are blocked, but let's try to understand!)

jvonau, 2019-01-09:
is squid enabled?

holta, 2019-01-09:
> is squid enabled?

Amazingly that fixed the problem.  I changed /etc/iiab/local_vars.yml to `squid_enabled: False` and then ran running ./iiab-network, confirming that Squid was turned off by running `systemctl status squid`

Then I tested Android 8.1 against the RPi 3 B+ which works "properly" now (correctly blocking access to http://mit.edu).

FYI Android (8.1 in this case) remains a royal pain...if you don't catch the "Connected, no Internet" notif the very 1st time you connect to IIAB's WiFi you're screwed...as you cannot later recover that notification to click/touch it, then hitting "Yes / Stay Connected"

Even asking Android to "forget" the WiFi connection is not always enough.  Insane, but I really did have to switch to another (live) WiFi network and then return to trying the original IIAB's WiFi/SSID in question.  Nuts but yeah &mdash; regrettably our FAQ instructions here will likely have to be expanded later... http://wiki.laptop.org/go/IIAB/FAQ#My_Android_device_says_.22Connected.2C_no_Internet.22_and_won.27t_browse_http:.2F.2Fbox

holta, 2019-01-09:
@georgejhunt & All:

Please feel free to close this tkt if you feel Android 7.x, 8.x & 9.x testing with Captive Portal is sufficient?

Regardless: let's definitely open a new ticket to ring out UX glitches + associated newbie educator training issues &mdash; so smartphones around the planet increasingly work with IIAB within seconds &mdash; _without_ (too many!) never-ending usability nightmares~

holta, 2019-01-10:
Please see associated bug report from @tim-moody:

#1376 Captive Portal w/ Android 7.x & 8.x should not require you to click "Connected, no Internet" / "Stay connected"

holta, 2019-02-23:
Is this now closeable as a result of https://github.com/iiab/iiab/issues/1504#issuecomment-466052240 ??

holta, 2020-01-04:
Seems to be solved now (sufficiently, as reasonably as we can expect for now) e.g. on Android 8.1, as written up at https://github.com/iiab/iiab/issues/1182#issuecomment-570745798

PR #1363 (2019-01-09): Captive Portal: patch capture-wsgi.py to test Android 7.x, 8.x, 9.x
Fixes #1323 ...we hope?!

Thanks to @georgejhunt

Seems to work for Android 8.1 but needs further testing to make sure this does not break Captive Portal where it worked in the past e.g. on @floydianslips' Google's Pixel XL phone running Android 9.
closed 2020-01-04 by holta