iiab/iiab #1330: Update Doc for New Dynamic Menuing

holta, 2018-12-10:
@tim-moody can you help update https://github.com/iiab/iiab/wiki/IIAB-Menuing now that the much-revised https://github.com/iiab/iiab-admin-console/pull/95 has been merged?

holta, 2018-12-10:
@tim-moody please also revise https://github.com/iiab/iiab/wiki/IIAB-Installation#add-content to mention new Dynamic Menuing arrangements if possible?

holta, 2019-01-15:
Brief summary of IIAB 6.7's drag+drop menuing & Admin Console advances:

https://github.com/iiab/iiab/wiki/IIAB-Menuing#2019-additions

A video will be needed as this evolves-

holta, 2019-01-15:
@tim-moody http://10.8.0.18/admin > Help > Help Sections: Content Menus button/link does not work &mdash; in current master.

Is this an oversight you might easily fix?

holta, 2019-01-15:
> @tim-moody http://10.8.0.18/admin > Help > Help Sections: Content Menus button/link does not work — in current master.

https://github.com/iiab/iiab-admin-console/pull/122 merges in a "Content Menus" help file.

tim-moody, 2019-01-15:
Worked for me because I copied manually. updated.

closed 2019-01-15 by tim-moody