iiab/iiab #1332: Check "How do I view usage statistics?" in FAQ for IIAB 6.7 release

holta, 2018-12-11:
@tim-moody, http://FAQ.IIAB.IO Item 26 looks possibly stale ("How do I view usage statistics?")

Can you confirm everything there is still accurate?

http://wiki.laptop.org/go/IIAB/FAQ#How_do_I_view_usage_statistics.3F

holta, 2018-12-14:
Refs: #1262 #1268

holta, 2019-02-07:
This [FAQ entry #29](http://wiki.laptop.org/go/IIAB/FAQ#How_do_I_view_usage_statistics.3F) urgently needs a cleanup, but let's punt to [IIAB 7.0](https://github.com/iiab/iiab/milestone/5) if we don't have time in the coming week.

holta, 2019-05-10:
@tim-moody continues to work hard on RTC (clock) ideas in support of this higher-level analytics gathering goal, perhaps based on the clocks of nearby teachers' Android phones etc.

(This is a very tough problem, that an attached GPS module and/or RTC batteries could also solve in theory, but there might be more pragmatic angle here surveying nearby Android/Windows devices.  In any case, this will likely have to wait for the IIAB 7.1 release cycle in Q3 2019 at the very earliest...)

jvonau, 2023-01-19:
What about the latest darling of the bunch in Matomo that only became functional after issue #3441 was filed despite being in the code base since #3304?