iiab/iiab #1366: Captive Portal fails when IIAB installed on Ubuntu?

holta, 2019-01-09:
Am opening a new tkt as #1323 appears to largely solve Android 8/9 issues?

Whereas "almost nothing works" (or so it would appear) when IIAB & its captive portal are installed on Ubuntu 18.04 or 16.04: https://github.com/iiab/iiab/issues/1323#issuecomment-452546824

holta, 2019-01-10:
@georgejhunt has not tested Captive Portal on Ubuntu 18.04 or any other OS's aside from Raspbian on RPi 3 (or 3 B+).

While important, this work may need to be delayed until IIAB 7.0.

For now I've documented this problem at the bottom of:
https://github.com/iiab/iiab/wiki/IIAB-6.7-Release-Notes#known-issues

holta, 2020-01-06:
@georgejhunt there are more than a few people that would love Captive Portal to work on Ubuntu, but I'm assuming this is not possible before IIAB 7.1 is released on/around 2020-01-30 ?

jvonau, 2020-04-29:
works on U20 with wifi firmware

closed 2020-04-29 by holta