iiab/iiab #1382: KA Lite fails when IIAB installed on Ubermix

holta, 2019-01-12:
/library/ka-lite/server.log shows no problems at first:
```
root@box:~# systemctl restart kalite-serve
root@box:~# cat /library/ka-lite/server.log
Going to daemon mode, logging to /library/ka-lite/server.log

To access KA Lite from another connected computer, try the following address(es):
        http://10.8.0.78:8008/

        http://172.18.96.1:8008/

        http://192.168.0.102:8008/

To access KA Lite from this machine, try the following address:
        http://127.0.0.1:8008/

[INFO] [2019-01-12 07:08:09,467] cherrypy.error: [12/Jan/2019:07:08:09] ENGINE Listening for SIGHUP.
[INFO] [2019-01-12 07:08:09,468] cherrypy.error: [12/Jan/2019:07:08:09] ENGINE Listening for SIGTERM.
[INFO] [2019-01-12 07:08:09,468] cherrypy.error: [12/Jan/2019:07:08:09] ENGINE Listening for SIGUSR1.
[INFO] [2019-01-12 07:08:09,469] cherrypy.error: [12/Jan/2019:07:08:09] ENGINE Bus STARTING
[INFO] [2019-01-12 07:08:09,469] cherrypy.error: [12/Jan/2019:07:08:09]  Loading and serving the Django application
[INFO] [2019-01-12 07:08:09,493] cherrypy.error: [12/Jan/2019:07:08:09] ENGINE Started monitor thread '_TimeoutMonitor'.
[INFO] [2019-01-12 07:08:09,635] cherrypy.error: [12/Jan/2019:07:08:09] ENGINE Serving on http://0.0.0.0:8008
[INFO] [2019-01-12 07:08:09,636] cherrypy.error: [12/Jan/2019:07:08:09] ENGINE Bus STARTED
```
But then /library/ka-lite/server.log shows the ERROR below `Could not parse version info string "3.6.2-CAPI-1.10.2 4d2925d6"` if anyone tries to visit http://box:8008  _(while the browser shows error message `This page isn’t working 10.8.0.78 is currently unable to handle this request. HTTP ERROR 500`)_
```
[ERROR] [2019-01-12 07:09:57,337] cherrypy.error: [12/Jan/2019:07:09:57] ENGINE GEOSException(u'Could not parse version info string "3.6.2-CAPI-1.10.2 4d2925d6"',)
Traceback (most recent call last):
  File "/usr/local/kalite/venv/local/lib/python2.7/site-packages/kalite/packages/dist/cherrypy/wsgiserver/wsgiserver2.py", line 1353, in communicate
    req.respond()
  File "/usr/local/kalite/venv/local/lib/python2.7/site-packages/kalite/packages/dist/cherrypy/wsgiserver/wsgiserver2.py", line 868, in respond
    self.server.gateway(self).respond()
  File "/usr/local/kalite/venv/local/lib/python2.7/site-packages/kalite/packages/dist/cherrypy/wsgiserver/wsgiserver2.py", line 2257, in respond
    response = self.req.server.wsgi_app(self.env, self.start_response)
  File "/usr/local/kalite/venv/local/lib/python2.7/site-packages/kalite/packages/dist/cherrypy/_cptree.py", line 299, in __call__
    return app(environ, start_response)
  File "/usr/local/kalite/venv/local/lib/python2.7/site-packages/kalite/packages/bundled/django/core/handlers/wsgi.py", line 255, in __call__
    response = self.get_response(request)
  File "/usr/local/kalite/venv/local/lib/python2.7/site-packages/kalite/packages/bundled/django/core/handlers/base.py", line 178, in get_response
    response = self.handle_uncaught_exception(request, resolver, sys.exc_info())
  File "/usr/local/kalite/venv/local/lib/python2.7/site-packages/kalite/packages/bundled/django/core/handlers/base.py", line 220, in handle_uncaught_exception
    if resolver.urlconf_module is None:
  File "/usr/local/kalite/venv/local/lib/python2.7/site-packages/kalite/packages/bundled/django/core/urlresolvers.py", line 342, in urlconf_module
    self._urlconf_module = import_module(self.urlconf_name)
  File "/usr/local/kalite/venv/local/lib/python2.7/site-packages/kalite/packages/bundled/django/utils/importlib.py", line 35, in import_module
    __import__(name)
  File "/usr/local/kalite/venv/local/lib/python2.7/site-packages/kalite/distributed/urls.py", line 14, in <module>
    from . import api_urls
  File "/usr/local/kalite/venv/local/lib/python2.7/site-packages/kalite/distributed/api_urls.py", line 12, in <module>
    import kalite.coachreports.api_urls
  File "/usr/local/kalite/venv/local/lib/python2.7/site-packages/kalite/coachreports/api_urls.py", line 3, in <module>
    from .api_resources import PlaylistProgressResource, PlaylistProgressDetailResource
  File "/usr/local/kalite/venv/local/lib/python2.7/site-packages/kalite/coachreports/api_resources.py", line 3, in <module>
    from tastypie.resources import Resource
  File "/usr/local/kalite/venv/local/lib/python2.7/site-packages/kalite/packages/dist/tastypie/resources.py", line 20, in <module>
    from django.contrib.gis.db.models.fields import GeometryField
  File "/usr/local/kalite/venv/local/lib/python2.7/site-packages/kalite/packages/bundled/django/contrib/gis/db/models/__init__.py", line 5, in <module>
    from django.contrib.gis.db.models.aggregates import *
  File "/usr/local/kalite/venv/local/lib/python2.7/site-packages/kalite/packages/bundled/django/contrib/gis/db/models/aggregates.py", line 2, in <module>
    from django.contrib.gis.db.models.sql import GeomField
  File "/usr/local/kalite/venv/local/lib/python2.7/site-packages/kalite/packages/bundled/django/contrib/gis/db/models/sql/__init__.py", line 2, in <module>
    from django.contrib.gis.db.models.sql.query import GeoQuery
  File "/usr/local/kalite/venv/local/lib/python2.7/site-packages/kalite/packages/bundled/django/contrib/gis/db/models/sql/query.py", line 4, in <module>
    from django.contrib.gis.db.models.fields import GeometryField
  File "/usr/local/kalite/venv/local/lib/python2.7/site-packages/kalite/packages/bundled/django/contrib/gis/db/models/fields.py", line 4, in <module>
    from django.contrib.gis import forms
  File "/usr/local/kalite/venv/local/lib/python2.7/site-packages/kalite/packages/bundled/django/contrib/gis/forms/__init__.py", line 2, in <module>
    from django.contrib.gis.forms.fields import GeometryField
  File "/usr/local/kalite/venv/local/lib/python2.7/site-packages/kalite/packages/bundled/django/contrib/gis/forms/fields.py", line 8, in <module>
    from django.contrib.gis.geos import GEOSException, GEOSGeometry
  File "/usr/local/kalite/venv/local/lib/python2.7/site-packages/kalite/packages/bundled/django/contrib/gis/geos/__init__.py", line 6, in <module>
    from django.contrib.gis.geos.geometry import GEOSGeometry, wkt_regex, hex_regex
  File "/usr/local/kalite/venv/local/lib/python2.7/site-packages/kalite/packages/bundled/django/contrib/gis/geos/geometry.py", line 16, in <module>
    from django.contrib.gis.geos.coordseq import GEOSCoordSeq
  File "/usr/local/kalite/venv/local/lib/python2.7/site-packages/kalite/packages/bundled/django/contrib/gis/geos/coordseq.py", line 9, in <module>
    from django.contrib.gis.geos.libgeos import CS_PTR
  File "/usr/local/kalite/venv/local/lib/python2.7/site-packages/kalite/packages/bundled/django/contrib/gis/geos/libgeos.py", line 131, in <module>
    _verinfo = geos_version_info()
  File "/usr/local/kalite/venv/local/lib/python2.7/site-packages/kalite/packages/bundled/django/contrib/gis/geos/libgeos.py", line 126, in geos_version_info
    raise GEOSException('Could not parse version info string "%s"' % ver)
GEOSException: Could not parse version info string "3.6.2-CAPI-1.10.2 4d2925d6"
```
Is it possible we need another line in /etc/tmpfiles.d/iiab.conf similar to what fixed {apache2, mongodb, munin} services in #1380 ?

holta, 2019-01-12:
I don't understand it, but FWIW the error `Could not parse version info string "3.6.2-CAPI-1.10.2 4d2925d6"` appears somewhat common for others in recent years:
https://stackoverflow.com/questions/18643998/geodjango-geosexception-error

benjaoming, 2019-01-12:
I think this issue is because of a bundled library `tasypie` that tries to import some GIS database stuff. Since it's not necessary, it could potentially be uninstalled on the host system. Try `sudo apt remove "libgeos-*"` - but observe the dependencies that will also be removed and decide whether you need them.

I can try to add a fix.

holta, 2019-01-12:
> I think this issue is because of a bundled library `tasypie` that tries to import some GIS database stuff. Since it's not necessary, it could potentially be uninstalled on the host system. Try `sudo [...]

That Worked !!  After I ran "systemctl restart kalite-serve" (which I'm not sure was even nec) http://box:8008 now works on Ubermix !

> I can try to add a fix.

Fantastic if you get the chance.

In any case, profound thanks @benjaoming for this genuine miracle.

holta, 2019-01-12:
Basic testing is complete for workaround PR #1385 but an upstream fix (https://github.com/learningequality/ka-lite/pull/5592) will be better yet in future!

closed 2019-01-12 by holta