iiab/iiab #1384: Admin Console doesn't load.

floydianslips, 2019-01-12:
### Expected behavior and actual behavior.

### Steps to reproduce the problem.

### Specifications like the version of the IIAB, operating system version, or hardware details.
Here is a link to the screen after login. 

https://pastebin.com/nXYiUWkx

It allows login but fails to init.

holta, 2019-01-12:
CLARIF: @floydianslips mentions this is occurring with IIAB 6.5 or 6.6

holta, 2019-01-12:
@tim-moody wonders if a hard power-off might have caused this problem, as @floydianslips  confirms is possible?  @tim-moody writes:

> the key is IIAB-CMDSRV failed to initialize

> the error is that cmdsrv is not running, but cmdsrv sets flags that it unsets when it shuts down, so possibly a false flag is about

> the standard debug technique is
cd /opt/admin/cmdsrv
systemctl stop iiab-cmdsrv
./iiab-cmdsrv
now errors are not trapped and you can see if something goes wrong in python

Earlier @tim-moody asked if updating to the lastest Admin Console might work?
```
cd /opt/iiab/iiab-admin-console
git pull
./install
```
@floydianslips  is nervous about updating from so many months ago.  But since this machine is many months old, I wonder if reinstalling the existing Admin Console from its existing code might help?
```
cd /opt/iiab/iiab-admin-console
./install
```

floydianslips, 2019-01-12:
Fixed by the master himself @tim-moody. Thanks Tim.

holta, 2019-01-12:
Mystery remains as to how this happened, but @tim-moody determined the flaw was:

> cmdsrv.0.2.db is missing

While it's possible ./install (i.e. a reinstall) from the existing IIAB 6.5 code would have fixed it... @tim-moody clarified:

>I patched it
just trying to stop cmdsrv as it is not accepting commands
front end works now
no idea how that happened
the main task are supposed to touch the file and set permissions
if that never ran then the file would be missing
if this used to work, then can't imagine how it got deleted
there was an upgrade script that may have tried to reset it

closed 2019-01-12 by floydianslips