iiab/iiab #1387: Debian 10: two outstanding issues, incl workarounds

holta, 2019-01-13:
Installing http://d.iiab.io/6.7 on Debian 10 pre-releases can work!  With these 2 known caveats:

- The [3 lines](https://github.com/iiab/iiab/blob/master/roles/httpd/templates/refresh-wiki-docs.sh#L36-L38) in /usr/bin/iiab-refresh-wiki-docs that use lynx to download from github.com need to be commented out for now (mysteriously GitHub is blocking all connections from lynx on Debian 10, even when lynx issues exactly the same headers as from Ubuntu/Raspbian or Debian 9!)  Possibly this is a subtle SSL/TLS problem, even when lynx claims to be using the same SSL as on the other distros?!
- ["Stop ejabberd service"](https://github.com/iiab/iiab/blob/master/roles/ejabberd/tasks/main.yml#L62-L68) froze during ejabberd's install (when installing BIG-sized IIAB) so `systemctl stop ejabberd` had to be run manually.  Installation then continued on its own: relaunching IIAB's install with `sudo iiab` was not even nec this time!

Ref: PR #1386

holta, 2019-01-14:
@benjaoming believes Kolibri _should_ install on Buster (Debian 10) so we need to dig deeper to try to find any error message(s) during the https://github.com/iiab/iiab/tree/master/roles/kolibri installation process!

holta, 2019-02-01:
_Many Thanks to @arky, @m-anish & @deldesir for your help here &mdash; as [IIAB 6.7](https://github.com/iiab/iiab/wiki/IIAB-6.7-Release-Notes) should finally be released next week we hope &mdash; and people will definitely try to install it on Debian pre-releases in coming months !!_

holta, 2019-02-02:
FYI Debian 10 / Buster's Alpha 5 is now being released &mdash; the announcement will appear here in coming minutes:
https://lists.debian.org/debian-devel-announce/2019/02/threads.html
https://www.debian.org/devel/debian-installer/

FWIW its installer includes initial support for Secure Boot and UEFI:
https://bits.debian.org/2019/02/testing-initial-secure-boot-support.html

holta, 2019-02-06:
fyi @deldesir is hoping to try https://cdimage.debian.org/cdimage/daily-builds/daily/arch-latest/amd64/iso-cd/debian-testing-amd64-netinst.iso if he can get Debian 10 pre-release running in Haiti.

deldesir, 2019-02-08:
Job for dnsmasq.service failed because the control process exited with error code.

holta, 2019-02-08:
> Job for dnsmasq.service failed because the control process exited with error code.
> What might cause this on stage 4 (4-server-options) of the install process?

Great Question: @jvonau what log/pastebin is most likely useful to reveal what's happening here?

holta, 2019-02-08:
@deldesir does "iiab-install --reinstall" (in /opt/iiab/iiab) fail every time?

 Try to get more details, e.g. from "journal to -u dnsmasq" or any relevant log file?

deldesir, 2019-02-09:
The same error even after full reinstall.

holta, 2019-02-11:
@deldesir, is your IIAB-on-Debian-10 install doing better now, with dnsmasq now working per #1469 ?

holta, 2019-02-12:
fyi Debian 10 is now in "soft freeze" and will enter "full freeze" a month from today on March 12th:
https://lists.debian.org/debian-stable-announce/2019/debian-stable-announce-201902/threads.html

holta, 2019-03-12:
@deldesir & @arky: Debian 10 "Buster" entered Final Freeze earlier today!

It would awesome if you could help review IIAB 7.0/master on there in coming weeks if/when you have a moment?

_(In particular with Debian 10 having enabled AppArmor by default, a couple small security tweaks might turn out to be necessary now...)_

holta, 2019-03-19:
Really Great News that IIAB 7.0/master is now installable on Debian 10 Buster's latest pre-releases, as @floydianslips confirmed overnight:
- lynx no longer fails to scrape GitHub docs (for http://box/info)
- ejabberd no longer blocks IIAB installs, in general
- Kolibri 0.12.1 now works (http://box/kolibri)

Remaining Issue, applicable to an increasing number of OS's / distros:
- MongoDB does not exist: #1437

holta, 2019-05-05:
Debian 10 "Buster" is arriving very shortly (in coming weeks?) so https://itsfoss.com/debian-10-buster/ summarizes some changes we should watch for:
- Kernel 4.19.0-4
- AppArmor Enabled by Default
- Nodejs 10.15.2
- NFtables replaces iptables for "better and easier syntax, better support for dual-stack ipv4-v6 firewalls"
- Any existing Postgresql databases used will need to be reindexed "due to updates in glibc locale data"
- Implements /usr merge (https://www.freedesktop.org/wiki/Software/systemd/TheCaseForTheUsrMerge/) catching up with Fedora 17 and other OS's/distros, creating these 4 symlinks:
  /bin → /usr/bin
  /sbin → /usr/sbin
  /lib → /usr/lib
  /lib64 → /usr/lib64
- Secure-boot support
- ...

holta, 2019-06-03:
Refs:

PR #1697 Not just is_ubuntu_18: make U19/U20 viable [Debian 10 already OK?] [extends use of php{{ php_version }}-sqlite3 to future OS's for now...somewhat controversially]
#1717 openvpn@xscenet fails to use /etc/iiab/openvpn_handle with OpenVPN 2.4.6 ?

holta, 2019-06-12:
FYI Debian 10 "Buster" is expected to be release on "Saturday July 6th":
https://www.phoronix.com/scan.php?page=news_item&px=Debian-10.0-Buster-Release-Date

Its daily ISO's are available here, for testing with http://d.iiab.io &mdash; i.e. IIAB 7.0/master pre-releases:
https://cdimage.debian.org/mirror/cdimage/daily-builds/daily/current/amd64/iso-cd/

holta, 2019-06-15:
Installing BIG-sized http://d.iiab.io worked onto 2019-06-15's Debian 10 Buster pre-release, with the final OS expected 2019-07-06 &mdash; except for Sugarizer which has to be change to False/False in local_vars.yml

I consider this a success, and will close this tkt now.

The remaining issue of #1437 (MongoDB not being included on many/most Linuxes going forward) will have to solved in a larger way...so that Sugarizer can hopefully continue to run and many/most Linuxes incoming years.

closed 2019-06-15 by holta