iiab/iiab #1394: IIAB fails to correctly autodetect VM networking setup

m-anish, 2019-01-15:
### Expected behavior and actual behavior.

I am setting up IIAB as per the specs below. While the install proceeds smoothly, the network mode chosen is gateway, and the dhcp server sits on the public bridged interface. Ideally, it should have auto-detected the dhcp server on that interface and set it self up as an appliance. Re-running iiab-network also does not fix the problem.

### Steps to reproduce the problem.

### Specifications like the version of the IIAB, operating system version, or hardware details.
I am testing the IIAB one line install with this as the latest commit
b2be3ccae19bb52be80b3f399989fcafd5edca49

Setup:
Vagrant, Ubuntu 18.04 (Bionic) VM, IIAB One line install with mininal vars file
Network setup contains two interfaces
1. default NAT
2. public bridged interface

holta, 2019-01-15:
Thanks @jvonau who's currently helping over live chat.  In reply, @m-anish has posted:
```
root@box:/opt/iiab/iiab/vars# ip r
default via 10.0.2.2 dev enp0s3 proto dhcp src 10.0.2.15 metric 100 
10.0.2.0/24 dev enp0s3 proto kernel scope link src 10.0.2.15 
10.0.2.2 dev enp0s3 proto dhcp scope link src 10.0.2.15 metric 100 
172.18.96.0/19 dev br0 proto kernel scope link src 172.18.96.1
```
And for `ip a` :
http://paste.ubuntu.com/p/qz5GrvgbK4/

And for `brctl show` :
http://paste.ubuntu.com/p/82xczbbvqw/

holta, 2019-01-15:
Chat except:

@jvonau: I'd just remove the internal Nat from the vagrant file..  or pre-populate /etc/iiab/local_vars.yml with the change mentioned above..

@m-anish: sure. i will do that.. 
but do you think this needs to be fixed sometime?

@jvonau: Or bug ansible's devs.. 
Nothing that I'm going to automate, the overrides are there to provide the functionality wanted.

@m-anish: okay.fair enough
but this is something that can break somebodys network

holta, 2019-01-15:
@m-anish: do you think i could do something like this....

https://www.vagrantup.com/docs/networking/public_network.html#default-router

the "default router" section

@jvonau: What does your [vagrant] file look like?

@m-anish: https://termbin.com/6bn9

basically one extra line... config.vm.network "public_network"

the internet seems to suggest that removing the default nat interface is a) not possible or b) not a good idea

holta, 2019-01-15:
@jvonau clarifies:

In the [ /opt/iiab/iiab/iiab-install.log ] is the output of ansible setup a dump of the detected variables.. [0-init](https://github.com/iiab/iiab/tree/master/roles/0-init) is where the gateway gets detected check there for the variable used from ansible

m-anish, 2019-02-05:
@jvonau says

...make a suggestion (to ansible) that under the individual network device listing that the gateway be mentioned alongside IP address and netmask and Network

m-anish, 2019-02-05:
So, as I understand in a wider context (outside VMs as well) we do not support IIAB to work in appliance mode if it happens to be part of two independent networks, where it needs to be in appliance mode to both those networks.

m-anish, 2019-02-05:
Setting iiab_lan_enabled to False does not seem to work. 
[iiab-install.log](https://github.com/iiab/iiab/files/2833209/iiab-install.log)

holta, 2019-02-05:
> @jvonau Setting iiab_lan_enabled to False does not seem to work.
> [iiab-install.log](https://github.com/iiab/iiab/files/2833209/iiab-install.log)

Easier to read instantly in browser here: http://paste.ubuntu.com/p/C9Z4M9h8nP/

holta, 2019-02-05:
@jvonay writes:

> Anish think I see the issue with why that didn't work, looks like there may be some '' that are not needed could you change the variable to `user_lan_iface: none` and then iiab-network as a test please..

holta, 2019-02-07:
@jvonau summary from our community/team call (http://minutes.iiab.io) today:

> “when” clauses in computed_network.yml need adjusting, “iiab-lan-enabled:” impact with Admin Console?  Did syntax change in underlying Ansible?  ipv6 popping up on some interfaces, any idea why?

@m-anish suggests:

> This is a bug, but prob not a blocker for [IIAB 6.7](https://github.com/iiab/iiab/milestone/4) release...

holta, 2019-05-04:
@m-anish can you offer suggestions towards bringing this issue to a close — prior to IIAB 7.0's release ideally?

holta, 2019-07-29:
@m-anish plz confirm if this is solved by PR #1906 ?

holta, 2019-11-01:
@m-anish this is 10 months old and appears solved.  But if any ongoing issues remain, plz re-open and explain, Thanks!

closed 2019-11-01 by holta