iiab/iiab #1395: Admin Console Drap+Drop fails on Firefox (works in Chrome browser)

holta, 2019-01-15:
@tim-moody http://10.8.0.18/admin > Content Menus > Content Item List does not allow drag+drop from right column ("Available Menu Item Definitions") to left column ("Menu Items in Current Menu") in Firefox 64.0.2

Even after clicking "Load Menu" (necessary, I finally figured that out!)

PS such right-to-left drag+drop _does_ work with Chrome 71.0.3578.98 &mdash; if this is not fixable in Firefox, can you make a very prominent warning appear outlining which browsers are most/least likely to work?  (As of now, no notice is given, causing confusion.)

tim-moody, 2019-01-15:
sounds like https://bugzilla.mozilla.org/show_bug.cgi?id=339293

holta, 2019-01-15:
@tim-moody clarifies:

> iiab.me/test/js-menu/ddrop.htm also doesn't work in FF

holta, 2019-01-15:
Likely fixed by: https://github.com/iiab/iiab-admin-console/pull/123

closed 2019-01-15 by holta