iiab/iiab #1410: Make en-oob.json far more approachable & inviting (incl video screencast if poss!)

holta, 2019-01-17:
I can helping with the writing making [en-oob.json](https://github.com/iiab/iiab-admin-console/blob/master/roles/js-menu/files/menu-files/menu-defs/en-oob.json) more enticing, while @georgejhunt is working on a short screencast to show off new Dynamic Menuing so implementers visually+rapidly understand their options.

@kananigit suggests using VokoScreen on Ubuntu?  Other screencasting packages are also possible: https://itsfoss.com/best-linux-screen-recorders/

Refs: #1031 "sampler.html page (used by http://box/home/index.html OOB) needs an overhaul [actually removal!]"

holta, 2019-01-24:
I will add a link to @georgejhunt's video in coming days:

> From: George Hunt
> Date: Sat, Jan 19, 2019 at 3:19 PM
> Subject: First iiab video -- 12 minutes, 30MB
>
> I played around some trying to find a free open source tool set to use
to generate screen casts. But in the end, it seemed better to use what
was already available to me on my Mac.
>
> I used QuickTime player to record, iMovie to edit, and VLC to shrink
from 200MB to 30MB.
>
> The results are at https://youtu.be/aYdF4STl97s

holta, 2019-01-24:
@georgejhunt might also:

- _(short term)_ publish an MP4 and WebM version of the above video, for those wanting to play it offline, in support of almost all browsers/modern OS's.
- _(medium term)_ fix a couple factual errors in the vid, that @tim-moody pointed out
- _(long term)_ re-record the 13min video as a series of much shorter 2-to-4min videos on different introductory topics.

georgejhunt, 2019-01-25:
I edited the 13 minute source into smaller chunks, and created webm format
(which as Tim said, just plays in a browser with no player).

The new stuff is at http://d.iiab.io/content/video.

holta, 2019-01-26:
> @georgejhunt can you please publish these 3 to YouTube ?

1) This is important so we don't abuse donated bandwidth.
2) Secondarily, was .mp4 avoided at http://d.iiab.io/content/videos/ for a reason?  (I was talking to volunteer who publishes community videos yesterday, and .mp4 is by far the most popular video format as of today anyway!)

holta, 2019-01-31:
@m-anish willing to help with short IIAB DIY videos, and recommends:
- https://creatoracademy.youtube.com/page/browse tips to enhance your YouTube channel
- https://obsproject.com software to stream directly to YouTube (overlaying diff screens capture vs. logo vs. scene of room)

holta, 2019-01-31:
@georgejhunt thanks for publishing the three shorter/topical (nicely edited) .mp4 videos here!
http://d.iiab.io/content/videos/

Can you publish these to YouTube in the coming 24h if poss?

georgejhunt, 2019-01-31:
Done.
https://youtu.be/mWEDMY5a-WY
https://youtu.be/P-NYNTe17gg
https://youtu.be/mmwyLVlEqrg

holta, 2019-02-05:
> https://youtu.be/mmwyLVlEqrg

Small typo here: opening title of video says "IIAB 2.7" instead of 6.7 :)

holta, 2019-02-05:
@georgejhunt a small [TOC](http://d.iiab.io/content/videos/HEADER.html) was added at the top of http://d.iiab.io/content/videos

LMK if this needs revising?

holta, 2019-02-08:
Tangentially related Admin Console / Menuing tickets...as we prep for IIAB 6.7 RC3 on Feb 11:

15 issues:
https://github.com/iiab/iiab-admin-console/issues

2 PR's:
https://github.com/iiab/iiab-admin-console/pulls

holta, 2019-02-11:
Central Challenge—we need to articulate to the "public" (novice implementers) how we can all improve [menu def] descriptions & logos together:

https://github.com/iiab/iiab/issues/1002#issuecomment-462183401

holta, 2019-02-20:
Crucially, http://FAQ.IIAB.IO Number 14 (http://wiki.laptop.org/go/IIAB/FAQ#How_do_I_customize_my_server.27s_home_page.3F) needs to be rewritten.

I'll start work on that today, so that IIAB 6.7 can be substantially more understandable to "activist teachers" and the global public eagerly wanting to give this a try, customizing their own Internet-in-a-Box to the needs of their own community.

Future: editing of Content Pack (menuItemDef) descriptions & logos might be possible from the browser itself as part of IIAB 7.0, per @tim-moody's ongoing R&D: https://github.com/iiab/iiab/issues/1002#issuecomment-462183401

closed 2019-02-21 by holta