iiab/iiab #1413: Test Calibre 3.37 on Ubuntu 18.04 & Raspbian

holta, 2019-01-18:
.debs for Calibre 3.38 will be available for Raspbian here, within ~10 days presumably:

http://raspbian.raspberrypi.org/raspbian/pool/main/c/calibre/
http://archive.raspbian.org/raspbian/pool/main/c/calibre/

See Calibre 3.37 released 2019-01-18:

https://calibre-ebook.com/whats-new

Ref: #1340 for Calibre 3.37

m-anish, 2019-01-26:
Okay, so there is a problem!

I installed calibre, and when I access through a browser on `http://<iiab-ip>:8080`, I am asked for a username/password, to which I reply Admin/changeme and I am able to login properly. 

But when I access the same at `http://<iiab-ip>/calibre`, and enter the same credentials, I get an error page - screenshot below

[image]

FWIW, calibre-web works just fine on `http://<iiab-ip>/books` but doesn't respond on `http://<iiab-ip>:8083` but that might just be an iptables thing because from the configuration, I can see it is programmed to allow traffic on 8080 but not 8083

holta, 2019-01-26:
@m-anish http://box/calibre was never fully implemented.

http://box:8080 remains the default URL for Calibre since 2017 &mdash; as documented in http://FAQ.IIAB.IO Item 3 here:
http://wiki.laptop.org/go/IIAB/FAQ#What_can_I_do_with_E-books_and_Internet-in-a-Box.3F

Also documented here:
https://github.com/iiab/iiab/wiki/IIAB-Networking#list-of-open-ports--services

_(But if someone can implement http://box/calibre that would be truly great!)_

m-anish, 2019-01-26:
[I commented there.](https://github.com/iiab/iiab/issues/1050#issuecomment-457840154)

holta, 2019-01-26:
Raspbian testing remains!  When Calibre 3.38.1 .debs are ready here:

http://raspbian.raspberrypi.org/raspbian/pool/main/c/calibre/
http://archive.raspbian.org/raspbian/pool/main/c/calibre/

holta, 2019-02-01:
No longer nec as a result Calibre 3.39.1 (#1444)

closed 2019-02-01 by holta