iiab/iiab #1418: Captive portal test matrix

m-anish, 2019-01-20:
|             | Gateway</br>popup | Gateway</br>url in browser | Lan Controller<br>popup |
| --------- | ------- |------- |-------- |
| Android 6| <ul><li>[ ] </li></ul> | <ul><li>[ ] </li></ul> | <ul><li>[ ] </li></ul> |
| Android 7| <ul><li>[ ] </li></ul> | <ul><li>[ ] </li></ul> | <ul><li>[ ] </li></ul> |
| Android 8| <ul><li>[ ] </li></ul> | <ul><li>[ ] </li></ul> | <ul><li>[ ] </li></ul> |
| Win 8.1   | <ul><li>[ ] </li></ul> | <ul><li>[ ] </li></ul> | <ul><li>[ ] </li></ul> |
| Win 10    | <ul><li>[ ] </li></ul> | <ul><li>[ ] </li></ul> | <ul><li>[ ] </li></ul> |
| iOS         | <ul><li>[ ] </li></ul> | <ul><li>[ ] </li></ul> | <ul><li>[ ] </li></ul> |
| Mac        | <ul><li>[ ] </li></ul> | <ul><li>[ ] </li></ul> | <ul><li>[ ] </li></ul> |
| add here | <ul><li>[ ] </li></ul> | <ul><li>[ ] </li></ul> | <ul><li>[ ] </li></ul> |

holta, 2019-01-20:
A Y/N answer might not be helpful.

e.g. when Android 8.1 on Moto G5 Works but others (who need to provide details of their phone model & exact Android version & screenshots) have not had any luck with Android 8.

m-anish, 2019-01-20:
> A Y/N answer might not be helpful.
> 
> e.g. when Android 8.1 on Moto G5 Works but others (who need to provide details of their phone model & exact Android version & screenshots) have not had any luck with Android 8.

Feel free to edit any way you like, either adding more columns, or replacing the tick with text ... i'd prefer more columns.. you may even remove the generic android 8, 7, 6 etc. I just added those as placeholders.