iiab/iiab #1425: Consider pandoc 2.5+ for IIAB 7.0 (scarpes/updates http://box/info offline documentation)

holta, 2019-01-25:
My preference is that we upgrade pandoc from 1.16 (part of Ubuntu 18.04) and 1.17 (part of the latest Raspbian) to 2.5+ for cleaner / more usable HTML in our offline docs: https://github.com/jgm/pandoc/releases

Unless @m-anish finds something much more stable, reliable and hassle-free.  He mentions static site generators Jekyll (https://jekyllrb.com/docs/) and Hugo (https://gohugo.io)

Myself I'd strongly prefer sites we do NOT have to be maintained, e.g. include guaranteed hassle-free backups & username/password administration we do NOT have to touch &mdash; so our community can scale, and so we do not lose more work as did in recent years when both our WordPress and Discourse installations were entirely lost.

Right now http://wiki.laptop.org and http://github.com/iiab/iiab/wiki each work very well -- but they each have their shortcomings, that we might move away from long-term possibly?  One option might be https://readthedocs.org fits into IIAB's " constantly offlining/scraping new docs for http://box/info " workflow somehow?

Refs: #803

m-anish, 2019-01-26:
Looks interesting. 

What should be the next steps on this?

holta, 2019-03-04:
@m-anish wrote:
> Looks interesting.
> 
> What should be the next steps on this?

Pressure Debian 10 "Buster" to get with the program :-)

Possibly @arky knows Debian developers who can take Pandoc 2.7 ([released yesterday](https://github.com/jgm/pandoc/releases)) seriously, rather than Pandoc 2.2.1 @ https://packages.debian.org/buster/pandoc from almost a year ago.

holta, 2019-05-04:
Let's defer to IIAB 7.1, when Debian 10 and hence "Raspbian 10" may solve this for us!?

holta, 2019-05-05:
It's unfortunate that Ubuntu 19.04 just shipped with a year-old version of pandoc (2.2.1 from 2018-05-10):
https://launchpad.net/ubuntu/disco/+source/pandoc

Likewise for Debian 10 Buster, whose final release is expected in coming weeks, which will likely be used as the basis for Raspbian on Raspberry Pi:
https://packages.debian.org/search?keywords=pandoc

Whereas development of [pandoc](https://pandoc.org/) has in fact been very steady over the past year!
https://pandoc.org/releases.html
https://github.com/jgm/pandoc/releases

holta, 2019-06-12:
fyi pandoc 2.7.3 is now released, containing 258 commits since March 3rd's pandoc 2.7 &mdash; changelog is here:
https://github.com/jgm/pandoc/releases/tag/2.7.3

jvonau, 2020-04-29:
stale?

holta, 2020-04-29:
It's unfortunate Ubuntu 20.04 LTS did not go with a much more recent pandoc, but at least they have:

```
root@box:~# apt -a list pandoc
Listing... Done
pandoc/focal,now 2.5-3build2 arm64 [installed]
```

...as I ran on RPi 4.

holta, 2020-04-29:
Pretty sad that Ubuntu 20.04 LTS went with pandoc 2.5 (https://github.com/jgm/pandoc/releases/tag/2.5) from 2018-11-27.

When pandoc 2.9.x has been available since 2019-12-12 !