iiab/iiab #1436: deb 9 and 10 have bind9 and isc-dhcp-server

tim-moody, 2019-01-30:
dns_service: bind9
dhcp_service: isc-dhcp-server
dns_user: bind

there may be others.

we should either control here (my preference) or with named_install in local vars, but not both

holta, 2019-01-30:
@m-anish, @arky & @jvonau can you take a look?  @tim-moody seems to be suggesting we overhaul the 3+3 lines below to use dnsmasq instead?

https://github.com/iiab/iiab/blob/master/vars/debian-9.yml#L5-L7
https://github.com/iiab/iiab/blob/master/vars/debian-10.yml#L5-L7

Does this apply to Raspbian, Ubuntu 18.04 and others distros too?

https://github.com/iiab/iiab/blob/master/vars/raspbian-9.yml#L8-L10
https://github.com/iiab/iiab/blob/master/vars/ubuntu-18.yml#L5-L7
https://github.com/iiab/iiab/blob/master/vars/centos-7.yml#L5-L6

jvonau, 2019-01-30:
Just provides a mapping of the service to deb package names just like apache, squid, mysql.

tim-moody, 2019-01-30:
so the deb package name for dnsmasq is bind9?

holta, 2019-01-30:
Should these variable names be improved (or a line just above clarifying their purpose?) so newcomers understand what these variables within each [/opt/iiab/iiab/vars/[OS].yml](https://github.com/iiab/iiab/tree/master/vars) really mean & what they are for &mdash; without us accidentally creating confusion?

tim-moody, 2019-01-30:
let's just add a comment above that these names only apply if named_install and/or ?dhcpd_install is true

and not to dnsmasq

holta, 2019-01-30:
> let's just add a comment above that these names only apply if named_install and/or ?dhcpd_install is true
> 
> and not to dnsmasq

@jvonau is the above accurate/complete?

PR #1441 (2019-02-01): Make vars/<OS>.yml files less confusing (with dnsmasq, we can generally ignore "dns_service: named" & "dns_service: bind9")
Fixes #1436 sufficiently for now?
closed 2019-02-01 by holta