iiab/iiab #1443: [Docs][FAQ]  Can you create WiFi hotspot using old netbooks/laptop?

arky, 2019-02-01:
Creating an outline of a common question from the community. 

**[FAQ]  Can you create WiFi hotspot using old netbooks/laptop?**
Yes, it is possible to create a WiFi hotspot (using hostapd) on some netbooks and laptops with Qualcomm Atheros chipsets.These older Atheros chipsets have good support in gnu/Linux  (Refer https://wiki.debian.org/ath9k). 

Test shows that IIAB 6.7 hostapd WiFi hotspot works with Qualcomm Atheros AR9285 PCI-Express Wireless device found in Acer Aspire One.

holta, 2019-02-01:
Added as http://FAQ.IIAB.IO Item 16, Thanks @arky!

@jvonau, @georgejhunt, @tim-moody @m-anish can you add to this?
http://wiki.laptop.org/go/IIAB/FAQ#Can_I_create_a_Wi-Fi_hotspot_using_an_old_laptop.3F

holta, 2019-02-06:
Please re-open this tkt if nec, to expand this very useful survey of laptops which work creating internal hotspots for IIAB 7.0!?

closed 2019-02-06 by holta