iiab/iiab #1449: Upgrade/Validate sugarizer-server v1.1.0 on release "March 2019"

holta, 2019-02-04:
Building on PR #1430 here:
https://github.com/iiab/iiab/blob/master/roles/sugarizer/defaults/main.yml#L15-L18

When @llaske's v1.1.0's back-end is released here:
https://github.com/llaske/sugarizer-server/releases

holta, 2019-05-02:
Many thanks to @llaske who just clarified:

> 1) The dev branch of Sugarizer Server is very stable (currently in production at our Saint-Ouen deployment) so I'm packaging the release. Will be labeled version 1.1 and could be expected in 2/3 weeks so, yes before the end of the month.
>
> 2) The next release of Sugarizer will be labeled version 1.2. No release date yet. Probably in July.

Ref: #1429

holta, 2019-05-12:
Sugarizer 1.1 is now released, and appears to be a very significant upgrade, and hopefully a great addition to IIAB 7.0 very shortly:

- https://groups.google.com/forum/#!topic/unleashkids/PaFU8EsxUIM
- https://github.com/llaske/sugarizer-server/releases/tag/v1.1.0
- https://github.com/llaske/sugarizer-server/blob/master/CHANGELOG.md
- https://github.com/llaske/sugarizer-server/blob/master/README.md

_Thank You @llaske!_

Question: Lionel do you know of any strategies emerging to run Sugarizer & Sugarizer Server natively on the imminent Debian 10 Buster and hence Raspbian 10 Buster expected soon after?  (Similar to CentOS 8, both major OS final releases expected in not so many weeks &mdash; neither of which will include MondoDB in their repos we're told &mdash; posing some tough questions that we will need help navigating together!)

llaske, 2019-05-17:
Look like it's possible to install MongoDB on Debian Buster from source code.
See https://stackoverflow.com/questions/47638143/install-mongodb-on-debian-buster/50032030#50032030

closed 2019-12-21 by holta