iiab/iiab #1450: [Docs][FAQ] Assigning a static IP Address

arky, 2019-02-05:
Creating a FAQ of another common question from the community. I am not quite sure how the 'interface' entry is generated, I guess ansible selects the currently active interface 'wlan0' or 'eth0'?

**[Docs][FAQ] Assigning a static IP Address**
To set a static IP address add the following entries to /etc/iiab/local_vars.yml and run the `iiab-network' command.
```
wan_ip: 192.168.1.10
wan_netmask: 255.255.255.0
wan_gateway: 192.168.1.1
wan_nameserver: 192.168.1.1

```

This will generate following entries in 
/etc/dhcpcd.conf
```
interface eth0
static ip_address=192.168.1.10/24
static routers=192.168.1.1
domain_name_servers=192.168.1.1
```

holta, 2019-02-05:
@jvonau appears concerned the above may apply to RPi only?

tim-moody, 2019-02-06:
For me going into the Admin Console, checking static ip, entering the address, save config and run install has worked in the past on a nuc with ubuntu (without touching any of the above variables)

m-anish, 2019-02-06:
Okay I tested this on a VM with a single public bridged network, on the WAN side to the IIAB. Base OS - Ubuntu 18.04

Setting those values caused the wan interface not to get any IP address.

Also, I dont see any `/etc/dhcpd.conf` file

What debug info will help? This was a vanilla minimal sized install.

holta, 2019-02-06:
@m-anish recommends we add @arky's suggestion to http://FAQ.IIAB.IO now that PR #1455 has been merged.

holta, 2019-02-06:
http://FAQ.IIAB.IO Item 19 is now published:

http://wiki.laptop.org/go/IIAB/FAQ#How_do_I_assign_a_static_IP_Address.3F

_Thanks all (and feel free to improve on it there!)_

closed 2019-02-06 by holta

holta, 2019-02-06:
> http://wiki.laptop.org/go/IIAB/FAQ#How_do_I_assign_a_static_IP_Address.3F

Tightened up / moved to:
http://wiki.laptop.org/go/IIAB/FAQ#How_do_I_set_a_static_IP_Address.3F