iiab/iiab #1465: Include Minetest client software for Windows/Android for offline communities?

holta, 2019-02-08:
IIAB could possibly offer http://box/software as a synonym for the pre-existing http://box/local_content == http://box/usb

Reason: what client software to offer each community tends to be highly-dependent on local needs / local language / curriculum / IT particulars / etc...as outlined @ https://github.com/iiab/iiab/pull/1454#issuecomment-461515928

Refs: PR #1454, PR #1461, #1462, #1464

holta, 2019-02-08:
These 3 (from https://www.minetest.net/downloads/) are probably enough for starters:
- https://github.com/minetest/minetest/releases/download/0.4.17.1/minetest-0.4.17.1-win64.zip 14.0 MB
- https://github.com/minetest/minetest/releases/download/0.4.17.1/minetest-0.4.17.1-win32.zip 13.7 MB
- https://github.com/minetest/minetest/releases/download/0.4.17.1/Minetest-0.4.17.1-2.apk 11.3 MB

(Less than 40MB total.)

tim-moody, 2025-04-11:
Side loading android is problematic, but the windows and/or Linux client might be worthwhile. Taking it a step at a time makes sense.