iiab/iiab #1467: Get Asterisk/FreePBX working on RPi too!

holta, 2019-02-09:
Ideally Asterisk/FreePBX should world on $35 Raspberry Pi's not just pricier NUC's.

If so, this could build directly on @m-anish's PR #1405 ?
And/or use http://www.raspberry-asterisk.org/ alongside?

(Debian 9.7 and Debian 10 support might also be considered in future...)

m-anish, 2019-02-10:
Yep. This would be great to have! Would like to point out that while they will push the available resources on a Pi, it _should_ be possible to run them atleast for small installations without problems.

holta, 2019-02-11:
@m-anish got PR #1478 (not the final word, as "./runrole pbx" fails on repeat runs) to compile/install on Raspbian Lite on RPi 3 B+

Amazing!

(Let's declare "experimental support" here and recommend everyone who can test this...)

holta, 2019-02-11:
_@m-anish what further testing should we encourage on RPi 3 & RPi 3 B+ ?_

FYI our initial/experimental RPi support for Asterisk/FreePBX is now documented/updated in:
- the table at http://FAQ.IIAB.IO Item 9 ("What services (IIAB apps) are suggested during installation?")
- default_vars.yml and {MIN, MEDIUM, BIG} local_vars.yml, per PR #1494
- [roles/pbx/README.rst](https://github.com/iiab/iiab/tree/master/roles/pbx#pbx-readme), per PR #1491

holta, 2019-02-18:
Done!  Still needs work of course, as outlined @ https://github.com/iiab/iiab/issues/1493#issuecomment-463113803

closed 2019-02-18 by holta