iiab/iiab #1469: dnsmasq fails often in captive-portal / 4-server-options on multiple OS's

holta, 2019-02-09:
Ubuntu 16.04 and Debian 10 are not the only ones.

On some distros it happens every time, whereas on others it's intermittent ("./iiab-install --reinstall" can *sometimes* work to overcome this failure in 4-server-options).

In all cases, IIAB stops installing here, when dnsmasq fails to start:

https://github.com/iiab/iiab/blob/master/roles/captive-portal/tasks/main.yml#L141-L145

Thank you @deldesir @jvonau @georgejhunt for continuing to monitor this pattern towards identifying the root cause — so we can put an end to this in coming weeks if not days!

holta, 2019-02-09:
If we do not identify the root cause in coming days, we should consider a workaround so that IIAB 6.7 isn't stillborn/sabotaged across multiple OS's/distros (-:

deldesir, 2019-02-09:
For now installation goes without it when captive_portal_install: False and captive_portale_enabled: False un /etc/iiab/local_vars.yml.

"systemctl start dnsmasq" stills fails even when run manually. 

I'll wait installation to finish without captive portal and after I'll play with tasks of captive-portal role.

holta, 2019-02-09:
Thanks @deldesir!  Please let us know if roles/network succeeds during the last stage of iiab-install, and when run manually as ./iiab-network (or "./runrole network") — in all cases making sure you're in /opt/iiab/iiab

(Or possibly dnsmasq has a serious glitch on Debian 10 for the moment?)

deldesir, 2019-02-09:
Not dnsmasq fault nor captive-portal. 
netstat -lnp | grep ":53 " showed connmand took the port. Now dnsmasq start after I removed connman. 

Now I am rerunning iiab-install --reinstall with captive portal set to true for both install and enabled.

holta, 2019-02-09:
> ConnMan is an internet connection manager for embedded devices

Appropriately named ;)

How did you remove connman ?

Why is this installed as part of your Debian 10, do you know?

holta, 2019-02-09:
> Why is this installed as part of your Debian 10, do you know?

Clarify: @deldesir installed this (connman package) no worries, now we know!

Larger question about IIAB fragility in general — as we try to make dnsmasq resilient across distros, what error reporting (e.g. checking what daemon might already be using port 53?) might best communicate different ways that dnsmasq fails across different distros?

deldesir, 2019-02-09:
App remove it and then purge it. I installed it cause I wanted an easy way to manage wifi card with problematic ath10k firmware

georgejhunt, 2019-02-09:
I think we should change the default to off for captive-portal on all
hardware that is not rpi.
Our instructions could then advise the user to do the IIAB install, until
it is successful. And then configure the wifi, until it works.

Finally, they can "./runrole captive-portal", and have most of the
dependencies sorted out.

holta, 2019-02-10:
> I think we should change the default to off for captive-portal on all hardware that is not rpi.

@georgejhunt how/where do you suggest this happen?

A couple related points:
1) Right now captive-portal is True/True in MEDIUM-sized and BIG-sized local_vars.yml and we should keep stuff like that, to expand Captive Portal testing wherever possible!
2) Folks in India (that @m-anish works with) very much want Captive Portal to work on Ubuntu 18.04 — is extending what we have on Raspbian to other distro(s) likely very hard?

holta, 2019-02-11:
@georgejhunt asks a thoughtful question: if we do suppress dnsmasq glitches within 4-server-options, does this really help if they'll (presumably) just reappear later during the roles/network stage?

georgejhunt, 2019-02-11:
I installed debian 10 on a test NUC. After fixing the lack of a proper
entry in /etc/release, it ran to completion on medium without error.

So no information relating to dnsmasq was gained

holta, 2019-02-11:
While similar in some ways, is this unrelated to #1452 in the end?

holta, 2019-03-19:
@jvonau this error no longer occurs on Debian 10 according to @floydianslips 

I also no longer see it on Ubuntu 16.04 (if we really still care about that, but FYI!)

Should we declare victory and close this?

holta, 2019-05-04:
This still happens on Ubuntu 16.04, but that is now more than 3 years old and it really is is time for IIAB implementers to use more modern OS's.

closed 2019-05-04 by holta