iiab/iiab #1493: freepbx service won't start: "freepbx.service: Failed to execute command: Permission denied" -- after './runrole pbx' and './iiab-install --reinstall' both fail

holta, 2019-02-11:
@m-anish I hope you have to time to look into this?

Does the `Failed at step EXEC spawning /usr/sbin/fwconsole: Permission denied` error below shed any light?

Or is this the same underlying `[freepbx_log.INFO]: Connection attmempt to AMI failed [] []` as posted at https://github.com/iiab/iiab/pull/1478#issuecomment-462218170 ?

> root@box:~# systemctl status freepbx
● freepbx.service - FreePBX VoIP Server
   Loaded: loaded (/etc/systemd/system/freepbx.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Mon 2019-02-11 15:52:24 EST; 42s ago
  Process: 876 ExecStart=/usr/sbin/fwconsole start (code=exited, status=203/EXEC)
 Main PID: 876 (code=exited, status=203/EXEC)
>
> Feb 11 15:52:24 box.lan systemd[1]: Starting FreePBX VoIP Server...
Feb 11 15:52:24 box.lan systemd[876]: freepbx.service: Failed to execute command: Permission denied
Feb 11 15:52:24 box.lan systemd[876]: freepbx.service: Failed at step EXEC spawning /usr/sbin/fwconsole: Permission denied
Feb 11 15:52:24 box.lan systemd[1]: freepbx.service: Main process exited, code=exited, status=203/EXEC
Feb 11 15:52:24 box.lan systemd[1]: freepbx.service: Failed with result 'exit-code'.
Feb 11 15:52:24 box.lan systemd[1]: Failed to start FreePBX VoIP Server.

holta, 2019-02-11:
As we try to diagnose this, here are a few speculative theories, from @m-anish earlier today:

> on the min sized install runrole pbx completes just fine
> so something else in the big sized install must be mucking things up
> my guess is mysql is changed in a way freepbx doesnt like

> ill fire off a big sized install in a vm
> to see whats going on with pbx in that
> i am guessing something else is causing it to fail

m-anish, 2019-02-12:
* Was this a pristine install?
* Is pbx enabled and pbx install set to true?
* This looks more like a systemd failure rather than freepbx

holta, 2019-02-12:
> * Was this a pristine install?

This is the very same BIG-sized install of 9h ago (right after PR #1478 was merged) on a fresh Ubuntu Server 18.04 VM,  No code has changed since, only documentation.

> * Is pbx enabled and pbx install set to true?

Yes and yes.

> * This looks more like a systemd failure rather than freepbx

If re-running `./runrole pbx` has such serious ramifications we definitely need to dig deepeer.  Can you log in to 10.8.0.22?

m-anish, 2019-02-12:
I'm running (from scratch) a BIG sized install right now. I'm gonna check the permissions on 
`/usr/bin/fwconsole`
and
`/var/lib/asterisk/bin/fwconsole` (to which it is symlinked)
and compare your and mine setup for starters.

m-anish, 2019-02-12:
My guess is that freepbx didn't install fully or properly, originally, because:
1. If it installs properly, then on subsequent install runs, the installation process is simply skipped
2. `/var/lib/asterisk/bin/fwconsole` does not have executable permissions (+x) which looks very strange. Even when I explicitly `chmod a+x` it, I can't run it. Gonna check the install log now.

holta, 2019-02-12:
This issue (#1493) appears not to be reproducible with the lastest IIAB master (installing using http://d.iiab.io/6.7 onto Ubuntu 18.04 or Raspbian Lite).

Will keep monitoring and hopefully close before end of day?

m-anish, 2019-02-12:
Sounds good

holta, 2019-02-12:
Also important to mention that Calibre was not installed in my most recent BIG-sized installs on both above-mentioned OS's, per @jvonau's suggestion.

This (appears ??) to have cleared up a lot of issues, possibly also include Kolibri issue #1489 ?

m-anish, 2019-02-13:
For the record, I ran the BIG sized install, and I reran it on the same 18.04 VM and it worked just fine. 

So it might be some other playbook/module mucking it up. 

I had also run out of RAM and HDD space on my VM. 1GB RAM and 10GB HDD space, and had to change that. It might have been what caused it at your end?

holta, 2019-02-13:
fyi "systemctl start asterisk" needed to be manually on one RPi, before "systemctl start freepbx" would work.

Let's keep this ticket open for a bit longer in case any patterns emerge on either RPi or Ubuntu/Debian...

holta, 2019-02-13:
These timeout issues are ongoing, but largely under control on Big Iron & VMs (Ubuntu 18.04 and Debian 9.7 too so far).

On Raspberry Pi 3 timeouts are more common, while the 'pbx' playbook tries to restart the 'freepbx' systemd service.  Can the existing Asterisk-on-RPi community help us understand this!

Let's try hard to improve both above situations for IIAB 7.0's target springtime !

m-anish, 2019-02-14:
> Let's try hard to improve both above situations for IIAB 7.0's target springtime !

Agreed!

holta, 2019-02-14:
Please help us confirm this theory, those who can: seeing as "systemctl restart freepbx" fails well more than 50% of the time when run a BIG-sized install of IIAB 6.7 on RPi [*] is it possible it is substantially more reliable when run as part of a MIN-sized install?

[*] this has been tested on both Raspbian Lite and Raspbian Desktop OS's, and both RPi 3 and RPi 3 B+ hardware.

Documented for now in: https://github.com/iiab/iiab/blob/master/roles/pbx/README.rst#raspberry-pi-known-issue

holta, 2021-08-02:
@lemueldsouza is this timeout possibly no longer an issue 2.5 years later?

That would be great news, so long as the PHP 7.4 dilemma below can be overcome: (in coming months, if not today)

- PR #2896
- #2897

holta, 2021-08-03:
@lemueldsouza agrees we should probably close this 2 years later, as likely no longer relevant.

@m-anish: please do re-open if this or similar if any such re-emerges!

closed 2021-08-03 by holta