iiab/iiab #1495: "Asterisk - Install dependencies" fails on initial install on Raspbian Desktop

holta, 2019-02-11:
@m-anish can you look into this initial install failure on Raspbian Desktop on RPi 3 B+?  Note this machine (10.8.0.54) already had a BIG-sized version of IIAB 6.7/master installed on it from just a few days ago &mdash; the failure below occurred after I did "git pull" and upgraded its /etc/iiab/local_vars.yml to the latest BIG-sized from PR #1478 :

> TASK [pbx : Asterisk - Install dependencies] ***********************************
fatal: [127.0.0.1]: FAILED! => {"cache_update_time": 1549919760, "cache_updated": false, "changed": false, "msg": "'/usr/bin/apt-get -y -o \"Dpkg::Options::=--force-confdef\" -o \"Dpkg::Options::=--force-confold\"     install 'libnewt-dev' 'libncurses5-dev' 'subversion' 'libsqlite3-dev' 'libjansson-dev' 'libxml2-dev' 'uuid-dev'' failed: E: Unable to correct problems, you have held broken packages.\n", "rc": 100, "stderr": "E: Unable to correct problems, you have held broken packages.\n", "stderr_lines": ["E: Unable to correct problems, you have held broken packages."], "stdout": "Reading package lists...\nBuilding dependency tree...\nReading state information...\nSome packages could not be installed. This may mean that you have\nrequested an impossible situation or if you are using the unstable\ndistribution that some required packages have not yet been created\nor been moved out of Incoming.\nThe following information may help to resolve the situation:\n\nThe following packages have unmet dependencies:\n libsqlite3-dev : Depends: libsqlite3-0 (= 3.16.2-5+deb9u1) but 3.26.0+fossilbc891ac6b-2 is to be installed\n", "stdout_lines": ["Reading package lists...", "Building dependency tree...", "Reading state information...", "Some packages could not be installed. This may mean that you have", "requested an impossible situation or if you are using the unstable", "distribution that some required packages have not yet been created", "or been moved out of Incoming.", "The following information may help to resolve the situation:", "", "The following packages have unmet dependencies:", " libsqlite3-dev : Depends: libsqlite3-0 (= 3.16.2-5+deb9u1) but 3.26.0+fossilbc891ac6b-2 is to be installed"]}

PS please feel free to log in and diagnose directly at 10.8.0.54, Thanks!

Refs: #1467, PR #1478, #1493

m-anish, 2019-02-12:
This is a false test environment imo. I would recommend testing only from scratch, and redoing the big sized install.

holta, 2019-02-12:
> This is a false test environment imo. I would recommend testing only from scratch, and redoing the big sized install.

This is exactly the kind of test environment I used to test all other IIAB apps since 2017 &mdash; a very recent of installation of IIAB/master on Raspbian, followed by "./runrole XYZ" &mdash; so this definitely doesn't bode well.

Anyway to be clear the entire process of installing, updating Raspbian Desktop followed by a BIG-sized IIAB install takes ~4h, so there are good reasons this does not happen but every month or so.

m-anish, 2019-02-12:
> `Unable to correct problems, you have held broken packages`

This is not an asterisk issue if your iiab has a configuration in which it cannot install. Asterisk has many dependencies and requires a (generally) well functioning server where things like package dependencies are sorted and not broken.

This is exactly what I meant when I said asterisk is a major install. It is not an addon you can untar in a folder and get on with life.

AFAICT, this particular issue is a 'wontfix' for me.

holta, 2019-02-12:
Serious underlying issue complicating all Raspbian Desktop installs: Node.js (nodejs) 8.x and Node-RED (nodered) 0.19.4 are already installed as part of the base OS.

I'll work on trying to clear those up today, so IIAB's installer can hopefully run smoothly on Raspbian Desktop and Ubermix/etc too.

holta, 2019-02-13:
PR #1497 merged, and this appears resolved.  Thanks all for continued testing on Debian 9.7/9.8, Ubuntu 18.04 and Raspbian Lite/Desktop.

closed 2019-02-13 by holta