iiab/iiab #1500: node-red: Use ansible lineinfile instead of settings.js template

m-anish, 2019-02-14:
To make the node-red playbook future-proof, we should use ansible's lineinfile module to make changes to the settings.js file. Right now we copy over the settings.js file, and if the upstream version changes significantly, it may cause regressions.

* See what changes are made to the settings.js file. They are mostly in the authentication section
* Instead of copying the file over from templates, use the lineinfile module.

m-anish, 2019-02-14:
@holta please mark this as 'good-first-issue' attracting potential volunteers and 7.0 milestone.

holta, 2019-02-17:
Specific example/details @ https://github.com/iiab/iiab/pull/1497#issuecomment-463097957