iiab/iiab #1501: iiab install fails intermittently due to the refresh-wiki-docs scraper script

m-anish, 2019-02-14:
### Expected behavior and actual behavior.

IIAB install fails at the refresh-wiki-docs step intermittently. This is due to internet issues, but this is also a reliability issue. No other iiab install step fails as consistently as this one.

### Steps to reproduce the problem.

* Live in India (or any other place with half-decent 4G-ish internet)
* Install IIAB
* Fails about 60% of the time at the refresh-wiki-docs stage

### Specifications like the version of the IIAB, operating system version, or hardware details.

* Testing 6.7 on 18.04.x and debian 9.7

### Workaround

* Simply set `nodocs: True` in `/etc/iiab/local_vars.yml` before running the installation, and it takes care of this issue.

holta, 2019-02-14:
Aside: the script `/usr/bin/iiab-wiki-refresh-docs` can be manually attempted if necessary/later.  e.g. if/when Internet becomes more reliable.

Source: https://github.com/iiab/iiab/blob/master/roles/httpd/templates/refresh-wiki-docs.sh

holta, 2019-03-03:
Cannot be reproduced in many other 4G environments — so offering alternatives will not easy until the root cause(s) are understood:

Is it possible that certain 4G ISP's intentionally degrade QOS to packets from browsers that appear as scrapers like `wget` and `lynx -reload` ?

holta, 2019-03-28:
Assigned to @m-anish who is the only one we know who can reproduce this, being in India with cellular/mobile ISP's that appear to be related to the problem.

Let's all help him as best we can as more information comes to light here!

m-anish, 2019-03-28:
Okay, this did not happen in 1-2 test runs I tried over the past couple of weeks. When I get some bandwidth, I will give this another spin to see if this was a transient. If this is fixed, we can close it, and reopen it later if it reoccurs. 

Give me a few days.

mir123, 2019-04-16:
Just got this same issue in México using optic fiber internet (Axtel). Tried two times, same story. Maybe related to HTTPS?

holta, 2019-04-16:
@mir123 in Mexico cannot even run `lynx` browser for some reason -- this fails with "unexpected network error":
```
Looking up lynx.invisible-island.net
Making HTTP connection to lynx.invisible-island.net
Sending HTTP request.
HTTP request sent; waiting for response.
HTTP/1.1 301 Moved Permanently
Data transfer complete
HTTP/1.1 301 Moved Permanently
Using https://lynx.invisible-island.net/
Looking up lynx.invisible-island.net
Making HTTPS connection to lynx.invisible-island.net
Verified connection to lynx.invisible-island.net (cert=lynx.invisible-island.net)
Certificate issued by: /C=US/O=Let's Encrypt/CN=Let's Encrypt Authority X3
Secure 256-bit TLS1.3 (ECDHE_RSA_AES_256_GCM_SHA384) HTTP connection
Sending HTTP request.
HTTP request sent; waiting for response.
Alert!: Unexpected network read error; connection aborted.
Can't Access `https://lynx.invisible-island.net/'
Alert!: Unable to access document.

lynx: Can't access startfile 
```

His original/longer error (when running `iiab-install`) is pasted in here:
```
fatal: [127.0.0.1]: FAILED! => {"changed": true, "cmd": ["/usr/bin/iiab-refresh-wiki-docs"], "delta": "0:00:09.878709", "end": "2019-04-16 13:21:10.997840", "msg": "non-zero return code", "rc": 1, "start": "2019-04-16 13:21:01.119131", "stderr": "+ set -e
+ source /etc/iiab/iiab.env
++ IIAB_BASE_PATH=/opt/iiab
++ IIAB_DIR=/opt/iiab/iiab
++ IIAB_RELEASE=7.0
++ IIAB_REVISION=0
++ IIAB_GATEWAY_ENABLED=False
++ LAN_IP=172.18.96.1
++ OS=raspbian
++ OS_VER=raspbian-9
++ WWWROOT=/library/www/html
++ STAGE=3
++ FQDN=box.lan
++ IIAB_USB_LIB_SHOW_ALL=True
+ INPUT=/tmp/iiab-wiki
+ OUTPUT=/tmp/iiab-wiki.out
+ DESTPATH=/library/www/html/info
+ rm -rf /tmp/iiab-wiki
+ rm -rf /tmp/iiab-wiki.out
+ mkdir -p /tmp/iiab-wiki
+ mkdir -p /tmp/iiab-wiki.out
+ git clone https://github.com/iiab/iiab.wiki.git /tmp/iiab-wiki
Cloning into '/tmp/iiab-wiki'...
++ ls /tmp/iiab-wiki
+ for f in `ls $INPUT`
+ FTRIMMED=Home
+ '[' Home = Home ']'
+ FTRIMMED=index
+ pandoc -s /tmp/iiab-wiki/Home.md -o /tmp/iiab-wiki.out/index.html
+ for f in `ls $INPUT`
+ FTRIMMED=IIAB-6.3-Release-Notes
+ '[' IIAB-6.3-Release-Notes = Home ']'
+ pandoc -s /tmp/iiab-wiki/IIAB-6.3-Release-Notes.md -o /tmp/iiab-wiki.out/IIAB-6.3-Release-Notes.html
+ for f in `ls $INPUT`
+ FTRIMMED=IIAB-6.4-Release-Notes
+ '[' IIAB-6.4-Release-Notes = Home ']'
+ pandoc -s /tmp/iiab-wiki/IIAB-6.4-Release-Notes.md -o /tmp/iiab-wiki.out/IIAB-6.4-Release-Notes.html
+ for f in `ls $INPUT`
+ FTRIMMED=IIAB-6.5-Release-Notes
+ '[' IIAB-6.5-Release-Notes = Home ']'
+ pandoc -s /tmp/iiab-wiki/IIAB-6.5-Release-Notes.md -o /tmp/iiab-wiki.out/IIAB-6.5-Release-Notes.html
+ for f in `ls $INPUT`
+ FTRIMMED=IIAB-6.6-Release-Notes
+ '[' IIAB-6.6-Release-Notes = Home ']'
[... 56 lines cut ...]
Alert!: Unexpected network read error; connection aborted.
Can't Access `https://github.com/XSCE/xsce/wiki/IIAB-6.2-Release-Notes'
Alert!: Unable to access document.

lynx: Can't access startfile ", "stderr_lines": ["+ set -e", "+ source /etc/iiab/iiab.env", "++ IIAB_BASE_PATH=/opt/iiab", "++ IIAB_DIR=/opt/iiab/iiab", "++ IIAB_RELEASE=7.0", "++ IIAB_REVISION=0", "++ IIAB_GATEWAY_ENABLED=False", "++ LAN_IP=172.18.96.1", "++ OS=raspbian", "++ OS_VER=raspbian-9", "++ WWWROOT=/library/www/html", "++ STAGE=3", "++ FQDN=box.lan", "++ IIAB_USB_LIB_SHOW_ALL=True", "+ INPUT=/tmp/iiab-wiki", "+ OUTPUT=/tmp/iiab-wiki.out", "+ DESTPATH=/library/www/html/info", "+ rm -rf /tmp/iiab-wiki", "+ rm -rf /tmp/iiab-wiki.out", "+ mkdir -p /tmp/iiab-wiki", "+ mkdir -p /tmp/iiab-wiki.out", "+ git clone https://github.com/iiab/iiab.wiki.git /tmp/iiab-wiki", "Cloning into '/tmp/iiab-wiki'...", "++ ls /tmp/iiab-wiki", "+ for f in `ls $INPUT`", "+ FTRIMMED=Home", "+ '[' Home = Home ']'", "+ FTRIMMED=index", "+ pandoc -s /tmp/iiab-wiki/Home.md -o /tmp/iiab-wiki.out/index.html", "+ for f in `ls $INPUT`", "+ FTRIMMED=IIAB-6.3-Release-Notes", "+ '[' IIAB-6.3-Release-Notes = Home ']'", "+ pandoc -s /tmp/iiab-wiki/IIAB-6.3-Release-Notes.md -o /tmp/iiab-wiki.out/IIAB-6.3-Release-Notes.html", "+ for f in `ls $INPUT`", "+ FTRIMMED=IIAB-6.4-Release-Notes", "+ '[' IIAB-6.4-Release-Notes = Home ']'", "+ pandoc -s /tmp/iiab-wiki/IIAB-6.4-Release-Notes.md -o /tmp/iiab-wiki.out/IIAB-6.4-Release-Notes.html", "+ for f in `ls $INPUT`", "+ FTRIMMED=IIAB-6.5-Release-Notes", "+ '[' IIAB-6.5-Release-Notes = Home ']'", "+ pandoc -s /tmp/iiab-wiki/IIAB-6.5-Release-Notes.md -o /tmp/iiab-wiki.out/IIAB-6.5-Release-Notes.html", "+ for f in `ls $INPUT`", "+ FTRIMMED=IIAB-6.6-Release-Notes", "+ '[' IIAB-6.6-Release-Notes = Home ']'", "+ pandoc -s /tmp/iiab-wiki/IIAB-6.6-Release-Notes.md -o /tmp/iiab-wiki.out/IIAB-6.6-Release-Notes.html", "+ for f in `ls $INPUT`", "+ FTRIMMED=IIAB-6.7-Release-Notes", "+ '[' IIAB-6.7-Release-Notes = Home ']'", "+ pandoc -s /tmp/iiab-wiki/IIAB-6.7-Release-Notes.md -o /tmp/iiab-wiki.out/IIAB-6.7-Release-Notes.html", "+ for f in `ls $INPUT`", "+ FTRIMMED=IIAB-7.0-Release-Notes", "+ '[' IIAB-7.0-Release-Notes = Home ']'", "+ pandoc -s /tmp/iiab-wiki/IIAB-7.0-Release-Notes.md -o /tmp/iiab-wiki.out/IIAB-7.0-Release-Notes.html", "+ for f in `ls $INPUT`", "+ FTRIMMED=IIAB-Architecture", "+ '[' IIAB-Architecture = Home ']'", "+ pandoc -s /tmp/iiab-wiki/IIAB-Architecture.md -o /tmp/iiab-wiki.out/IIAB-Architecture.html", "+ for f in `ls $INPUT`", "+ FTRIMMED=IIAB-Contributors-Guide", "+ '[' IIAB-Contributors-Guide = Home ']'", "+ pandoc -s /tmp/iiab-wiki/IIAB-Contributors-Guide.md -o /tmp/iiab-wiki.out/IIAB-Contributors-Guide [... cut ...]
total size is 173,099  speedup is 0.99", "stdout_lines": ["sending incremental file list", "./", "IIAB-6.3-Release-Notes.html", "IIAB-6.4-Release-Notes.html", "IIAB-6.5-Release-Notes.html", "IIAB-6.6-Release-Notes.html", "IIAB-6.7-Release-Notes.html", "IIAB-7.0-Release-Notes.html", "IIAB-Architecture.html", "IIAB-Contributors-Guide.html", "IIAB-Installation.html", "IIAB-Menuing.html", "IIAB-Networking.html", "IIAB-Platforms.html", "IIAB-Variables.html", "Raspberry-Pi-IIAB-Images.html", "index.html", "", "sent 174,328 bytes  received 304 bytes  349,264.00 bytes/sec", "total size is 173,099  speedup is 0.99"]}
```

holta, 2019-04-16:
CLARIF: @mir123 ran `./iiab-install` successfully a month or 2 ago.

i.e. this is a classic/annoying problem of trying to outguess ISP's traffic shaping policies &mdash; that end up blocking developing world contributors from "scraping" documention to redistribute it offline :-|

Let's try to understand these 2 data points better (Mexico & India ISP's) such that the root cause can be better explained and avoided...

holta, 2019-04-16:
Data points from @mir123 when testing these at the command line from **Raspbian Lite**, over a fiber optic connection in Mexico:
- `lynx` fails (it tries to connect to https://lynx.invisible-island.net/ and fails)
- `lynx http://mit.edu` works
- `lynx https://mit.edu` works

holta, 2019-04-16:
@m-anish please help us get "Developing World" people (i.e. those who can actually reproduce this) to understand how best to make `/usr/bin/iiab-wiki-refresh-docs` _survivable_ across a broader array of developing world ISP's.

Recap: this issue has never been reproducible outside of developing world ISP's to date, so it requires participation of developing world folk to understand it and solve it meaningfully.

Short Term: gracefully failing such that critical commands like `./iiab-install` can continue (or provide an actionable error message in plain english) would be infinitely better than the current situation.

Long Term: someone needs to figure out the root cause as to why several/many developing world ISP's keep dropping lynx https packets on the floor (during network congestion especially...)

holta, 2019-04-17:
_Possibly Related:_ @mir123 in Mexico was unable to run Node.js's official install command `curl -sL https://deb.nodesource.com/setup_10.x | bash -` possibly also due to an https issue with his major Mexican ISP (?)

As reported here:

#1603 Node.js won't reinstall on Raspbian, over Mexican ISP, perhaps due to https failure?

holta, 2019-05-04:
@m-anish do you have time to look into this?

Or should it be deferred to Q3's IIAB 7.1?

m-anish, 2020-01-06:
I'll do a min sized install a few times to see if this is happening. 

Thankfully, internet is unreliable here these days, so should bring out the problem if still there.

holta, 2020-01-06:
@m-anish you can manually/repeatedly run the command `/usr/bin/iiab-refresh-wiki-docs` to evaluate the nature of ISP disruptions.

(This might save you a lot of time as compared to redoing an entire MIN-sized install if IIAB?)

holta, 2020-01-09:
@m-anish suggests we close this (for now!) as he's no longer affected.

boholder, 2021-09-24:
[image]

I got the same error log as @ mir123 just at this reply time.

env :
debian11 (in virtualbox) (on windows host)
 I'm in China mainland BTW. 

I've set wget, curl, git, linux global (export http_proxy, https_proxy) proxy 
(otherwise I can't even clone the code from github, long praise the GFW) , 
but this error can still be repeated by executing `sudo /usr/bin/iiab-refresh-wiki-docs` alone.

Any advice or maybe I can help you fix | improve this problem by providing a manually-examming-test-env? Thanks ahead.

boholder, 2021-09-25:
Yes! I got the exact same error as the Mexican friend above when I installed iiab (lynx...XSCE...Can't access startfile). But fortunately lynx allows setting up proxies, and I learned from experience that proxies are the real solution to smoothing out any weird geo-network connection issues, this worked once again.

I also encountered some errors of the same type (due to suck geo-network):

1. Downloading something (sry I forgot, maybe is `OpenStreetMap testbalabala`) from `raw.githubusercontent.com` failed with `urlopen error [Errno 99] Cannot assign requested address`, solved by [searching](https://www.ipaddress.com/) and adding hosts manually (in `/etc/hosts`). It's really a weird error.

2. pip showed `Connect failure The read operation time out` when getting some liberaries, solved by replacing pip source with a mirror in China.

I'm in China mainland, and if I don't first configure a proxy for curl, our characteristic infrastructure here, GFW, will block my download communication with `raw.githubusercontent.com`, and I won't even be able to start installing iiab. 🙄

holta, 2021-09-25:
Huge thanks @boholder, for your explanations, so that everyone can try to make this much easier for people in all countries.