iiab/iiab #1502: Fix http://box:1880 (now fails) to http://box/node-red in Admin Console

holta, 2019-02-14:
@georgejhunt @tim-moody can you help here today?

Requested by @m-anish due to Apache change he made.

holta, 2019-02-14:
CORRECTION: the URL should be http://box/nodered

m-anish, 2019-02-14:
So the essential change I am looking for is a shift from `:1880` to `/nodered` in the menudefinition.

tim-moody, 2019-02-14:
what does this have to do with the admin console?

tim-moody, 2019-02-14:
since nodered_web_path is only in the role's default vars it is not visible to js-menu

holta, 2019-02-14:
Can this be close now as a result of https://github.com/iiab/iiab-admin-console/pull/149 ?

m-anish, 2019-02-14:
@tim-moody would it help if `nodered_web_path: nodered` was in default_vars?

m-anish, 2019-02-14:
I can confirm that this works as expected. Nodered menuitem on homepage links to `/nodered`

holta, 2019-02-14:
> @tim-moody would it help if `nodered_web_path: nodered` was in default_vars?

Happening in PRs #1508 #1509

closed 2019-02-14 by holta