iiab/iiab #1504: "Captive Portal doesn't work if Ethernet cable removed"

holta, 2019-02-14:
Reported by TK Kang 

Please work with him by email to solve this, those who can!

holta, 2019-02-14:
Please help us get this right:

"Caution: this prototype functionality might not work at all if an Ethernet cable is not simultaneously in use."
https://github.com/iiab/iiab/wiki/IIAB-6.7-Release-Notes#known-issues

Which also cautions:

[Captive Portal](http://wiki.laptop.org/go/IIAB/FAQ#Captive_Portal_Administration:_What_tips_.26_tricks_exist.3F) currently doesn't work if IIAB is installed on OS's other than Raspbian.  #1366

holta, 2019-02-15:
@georgejhunt is this confirmed & also resolved, as a result of PR #1510?

holta, 2019-02-21:
@tim-moody ran a test after @georgejhunt's PR #1510 fix:
- Android 8 & Windows 10 worked UNTIL REBOOT
  - CP continues to work if you yank Ethernet, so long as you don't reboot!
  - BUT CP does NOT when booting w/o Ethernet :-(
- Curiously this behavior is just like Nodogsplash
- SOMETHING needs an upstream connection...Tim wonders if IPv6 might be the underlying cause?

tim-moody, 2019-02-23:
I would leave all related issues open until all problems are resolved.

holta, 2019-02-25:
@georgejhunt do you believe TK's Ethernet issue is possibly now solved as a result of PR #1518 ?

georgejhunt, 2019-02-25:
No I do not think so. I was unable to do a before, after change that went
from no work, to work, on android.
I expect that whatever problems TK was having, will persist

holta, 2019-02-28:
@georgejhunt cannot reproduce this: can someone talk to TK to verify this is still an issue?

holta, 2020-01-03:
@georgejhunt did testing alongside PR #2070 and believes this can 
now be closed.

(But please re-open as nec!)

closed 2020-01-03 by holta