iiab/iiab #1516: Student Privacy / Medical Confidentiality best practices prior to copying an IIAB

holta, 2019-02-23:
What are the best practices now that we no longer use @georgejhunt's xs-betaprep and xs-sysprep commands?

This needs to be compiled into http://FAQ.IIAB.IO initially, and then into an actual command that can be run by less technical operators, who will increasingly be duplicating their IIAB microSD's per http://FAQ.IIAB.IO Number 41 ("[How do I back up, shrink & copy IIAB microSD cards?](http://wiki.laptop.org/go/IIAB/FAQ#How_do_I_back_up.2C_shrink_.26_copy_IIAB_microSD_cards.3F)")

_Background: the email thread "final production/provisioning tips from George Hunt" contains many good ideas here._

holta, 2019-02-23:
As this list of clinic medical confidentiality & school privacy IIAB practices is compiled and grows more complete, it will also be mentioned in these 2 places:
- http://wiki.laptop.org/go/IIAB/FAQ#How_do_I_back_up.2C_shrink_.26_copy_IIAB_microSD_cards.3F
- http://wiki.laptop.org/go/IIAB/Security

tim-moody, 2019-02-23:
I don't see confidentiality best practices in either link above. Where is the confidential information that needs protecting?

holta, 2019-02-23:
> Where is the confidential information that needs protecting?

That's what needs to be fleshed out, on this ticket ideally &mdash; i.e. where are all the places that student/teacher and patient data might be stored in an IIAB.

Such that regional mass-duplication (of derivative IIAB's, as many are requesting) is not ethically fraught.

holta, 2022-05-26:
Here are some suggestions/recommendations to de-identify an Internet-in-a-Box prior to duplicating it:

https://github.com/iiab/iiab-factory/blob/8626f10160b6588d7ee7acb0adda4fde9cf41ebf/box/rpi/min-sd#L79-L97

Specifically:

```
# clean up some files from development
rm -f /mnt/sdcard/library/awstats/*
rm -f /mnt/sdcard/var/log/apache2/*
rm -f /mnt/sdcard/var/log/nginx/*
rm -f /mnt/sdcard/etc/iiab/uuid
rm -f /mnt/sdcard/etc/remoteit/config.json*
rm -f /mnt/sdcard/library/www/html/local_content/USB*
rm -f /mnt/sdcard/usb*

# sanitize wpa_supplicant.conf but preserve country code
COUNTRY_CODE_LINE=`cat /mnt/sdcard/etc/wpa_supplicant/wpa_supplicant.conf | grep country=`
cat <<EOF > /mnt/sdcard/etc/wpa_supplicant/wpa_supplicant.conf
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
EOF
echo $COUNTRY_CODE_LINE >> /mnt/sdcard/etc/wpa_supplicant/wpa_supplicant.conf

#echo $FILENAME > /mnt/sdcard/etc/iiab/openvpn_handle
truncate --size 0 /mnt/sdcard/opt/admin/cmdsrv/cmdsrv.*.db
```

@tim-moody will likely improve on this in coming weeks &mdash; either to remove [/etc/machine-id](https://man7.org/linux/man-pages/man5/machine-id.5.html) &mdash; or replace this file with the string "uninitialized\n" as @jvonau has suggested.