iiab/iiab #1519: Opening up ports

mitra42, 2019-02-25:
@m-anish - as requested in email ... 

### Expected behavior and actual behavior.
Exected: Opening either locally `http://192.168.0.9:4244` or remotely `http://<address of router>:24244` should hit our Internet Archive server and start a process to load the IA front end.

Actual: Connection refused

### Steps to reproduce the problem.

### Specifications like the version of the IIAB, operating system version, or hardware details.
I have installed a smaller version of the Internet Archive server on an IIAB. 

Output of uname -a: `Linux box.lan 4.14.98-v7+ #1200 SMP Tue Feb 12 20:27:48 GMT 2019 armv7l GNU/Linux` its a RaspberryPi 3. 

Its hooked to a Router which forwards port 24244 to port 4244 on the RPI. I

The router also forwards ssh, so I can provide access if someone wants to login and poke around unfortunately the IP address of the router gets changed by the ISP but I can give you a current ip/port/id/password if you message me at mitra@archive.org

@holta suggested editing /usr/bin/iiab-gen-iptables and running either it, or `sudo ./runrole network`
neither of which worked.

mitra42, 2019-02-25:
I also figured that /usr/bin/iiab-gen-iptables gets overwritten by the contents of `/opt/iiab/iiab/roles/network/templates/gateway/iiab-gen-iptables` by `sudo ./runrole network` so I've edited that file instead, with no success.

mitra42, 2019-02-25:
As requested by @m-anish 
```
# cat /etc/iiab/iiab.env

# This is a configuration file for IIAB
# It can be sourced in a shell script or read into an application
IIAB_BASE_PATH=/opt/iiab
IIAB_DIR=/opt/iiab/iiab
IIAB_RELEASE=6.7
IIAB_REVISION=0
IIAB_GATEWAY_ENABLED=False
LAN_IP=172.18.96.1
OS=raspbian
OS_VER=raspbian-9
WWWROOT=/library/www/html
STAGE=9
FQDN=box.lan
IIAB_USB_LIB_SHOW_ALL=True
IIAB_WAN_DEVICE="wlan0"
IIAB_LAN_DEVICE="br0"
HOSTAPD_ENABLED=False
HTTPCACHE_ON=False
```

mitra42, 2019-02-25:
```
pi@box:/opt/iiab/iiab $ cat /etc/iiab/iiab.ini
[location]
iiab_base = /opt/iiab
iiab_dir = /opt/iiab/iiab
[version]
distribution = Debian
arch = armv7l
iiab_base_ver = 6.7
iiab_branch = HEAD
iiab_commit = 60b62d7ed0f581a4218f5376d6ee5a7ae71e05b3
install_date = 2019-02-24T00:18:25Z
install_xo = none
[runtime]
iiab_stage = 9
iiab_base_ver = 6.7
iiab_revision = 0
runtime_php = /usr/lib/php
runtime_branch = HEAD
runtime_commit = 60b62d7ed0f581a4218f5376d6ee5a7ae71e05b3
runtime_date = 2019-02-25T02:23:16Z
ansible_version = 2.7.8
kernel = 4.14.98-v7+
memory_mb = 927
swap_mb = 499
product_id = 
gw_active = True
internet_available = False
is_rpi = True
first_run = False
local_tz = Australia/Sydney
no_NM_reload = False
is_F18 = False
FQDN_changed = False
[iiab-admin]
name = iiab-admin
description = "Admin User"
iiab_admin_user = iiab-admin
[openvpn]
name = OpenVPN
description = OpenVPN enables live/remote support by connecting machines anywhere on the Internet, via a middleman server, using Virtual Private Network (VPN) techniques to create secure connections.
enabled = False
handle = 
cron_enabled = False
server = xscenet.net
server_virtual_ip = 10.8.0.1
server_port = 1194
[mysql]
name = MySQL
description = "MySQL is a widely used free and open source (GPLv2) database, offered by most web hosting services, on a diversity of platforms."
enabled = True
[usb-lib]
name = usb-lib
description = "usb-lib automounts Teacher Content on USB drives to /library/www/html/local_content, so students can browse it almost immediately at http://box/usb"
enabled = True
[kalite]
name = KA Lite
description = "KA Lite downloads Khan Academy videos for offline use, with exercises and accounts if students want to track their own progress."
path = /library/ka-lite
port = 8008
kalite_enabled = True
cron_enabled = True
[kiwix]
name = Kiwix
description = "Part of https://github.com/kiwix/kiwix-tools/ - kiwix-serve is the most used web server for ZIM files."
kiwix_url = /kiwix/
kiwix_path = /opt/iiab/kiwix
kiwix_port = 3000
iiab_zim_path = /library/zims
kiwix_library_xml = /library/zims/library.xml
kiwix_enabled = True
[awstats]
name = AWStats
description = "AWStats (originally known as Advanced Web Statistics) is a package written in Perl which generates static or dynamic html summaries based upon web server logs."
installed = True
enabled = True
[munin]
name = Munin
description = "Munin is a networked resource monitoring tool that can help analyze resource trends and \"what just happened to kill our performance?\" problems."
installed = True
enabled = True
[vnstat]
name = vnStat
description = "vnStat is a console-based network traffic monitor for Linux and BSD that keeps a log of network traffic for the selected interface(s)."
installed = True
enabled = True
[detected_network]
has_ifcfg_gw = none
prior_gateway_(device_gw2) = "wlan0"
dhcpcd_result = enabled
network_manager_active = False
systemd_networkd_active = False
wan_in_interfaces = False
wireless_list_1(wifi1) = not found-1
wireless_list_2(wifi2) = wlan0
num_wifi_interfaces = 1
discovered_wireless_iface = wlan0
discovered_wired_iface = eth0
num_lan_interfaces = 1
gui_static_wan = False
iiab_lan_iface = none
iiab_wan_iface = wlan0
[computed_network]
iiab_wan_enabled = True
user_wan_iface = wlan0
iiab_wan_iface = wlan0
iiab_lan_enabled = True
user_lan_iface = auto
iiab_lan_iface = br0
iiab_network_mode = Gateway
hostapd_enabled = False
host_ssid = Internet in a Box
host_wifi_mode = g
host_channel = 6
[network]
dansguardian_enabled = False
squid_enabled = False
wondershaper_enabled = False
iiab_network_mode_applied = Gateway
dhcpd_enabled = False
dhcp_service2 = dnsmasq
named_enabled = False
dnsmasq_enabled = True
no_net_restart = True
[squid]
enabled = False
[dansguardian]
enabled = False
[wondershaper]
enabled = False
```

holta, 2019-02-25:
> I also figured that /usr/bin/iiab-gen-iptables gets overwritten by the contents of `/opt/iiab/iiab/roles/network/templates/gateway/iiab-gen-iptables` by `sudo ./runrole network` so I've edited that file instead, with no success.

Can you post your change to `iiab-gen-iptables` if poss?

m-anish, 2019-02-25:
> > I also figured that /usr/bin/iiab-gen-iptables gets overwritten by the contents of `/opt/iiab/iiab/roles/network/templates/gateway/iiab-gen-iptables` by `sudo ./runrole network` so I've edited that file instead, with no success.
> 
> Can you post your change to `iiab-gen-iptables` if poss?

and perhaps `iptables-save` too?

jvonau, 2019-02-25:
iptables -L -v

mitra42, 2019-02-25:
I added - around line 100 where similar lines existed - to `iiab-gen-iptables`
```
$IPTABLES -A INPUT -p tcp --dport 4244 -m state --state NEW -i $wan -j ACCEPT
```

holta, 2019-02-25:
> I added - around line 100 where similar lines existed - to `iiab-gen-iptables`
> 
> ```
> $IPTABLES -A INPUT -p tcp --dport 4244 -m state --state NEW -i $wan -j ACCEPT
> ```

@mitra42 is your port definitely TCP not UDP ?

mitra42, 2019-02-25:
```
pi@box:/opt/iiab/iiab $ sudo iptables -L -v
Chain INPUT (policy ACCEPT 0 packets, 0 bytes)
 pkts bytes target     prot opt in     out     source               destination         
 2650  406K ACCEPT     all  --  lo     any     anywhere             anywhere            
    0     0 DROP       tcp  --  any    any     anywhere             anywhere             tcp dpt:sunrpc
    0     0 DROP       udp  --  any    any     anywhere             anywhere             udp dpt:sunrpc
    0     0 DROP       tcp  --  any    any     anywhere             anywhere             tcp dpt:mysql
    0     0 DROP       udp  --  any    any     anywhere             anywhere             udp dpt:mysql
    0     0 DROP       tcp  --  any    any     anywhere             anywhere             tcp dpt:postgresql
    0     0 DROP       udp  --  any    any     anywhere             anywhere             udp dpt:postgresql
    0     0 DROP       tcp  --  any    any     anywhere             anywhere             tcp dpt:5984
    0     0 DROP       udp  --  any    any     anywhere             anywhere             udp dpt:5984
  513 37516 ACCEPT     all  --  any    any     anywhere             anywhere             state RELATED,ESTABLISHED
    2   355 ACCEPT     all  --  br0    any     anywhere             anywhere             state NEW
   78 23882 ACCEPT     udp  --  any    any     anywhere             anywhere             udp dpt:mdns
    0     0 ACCEPT     tcp  --  wlan0  any     anywhere             anywhere             tcp dpt:ssh state NEW
    0     0 ACCEPT     tcp  --  wlan0  any     anywhere             anywhere             tcp dpt:http state NEW
    2   128 ACCEPT     tcp  --  wlan0  any     anywhere             anywhere             tcp dpt:4244 state NEW
    0     0 ACCEPT     tcp  --  wlan0  any     anywhere             anywhere             tcp dpt:3000 state NEW
    0     0 ACCEPT     tcp  --  wlan0  any     anywhere             anywhere             tcp dpt:8008 state NEW
    0     0 ACCEPT     tcp  --  wlan0  any     anywhere             anywhere             tcp dpt:8009 state NEW
    0     0 ACCEPT     tcp  --  wlan0  any     anywhere             anywhere             tcp dpt:http-alt state NEW
    0     0 ACCEPT     tcp  --  wlan0  any     anywhere             anywhere             tcp dpt:8083 state NEW
    0     0 ACCEPT     tcp  --  wlan0  any     anywhere             anywhere             tcp dpt:ipp state NEW
    0     0 ACCEPT     tcp  --  wlan0  any     anywhere             anywhere             tcp dpt:8089 state NEW
    0     0 ACCEPT     tcp  --  wlan0  any     anywhere             anywhere             tcp dpt:1880 state NEW
    0     0 ACCEPT     tcp  --  wlan0  any     anywhere             anywhere             tcp dpt:1883 state NEW
    0     0 ACCEPT     tcp  --  wlan0  any     anywhere             anywhere             tcp dpt:9091 state NEW
    0     0 ACCEPT     tcp  --  wlan0  any     anywhere             anywhere             tcp dpt:51413 state NEW
    0     0 ACCEPT     udp  --  wlan0  any     anywhere             anywhere             udp dpt:30000 state NEW
   24   904 DROP       all  --  wlan0  any     anywhere             anywhere            

Chain FORWARD (policy ACCEPT 0 packets, 0 bytes)
 pkts bytes target     prot opt in     out     source               destination         
    0     0 ACCEPT     all  --  wlan0  br0     anywhere             anywhere             state RELATED,ESTABLISHED
    0     0 ACCEPT     all  --  br0    wlan0   anywhere             anywhere            
    0     0 DROP       all  --  wlan0  br0     anywhere             anywhere            

Chain OUTPUT (policy ACCEPT 3012 packets, 449K bytes)
 pkts bytes target     prot opt in     out     source               destination         
pi@box:/opt/iiab/iiab $ 
```

mitra42, 2019-02-25:
```
pi@box:/opt/iiab/iiab $ sudo iptables-save
# Generated by iptables-save v1.6.0 on Mon Feb 25 13:51:37 2019
*nat
:PREROUTING ACCEPT [35:2124]
:INPUT ACCEPT [10:1184]
:OUTPUT ACCEPT [62:6072]
:POSTROUTING ACCEPT [62:6072]
COMMIT
# Completed on Mon Feb 25 13:51:37 2019
# Generated by iptables-save v1.6.0 on Mon Feb 25 13:51:37 2019
*filter
:INPUT ACCEPT [0:0]
:FORWARD ACCEPT [0:0]
:OUTPUT ACCEPT [3027:455482]
-A INPUT -i lo -j ACCEPT
-A INPUT -p tcp -m tcp --dport 111 -j DROP
-A INPUT -p udp -m udp --dport 111 -j DROP
-A INPUT -p tcp -m tcp --dport 3306 -j DROP
-A INPUT -p udp -m udp --dport 3306 -j DROP
-A INPUT -p tcp -m tcp --dport 5432 -j DROP
-A INPUT -p udp -m udp --dport 5432 -j DROP
-A INPUT -p tcp -m tcp --dport 5984 -j DROP
-A INPUT -p udp -m udp --dport 5984 -j DROP
-A INPUT -m state --state RELATED,ESTABLISHED -j ACCEPT
-A INPUT -i br0 -m state --state NEW -j ACCEPT
-A INPUT -p udp -m udp --dport 5353 -j ACCEPT
-A INPUT -i wlan0 -p tcp -m tcp --dport 22 -m state --state NEW -j ACCEPT
-A INPUT -i wlan0 -p tcp -m tcp --dport 80 -m state --state NEW -j ACCEPT
-A INPUT -i wlan0 -p tcp -m tcp --dport 4244 -m state --state NEW -j ACCEPT
-A INPUT -i wlan0 -p tcp -m tcp --dport 3000 -m state --state NEW -j ACCEPT
-A INPUT -i wlan0 -p tcp -m tcp --dport 8008 -m state --state NEW -j ACCEPT
-A INPUT -i wlan0 -p tcp -m tcp --dport 8009 -m state --state NEW -j ACCEPT
-A INPUT -i wlan0 -p tcp -m tcp --dport 8080 -m state --state NEW -j ACCEPT
-A INPUT -i wlan0 -p tcp -m tcp --dport 8083 -m state --state NEW -j ACCEPT
-A INPUT -i wlan0 -p tcp -m tcp --dport 631 -m state --state NEW -j ACCEPT
-A INPUT -i wlan0 -p tcp -m tcp --dport 8089 -m state --state NEW -j ACCEPT
-A INPUT -i wlan0 -p tcp -m tcp --dport 1880 -m state --state NEW -j ACCEPT
-A INPUT -i wlan0 -p tcp -m tcp --dport 1883 -m state --state NEW -j ACCEPT
-A INPUT -i wlan0 -p tcp -m tcp --dport 9091 -m state --state NEW -j ACCEPT
-A INPUT -i wlan0 -p tcp -m tcp --dport 51413 -m state --state NEW -j ACCEPT
-A INPUT -i wlan0 -p udp -m udp --dport 30000 -m state --state NEW -j ACCEPT
-A INPUT -i wlan0 -j DROP
-A FORWARD -i wlan0 -o br0 -m state --state RELATED,ESTABLISHED -j ACCEPT
-A FORWARD -i br0 -o wlan0 -j ACCEPT
-A FORWARD -i wlan0 -o br0 -j DROP
COMMIT
# Completed on Mon Feb 25 13:51:37 2019
pi@box:/opt/iiab/iiab $ 
```

mitra42, 2019-02-25:
@holta - yes its TCP

tim-moody, 2019-02-25:
and it is open above.  (-A INPUT -i wlan0 -p tcp -m tcp --dport 4244 -m state --state NEW -j ACCEPT)

I would verify the port forwarding by changing the router to forward 24244 to 80 and then access it externally with a browser or curl or wget or lynx. If you get back the home page port forwarding is working.

mitra42, 2019-02-25:
I'm seeing one thing that may be related - and anyway is making testing hard ! It seems to switch back and forth to a state where DNS resolution fails. 
`ping: mitra.biz: Temporary failure in name resolution`
I don't think this is causing the problem as it should at least respond without DNS resolution, but fixing that would help. 

Sometimes rebooting clears this problem for a while, but I'm loath to do that since I'm ssh-ed in remotely at the moment, and that could cause me to lose connection. 

By the way ... the same physical box, running NOOBS but not IIAB, with our server was running fine ... these problems seem to be IIAB specific.

mitra42, 2019-02-25:
@tim-moody Unfortunately I'm remote from the box at the moment (ssh-ed into it), and the router doesn't allow remote edits.  Port forwarding could be one problem (there were definitely others, since it wasn't working when accessing locally before I left the site).

mitra42, 2019-02-25:
Phew... looks like I've got the port fixed - though still need to get DNS fixed. In case someone else hits this type of problem the issues were.
a) Ports closed by default
b) incomplete instructions to open them (@holta, told me to edit the Git file, which I interpreted as editing `/usr/bin/iiab-gen-iptables` which gets overwritten by `sudo ./runrole network`) 
c) DNS lookup is intermittent (though doesn't seem to be causing this problem).
d) somewhere during the process, the server crashed - maybe one of these processes caused an unhandled error in the server. I'd checked the server was still running before starting this process but didn't notice that crash, so even when the ports were fixed I was still seeing issues.

By the way ... I noticed documentation and instructions on IIAB often requires guessing, and even as a relative expert I'm guessing wrong.  I wrote up notes during installation, should I post them as a separate issue here, or is there a better place? 

Should the DNS intermittency be posted a separate issue, or can someone help fix that here as well.

holta, 2019-02-25:
@mitra42 plz send the password to my email investigate the DNS / name resolution issue.

Also plz send the installation struggle notes you wrote up, Thx!

mitra42, 2019-02-25:
Password and installation notes/suggestions sent in email.

m-anish, 2019-02-25:
@mitra42 @holta  Is this resolved besides the additional documentation that may be created/improved resulting from this?

mitra42, 2019-02-25:
The port opening is solved, DNS opening not yet, 

documentation suggestions supplied to Adam who is opening up new issues related to it.

holta, 2019-02-25:
@jvonau's #1523 proposed fix turns out to be insufficient to solve the DNS (name resolution) failure when installing IIAB over WiFi.

Apologies.  This may take a couple days to get right.  Meantime rebooting certainly works.  Then after we solve dnsmasq's DNS properly, confusing/interim reboots will no longer be necessary.

PS also note that installing over Ethernet works far more cleanly in general, as is the general recommendation to avoid hassles of many kind.  Installing over Ethernet over saves you from the need to run `iiab-hotspot-on` (though yes we'll also work on making that toggle far more clean, i.e. #1520 should work in Admin Console and be explained much better in general!)

mitra42, 2019-02-25:
So maybe that suggestion for Ethernet needs to be clearer in the documentation. 

From what I read, I assumed the issue was about speed/reliability (of the network), not about something in the install.  Since a) my WiFi wasn't the bottleneck - my router<>internet connection  was  b) I didn't have the 5 meter Ethernet cable needed to get to the router and c) with only 5 meter distance its pretty reliable.  I didn't see any reason to use the Ethernet.

holta, 2019-02-25:
> Since a) my WiFi wasn't the bottleneck - my router<>internet connection was b) I didn't have the 5 meter Ethernet cable needed to get to the router and c) with only 5 meter distance its pretty reliable. I didn't see any reason to use the Ethernet.

You're not the only one!  This is such a common problem (people prefer WiFi installs over Ethernet installs in modern times, no matter what we tell them) that it's high we make installing IIAB over WiFi "just work" i.e. far more reliably & with cleaner explanation along the way :-)

Working on it in coming days...Thx!

holta, 2019-02-25:
I'm pasting in 5 files to help @jvonau understand why DNS failed during my most recent WiFi install of IIAB 7.0/master (http://d.iiab.io) onto MIN-sized Raspbian Lite on RPi 3 B+:
```
root@raspberrypi:~# cat /etc/resolv.conf
# Generated by resolvconf
nameserver 127.0.0.1
```
```
root@raspberrypi:~# more /etc/iiab/iiab.env
# This is a configuration file for IIAB
# It can be sourced in a shell script or read into an application
IIAB_BASE_PATH=/opt/iiab
IIAB_DIR=/opt/iiab/iiab
IIAB_RELEASE=7.0
IIAB_REVISION=0
IIAB_GATEWAY_ENABLED=False
LAN_IP=172.18.96.1
OS=raspbian
OS_VER=raspbian-9
WWWROOT=/library/www/html
STAGE=9
FQDN=box.lan
IIAB_USB_LIB_SHOW_ALL=True
IIAB_WAN_DEVICE="wlan0"
IIAB_LAN_DEVICE="br0"
HOSTAPD_ENABLED=False
HTTPCACHE_ON=False
```

129-line `/etc/iiab/iiab.ini` is here:
https://paste.ubuntu.com/p/MrTXbM9NHY/

2263-line `/opt/iiab/iiab/iiab-install.log` is here:
http://paste.ubuntu.com/p/wzFcgVtYBq/

19-line `/opt/iiab/iiab-admin-console/admin-install.log` is here:
https://paste.ubuntu.com/p/yVkY42PDNS/

holta, 2019-05-08:
@mitra42 can we close this now, given the just-merged PR #1636 dnsmasq fix, for initial network configuration, on many laptops/NUCs etc?

mitra42, 2019-05-08:
Sure

closed 2019-05-08 by mitra42