iiab/iiab #1520: Admin Console & instructions should offer iiab-hotspot-on functionality more intuitively

holta, 2019-02-25:
Recommendation from @mitra42:

> Once I got on [ http://box.la/admin ] ...
Internal WiFi AP - says its enabled, but is not showing up. There is a line 'hostapd is enabled.' but that is cryptic, luckily I remembered a message back at the top of install.txt about running 'iiab-hotspot-on'. Suggestion: a) check if 'hostapd is enabled' and report here. b) either if possible allow running 'iiab-hotspot-on' from the admin page, and if not give instructions on that  admin page on how to do so.

holta, 2019-02-25:
A Univ of Arkansas TA faced exactly the same issues earlier this week, after she installed IIAB using Wi-Fi — without realizing that her Ethernet was not working..

She eventually solved the problems by running `iiab-hotspot-on` but indeed this needs to be much more intuitive, e.g. the Admin Console should offer a quite vocal reminder to run that — or built-in functionality to do the same.

holta, 2019-02-25:
@jvonau suggests we might put a LOUD BANNER in the implementer's face every time they ssh in, similar to Raspbian & IIAB's security warnings about weak passwords, but also educating the implementer about the need to possibly run `sudo iiab` and `iiab-hotspot-on`:

If so emulating these 2 files:
https://github.com/iiab/iiab/blob/master/roles/iiab-admin/templates/lxde_ssh_warn.sh
https://github.com/iiab/iiab/blob/master/roles/iiab-admin/templates/profile_ssh_warn.sh

As installed by:
https://github.com/iiab/iiab/blob/master/roles/iiab-admin/tasks/main.yml#L24-L44

tim-moody, 2019-02-25:
how did they ssh in, over the wan? is the wan wlan0? will they lose the connection by running iiab-hostspot-on?

mitra42, 2019-02-25:
The places  it really needs IMHO are 
a) at the end of the "sudo iiab" process, since the only place I saw it tell me to do this was in the install.sh 
b) In the Admin page where you are changing WiFi settings and where the relationship between "hostapd  is enabled." and "iiab-hotspot-on" is really something only IIAB experts are going to know. (Not someone trying to install a IIAB to take to the field).

holta, 2019-02-25:
> how did they ssh in, over the wan? is the wan wlan0? will they lose the connection by running iiab-hostspot-on?

In the case of Univ of Arkansas, I (think) she has a screen attached to her Raspbian Desktop, and does not use ssh at all.  Indeed she says that typing in `iiab-hotspot-on` solved all her IIAB solutions problems in the end.

holta, 2019-02-26:
Ref / Work-In-Progress: #1527 "RPi installs w/ Ethernet AND WiFi very flaky: (1) disconnects all ssh sessions (2) forces auto-creation of unwanted hotspot"

holta, 2019-04-27:
Related:

#796 Rapid networking & other IIAB config changes that can't wait ~20min for Ansible

mitra42, 2019-07-02:
@holta did this make it in, (in terms of modifying my installation docs which currently say at the start 
```
* If at all possible insert Ethernet, otherwise it will work over WiFi with some extra steps.
```
And at the bottom ...
```
And if you want to run as a local WiFi hotspot (recommended) then from the ssh prompt..
```
iiab-hotspot-on
```

holta, 2019-07-02:
@mitra42:

@tim-moody has done a ton of work here, allowing an increasing amount of `iiab-hotspot-on` and `iiab-hotspot-off` functionality from IIAB's Admin Console (http://box.lan/admin) as he can explain more.

But also keep in mind that many intimately related use-cases are fraught here (during IIAB installation especially) and particularly if the implementer tries to install everything from their building's WiFi, and doesn't realize they _Can't Have It Both Ways._

(As such this is a work in progress, but please reiterate to those willing to listen, that IIAB installation will always be far easier to explain if a live-to-Internet Ethernet cable is handy.)

jvonau, 2021-08-29:
#2739

jvonau, 2022-05-10:
coming up on 3 years are we there yet?

holta, 2022-05-10:
> coming up on 3 years are we there yet?

_Thanks for surfacing this complicated UX question._

(I don't know if it's solvable, but it definitely merits investigation!)

In the same vein, a couple tangentially related UX/onboarding questions below &mdash; that likewise represent nearly universal hassles to almost all newcomers &mdash; desperately trying to avoid settings complexity! :scream:

- PR #3070
- #3097 and associated PR #3179

jvonau, 2025-01-27:
Are we there yet?

holta, 2025-01-27:
Still a UX conundrum, as losing connection to Admin Console (when IIAB WiFi hotspot is turned off) will inherently confuse many newcomers. 😱

- The 2 commands `sudo iiab-hotspot-on` and `sudo iiab-hotspot-off` might well be enough; certainly they have the advantage that they force a modicum of CLI familiarity.

- Unless Tim has inspiration — to make something similar in Admin Console — without overly frustrating newcomers?

tim-moody, 2025-01-27:
There are two models for using iiab with upstream routers, turn up_down on and use at same time as hotspot, or turn up_down off and switch between them. I have tried to support both, though the former is the default and the latter has not been tested for awhile. 

If up_down is off, then you are supposed to see a control to turn hotspot on or off, but only then.

As an aside, I think one of the reasons that cmdsrv was starting hostapd during init was to handle the case where someone turned it on or off and then restarted cmdsrv. The worry is that you can lock yourself out. I even added the ability to connect to Admin Console over Bluetooth as a backup.

tim-moody, 2025-01-27:
In adm cons I unchecked up_down, saved and ICO. Then I got this. Not sure why hotspot is off rather than on after the change.

[image]

(I am sshed in over ethernet.)

jvonau, 2025-01-27:
I'll have the long winded answer later tonight, have Dr appt and other errands. Short answer is you asked to not use ap0. How do you have "Switch Wifi to IIAB Hotspot" rigged up?

tim-moody, 2025-01-27:
>Short answer is you asked to not use ap0.

not sure what you're saying.

>How do you have "Switch Wifi to IIAB Hotspot" rigged up?

https://github.com/iiab/iiab-admin-console/blob/176a77ee4a580d2554aed2486cb446d785616b00/roles/cmdsrv/files/iiab-cmdsrv3.py#L1425-L1463

This is where the interfaces stand:

```
root@box:~# ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether 2c:cf:67:0a:aa:ca brd ff:ff:ff:ff:ff:ff
    inet 192.168.3.60/24 brd 192.168.3.255 scope global dynamic noprefixroute eth0
       valid_lft 83204sec preferred_lft 83204sec
    inet6 fe80::e522:357a:c6a1:eddb/64 scope link noprefixroute
       valid_lft forever preferred_lft forever
3: wlan0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc pfifo_fast state DOWN group default qlen 1000
    link/ether 2c:cf:67:0a:aa:cb brd ff:ff:ff:ff:ff:ff
4: br0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue state DOWN group default qlen 1000
    link/ether 56:4f:56:df:21:01 brd ff:ff:ff:ff:ff:ff
    inet 10.10.10.10/24 brd 10.10.10.255 scope global br0
       valid_lft forever preferred_lft forever
5: ap0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc pfifo_fast state DOWN group default qlen 1000
    link/ether 02:b7:40:ac:21:dc brd ff:ff:ff:ff:ff:ff permaddr 2e:cf:67:aa:aa:cb

```

tim-moody, 2025-01-27:
after reboot, however

```
root@box:~# ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether 2c:cf:67:0a:aa:ca brd ff:ff:ff:ff:ff:ff
    inet 192.168.3.60/24 brd 192.168.3.255 scope global dynamic noprefixroute eth0
       valid_lft 86397sec preferred_lft 86397sec
    inet6 fe80::e522:357a:c6a1:eddb/64 scope link noprefixroute
       valid_lft forever preferred_lft forever
3: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast master br0 state UP group default qlen 1000
    link/ether 2c:cf:67:0a:aa:cb brd ff:ff:ff:ff:ff:ff
4: br0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default qlen 1000
    link/ether 56:4f:56:df:21:01 brd ff:ff:ff:ff:ff:ff
    inet 10.10.10.10/24 brd 10.10.10.255 scope global br0
       valid_lft forever preferred_lft forever

```

[image]

jvonau, 2025-01-27:
> > Short answer is you asked to not use ap0.
> 
> not sure what you're saying.
>

In place of using ap0, wlan0 is inserted into the bridge with no hope of using the device with an upstream AP use `bridge -d link` to have a better understanding
 
> https://github.com/iiab/iiab-admin-console/blob/176a77ee4a580d2554aed2486cb446d785616b00/roles/cmdsrv/files/iiab-cmdsrv3.py#L1425-L1463
>
> This is where the interfaces stand:
>
> ```
> root@box:~# ip a
> 1: [...]

That would look normal, br0 and ap0 for the hotspot. The shuffling over to wlan0 occurs as a result of the revised files in place/absent from running the network role and takes effect upon reboot.

jvonau, 2025-01-28:
> after reboot, however
>

The stage was set with running the network role, would of been updated for up_down False while hostapd.conf & hotspot-on|off should reflect that change. 

> ```
> root@box:~# ip a
> 1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
>     link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
>     inet 127.0.0.1/8 scope [...]

Now you have wlan0 as part of br0 with no ap0 present

jvonau, 2025-01-28:
> There are two models for using iiab with upstream routers, turn up_down on and use at same time as hotspot, or turn up_down off and switch between them. I have tried to support both, though the [...]

Makes sense, if wifi was the only connection things get dicey can't have the only connection go away, hotspot on|off just changes enable|disable only for that reason, one needs to disable what was controlling wlan0 for the upstream connection.
 
> As an aside, I think one of the reasons that cmdsrv was starting hostapd during init was to handle the case where someone turned it on or off and then restarted cmdsrv. The worry is that you can lock yourself out. I even added the ability to connect to Admin Console over Bluetooth as a backup.

Bluetooth worked great under br0 the last time I tested it.

jvonau, 2025-01-28:
> > How do you have "Switch Wifi to IIAB Hotspot" rigged up?
> 
> https://github.com/iiab/iiab-admin-console/blob/176a77ee4a580d2554aed2486cb446d785616b00/roles/cmdsrv/files/iiab-cmdsrv3.py#L1425-L1463

The subject line is present in admin-console is there any reason to keep this issue open?

holta, 2025-01-28:
Feel free to re-open this issue if there's a need!

tim-moody, 2025-01-28:
>The stage was set with running the network role

The question is there a way to activate hostapd without rebooting. Would restarting NetworkManager do it?

tim-moody, 2025-01-28:
>Bluetooth worked great under br0 the last time I tested it.

I haven't tested it for a couple of years, but the main drawback is number of steps to establish a network connection over bluetooth on windows.

tim-moody, 2025-01-28:
I tried going back the other way by checking up down, save, ICO.

ansible failed

```
PLAY [all] *********************************************************************

TASK [Gathering Facts] *********************************************************
[WARNING]: Platform linux on host 127.0.0.1 is using the discovered Python
interpreter at /usr/bin/python3.11, but future installation of another Python
interpreter could change the meaning of that path. See
https://docs.ansible.com/ansible-
core/2.18/reference_appendices/interpreter_discovery.html for more information.
ok: [127.0.0.1]

TASK [0-init : ...IS BEGINNING ============================================] ***
ok: [127.0.0.1]

TASK [0-init : Ansible just ran /etc/ansible/facts.d/local_facts.fact to set 15 vars -- here we extract 6 of those -- iiab_stage: 9, rpi_model: Raspberry Pi 5 Model B Rev 1.0, devicetree_model: Raspberry Pi 5 Model B Rev 1.0, os_ver: raspbian-12, python_version: 3.11, php_version: 8.2] ***
ok: [127.0.0.1]

TASK [0-init : Create /etc/iiab/iiab.ini, if it doesn't exist] *****************
skipping: [127.0.0.1]

TASK [0-init : Set first_run flag] *********************************************
skipping: [127.0.0.1]

TASK [0-init : Copy iiab-update & iiab-summary & iiab-diagnostics & iiab-root-login from /opt/iiab/iiab/scripts/ to /usr/bin/] ***
ok: [127.0.0.1] => (item=iiab-update)
ok: [127.0.0.1] => (item=iiab-summary)
ok: [127.0.0.1] => (item=iiab-diagnostics)
ok: [127.0.0.1] => (item=iiab-root-login)

TASK [0-init : Symlink /usr/bin/iiab-upgrade -> /usr/bin/iiab-update] **********
ok: [127.0.0.1]

TASK [0-init : Create globally-writable directory /etc/iiab/diag (0777) so non-root users can run 'iiab-diagnostics'] ***
ok: [127.0.0.1]

TASK [0-init : Pre-check that IIAB's "XYZ_install" + "XYZ_enabled" vars (1) are defined, (2) are boolean-not-string variables, and (3) contain plausible values. Also checks that "XYZ_install" is True when "XYZ_installed" is defined.] ***
included: /opt/iiab/iiab/roles/0-init/tasks/validate_vars.yml for 127.0.0.1

TASK [0-init : Set vars_checklist for 45 + 45 + 40 vars ("XYZ_install" + "XYZ_enabled" + "XYZ_installed") to be checked] ***
ok: [127.0.0.1]

TASK [0-init : Assert that 45 "XYZ_install" vars are all... defined] ***********
ok: [127.0.0.1] => (item=hostapd)
ok: [127.0.0.1] => (item=dnsmasq)
ok: [127.0.0.1] => (item=bluetooth)
ok: [127.0.0.1] => (item=sshd)
ok: [127.0.0.1] => (item=tailscale)
ok: [127.0.0.1] => (item=remoteit)
ok: [127.0.0.1] => (item=admin_console)
ok: [127.0.0.1] => (item=squid)
ok: [127.0.0.1] => (item=cups)
ok: [127.0.0.1] => (item=samba)
ok: [127.0.0.1] => (item=usb_lib)
ok: [127.0.0.1] => (item=azuracast)
ok: [127.0.0.1] => (item=gitea)
ok: [127.0.0.1] => (item=jupyterhub)
ok: [127.0.0.1] => (item=lokole)
ok: [127.0.0.1] => (item=mysql)
ok: [127.0.0.1] => (item=mediawiki)
ok: [127.0.0.1] => (item=mosquitto)
ok: [127.0.0.1] => (item=nodejs)
ok: [127.0.0.1] => (item=nodered)
ok: [127.0.0.1] => (item=nextcloud)
ok: [127.0.0.1] => (item=wordpress)
ok: [127.0.0.1] => (item=kalite)
ok: [127.0.0.1] => (item=kolibri)
[... 807 lines cut ...]
ok: [127.0.0.1] => (item={'option': 'host_ssid', 'value': 'Internet in a Box'})
ok: [127.0.0.1] => (item={'option': 'host_wifi_mode', 'value': 'g'})
changed: [127.0.0.1] => (item={'option': 'wifi_up_down', 'value': True})
ok: [127.0.0.1] => (item={'option': 'host_country_code_applied', 'value': 'US'})
ok: [127.0.0.1] => (item={'option': 'host_channel', 'value': 6})

TASK [network : Restart services] **********************************************
included: /opt/iiab/iiab/roles/network/tasks/restart.yml for 127.0.0.1

TASK [network : Restart wpa_supplicant service] ********************************
skipping: [127.0.0.1] => (item=wpa_supplicant)
skipping: [127.0.0.1]

TASK [network : Enable & Restart networkd-dispatcher.service] ******************
changed: [127.0.0.1]

TASK [network : Clone wifi if needed] ******************************************
changed: [127.0.0.1]

TASK [network : Restart hostapd when WiFi is present but not when using WiFi as gateway] ***
fatal: [127.0.0.1]: FAILED! => {"changed": false, "msg": "Unable to restart service hostapd: Job for hostapd.service failed because the control process exited with error code.\nSee \"systemctl status hostapd.service\" and \"journalctl -xeu hostapd.service\" for details.\n"}

PLAY RECAP *********************************************************************
127.0.0.1 : ok=103 changed=26 unreachable=0 failed=1 skipped=71 rescued=0 ignored=1
```

tim-moody, 2025-01-28:
```
root@box:~# systemctl status hostapd
× hostapd.service - Hostapd IEEE 802.11 AP, IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator
     Loaded: loaded (/etc/systemd/system/hostapd.service; enabled; preset: enabled)
     Active: failed (Result: exit-code) since Tue 2025-01-28 14:51:57 GMT; 5min ago
   Duration: 22h 5min 11.270s
    Process: 5759 ExecStart=/usr/sbin/hostapd -B -P /run/hostapd.pid /etc/hostapd/hostapd.conf (code=exited, status=1/FAILURE)
        CPU: 4ms

Jan 28 14:51:57 box hostapd[5759]: nl80211: deinit ifname=ap0 disabled_11b_rates=0
Jan 28 14:51:57 box hostapd[5759]: Could not read interface ap0 flags: No such device
Jan 28 14:51:57 box hostapd[5759]: nl80211 driver initialization failed.
Jan 28 14:51:57 box hostapd[5759]: ap0: interface state UNINITIALIZED->DISABLED
Jan 28 14:51:57 box hostapd[5759]: ap0: AP-DISABLED
Jan 28 14:51:57 box hostapd[5759]: ap0: CTRL-EVENT-TERMINATING
Jan 28 14:51:57 box hostapd[5759]: hostapd_free_hapd_data: Interface ap0 wasn't started
Jan 28 14:51:57 box systemd[1]: hostapd.service: Control process exited, code=exited, status=1/FAILURE
Jan 28 14:51:57 box systemd[1]: hostapd.service: Failed with result 'exit-code'.
Jan 28 14:51:57 box systemd[1]: Failed to start hostapd.service - Hostapd IEEE 802.11 AP, IEEE 802.1X/WPA/WPA2/EAP/RADIUS 
```

tim-moody, 2025-01-28:
```
A start job for unit hostapd.service has begun execution.
░░
░░ The job identifier is 1986.
Jan 28 14:51:57 box hostapd[5759]: Could not read interface ap0 flags: No such device
Jan 28 14:51:57 box hostapd[5759]: nl80211: Driver does not support authentication/association or connect commands
Jan 28 14:51:57 box hostapd[5759]: nl80211: deinit ifname=ap0 disabled_11b_rates=0
Jan 28 14:51:57 box hostapd[5759]: Could not read interface ap0 flags: No such device
Jan 28 14:51:57 box hostapd[5759]: nl80211 driver initialization failed.
Jan 28 14:51:57 box hostapd[5759]: ap0: interface state UNINITIALIZED->DISABLED
Jan 28 14:51:57 box hostapd[5759]: ap0: AP-DISABLED
Jan 28 14:51:57 box hostapd[5759]: ap0: CTRL-EVENT-TERMINATING
Jan 28 14:51:57 box hostapd[5759]: hostapd_free_hapd_data: Interface ap0 wasn't started
Jan 28 14:51:57 box systemd[1]: hostapd.service: Control process exited, code=exited, status=1/FAILURE
░░ Subject: Unit process exited
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░
░░ An ExecStart= process belonging to unit hostapd.service has exited.
░░
░░ The process' exit code is 'exited' and its exit status is 1.
Jan 28 14:51:57 box systemd[1]: hostapd.service: Failed with result 'exit-code'.
░░ Subject: Unit failed
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░
░░ The unit hostapd.service has entered the 'failed' state with result 'exit-code'.
Jan 28 14:51:57 box systemd[1]: Failed to start hostapd.service - Hostapd IEEE 802.11 AP, IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticat>
░░ Subject: A start job for unit hostapd.service has failed
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░
░░ A start job for unit hostapd.service has finished with a failure.
░░
░░ The job identifier is 1986 and the job result is failed.

```

jvonau, 2025-01-28:
> > The stage was set with running the network role
> 
> The question is there a way to activate hostapd without rebooting. Would restarting NetworkManager do it?

>NM is restarted during the TASK [network : Reload systemd] ************************************************
skipping: [127.0.0.1]
TASK [network : Restart the NetworkManager service] ****************************
changed: [127.0.0.1]

We restart NM but damon-reload is conditional upon `when: not iiab_lan_iface == "br0"` so the new drop in exclude file below that was placed into the filesystem was not respected    

>TASK [network : Copy ap0-manage.conf for NetworkManager] ***********************
ok: [127.0.0.1]
TASK [network : Copy manage.conf for NetworkManager when wifi_up_down False] ***
skipping: [127.0.0.1]
TASK [network : Remove manage.conf for NetworkManager when wifi_up_down True] ***
changed: [127.0.0.1]

Think the reloading of systemd would be a benefit for not having to reboot, try adding `not no_net_restart`
ie:
`not no_net_restart or not iiab_lan_iface == "br0"`

to the reload systemd stanza in roles/network/tasks/NM-debian.yml

tim-moody, 2025-01-28:
made that change and went from up down checked to unchecked

ansible succeeded,m but 

```
root@box:/opt/iiab/iiab# systemctl status hostapd
× hostapd.service - Hostapd IEEE 802.11 AP, IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator
     Loaded: loaded (/etc/systemd/system/hostapd.service; enabled; preset: enabled)
     Active: failed (Result: exit-code) since Tue 2025-01-28 16:44:34 GMT; 1min 4s ago
   Duration: 6.254s
    Process: 5693 ExecStart=/usr/sbin/hostapd -P /run/hostapd.pid /etc/hostapd/hostapd.conf (code=exited, status=1/FAILURE)
   Main PID: 5693 (code=exited, status=1/FAILURE)
        CPU: 5ms

Jan 28 16:44:34 box hostapd[5693]: nl80211: Could not configure driver mode
Jan 28 16:44:34 box hostapd[5693]: nl80211: deinit ifname=wlan0 disabled_11b_rates=0
Jan 28 16:44:34 box hostapd[5693]: nl80211 driver initialization failed.
Jan 28 16:44:34 box hostapd[5693]: wlan0: interface state UNINITIALIZED->DISABLED
Jan 28 16:44:34 box hostapd[5693]: wlan0: AP-DISABLED
Jan 28 16:44:34 box hostapd[5693]: wlan0: CTRL-EVENT-TERMINATING
Jan 28 16:44:34 box hostapd[5693]: hostapd_free_hapd_data: Interface wlan0 wasn't started
Jan 28 16:44:34 box systemd[1]: hostapd.service: Main process exited, code=exited, status=1/FAILURE
Jan 28 16:44:34 box systemd[1]: hostapd.service: Failed with result 'exit-code'.
Jan 28 16:44:34 box systemd[1]: Stopped hostapd.service - Hostapd IEEE 802.11 AP, IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator.

```

tim-moody, 2025-01-28:
```
root@box:/opt/iiab/iiab# systemctl restart hostapd
root@box:/opt/iiab/iiab# systemctl status hostapd
● hostapd.service - Hostapd IEEE 802.11 AP, IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator
     Loaded: loaded (/etc/systemd/system/hostapd.service; enabled; preset: enabled)
     Active: active (running) since Tue 2025-01-28 16:52:16 GMT; 6s ago
   Main PID: 6515 (hostapd)
      Tasks: 1 (limit: 9572)
        CPU: 5ms
     CGroup: /system.slice/hostapd.service
             └─6515 /usr/sbin/hostapd -P /run/hostapd.pid /etc/hostapd/hostapd.conf

Jan 28 16:52:16 box systemd[1]: Started hostapd.service - Hostapd IEEE 802.11 AP, IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator.
Jan 28 16:52:16 box hostapd[6515]: wlan0: interface state UNINITIALIZED->COUNTRY_UPDATE
Jan 28 16:52:16 box hostapd[6515]: wlan0: interface state COUNTRY_UPDATE->ENABLED
Jan 28 16:52:16 box hostapd[6515]: wlan0: AP-ENABLED

```

tim-moody, 2025-01-28:
unchecked to checked now worked

jvonau, 2025-01-28:
You want to put the PR in or shall I? If you do just tag this issue in the PR

jvonau, 2025-01-28:
I'd be inclined to just have the user reboot, not enough info to form an opinion without seeing the ansible run

tim-moody, 2025-01-28:
>Odd I would of thought ansible would puke during the run when restarting hostapd failed

are you sure it restarted hostapd? I did that manually after the ansible run and everything was fine, without reboot.

jvonau, 2025-01-28:
> > Odd I would of thought ansible would puke during the run when restarting hostapd failed
> 
> are you sure it restarted hostapd? I did that manually after the ansible run and everything was fine, without reboot.

That is why I need to see the run, I recall some no fail code being in place somewhere

jvonau, 2025-01-28:
https://github.com/iiab/iiab/blob/master/vars/default_vars.yml#L23
>skip_role_on_error: True

tim-moody, 2025-01-28:
I think this was the run:

```
PLAY [all] *********************************************************************

TASK [Gathering Facts] *********************************************************
[WARNING]: Platform linux on host 127.0.0.1 is using the discovered Python
interpreter at /usr/bin/python3.11, but future installation of another Python
interpreter could change the meaning of that path. See
https://docs.ansible.com/ansible-
core/2.18/reference_appendices/interpreter_discovery.html for more information.
ok: [127.0.0.1]

TASK [0-init : ...IS BEGINNING ============================================] ***
ok: [127.0.0.1]

TASK [0-init : Ansible just ran /etc/ansible/facts.d/local_facts.fact to set 15 vars -- here we extract 6 of those -- iiab_stage: 9, rpi_model: Raspberry Pi 5 Model B Rev 1.0, devicetree_model: Raspberry Pi 5 Model B Rev 1.0, os_ver: raspbian-12, python_version: 3.11, php_version: 8.2] ***
ok: [127.0.0.1]

TASK [0-init : Create /etc/iiab/iiab.ini, if it doesn't exist] *****************
skipping: [127.0.0.1]

TASK [0-init : Set first_run flag] *********************************************
skipping: [127.0.0.1]

TASK [0-init : Copy iiab-update & iiab-summary & iiab-diagnostics & iiab-root-login from /opt/iiab/iiab/scripts/ to /usr/bin/] ***
ok: [127.0.0.1] => (item=iiab-update)
ok: [127.0.0.1] => (item=iiab-summary)
ok: [127.0.0.1] => (item=iiab-diagnostics)
ok: [127.0.0.1] => (item=iiab-root-login)

TASK [0-init : Symlink /usr/bin/iiab-upgrade -> /usr/bin/iiab-update] **********
ok: [127.0.0.1]

TASK [0-init : Create globally-writable directory /etc/iiab/diag (0777) so non-root users can run 'iiab-diagnostics'] ***
ok: [127.0.0.1]

TASK [0-init : Pre-check that IIAB's "XYZ_install" + "XYZ_enabled" vars (1) are defined, (2) are boolean-not-string variables, and (3) contain plausible values. Also checks that "XYZ_install" is True when "XYZ_installed" is defined.] ***
included: /opt/iiab/iiab/roles/0-init/tasks/validate_vars.yml for 127.0.0.1

TASK [0-init : Set vars_checklist for 45 + 45 + 40 vars ("XYZ_install" + "XYZ_enabled" + "XYZ_installed") to be checked] ***
ok: [127.0.0.1]

TASK [0-init : Assert that 45 "XYZ_install" vars are all... defined] ***********
ok: [127.0.0.1] => (item=hostapd)
ok: [127.0.0.1] => (item=dnsmasq)
ok: [127.0.0.1] => (item=bluetooth)
ok: [127.0.0.1] => (item=sshd)
ok: [127.0.0.1] => (item=tailscale)
ok: [127.0.0.1] => (item=remoteit)
ok: [127.0.0.1] => (item=admin_console)
ok: [127.0.0.1] => (item=squid)
ok: [127.0.0.1] => (item=cups)
ok: [127.0.0.1] => (item=samba)
ok: [127.0.0.1] => (item=usb_lib)
ok: [127.0.0.1] => (item=azuracast)
ok: [127.0.0.1] => (item=gitea)
ok: [127.0.0.1] => (item=jupyterhub)
ok: [127.0.0.1] => (item=lokole)
ok: [127.0.0.1] => (item=mysql)
ok: [127.0.0.1] => (item=mediawiki)
ok: [127.0.0.1] => (item=mosquitto)
ok: [127.0.0.1] => (item=nodejs)
ok: [127.0.0.1] => (item=nodered)
ok: [127.0.0.1] => (item=nextcloud)
ok: [127.0.0.1] => (item=wordpress)
ok: [127.0.0.1] => (item=kalite)
ok: [127.0.0.1] => (item=kolibri)
ok: [127.0.0.1] => (item=kiwix)
ok: [127.0.0.1] => (item=postgresql)
[... 839 lines cut ...]

TASK [network : Run /usr/bin/iiab-gen-iptables] ********************************
changed: [127.0.0.1]

TASK [network : Checking if WiFi slave is active] ******************************
changed: [127.0.0.1]

TASK [network : Restart hostapd if WiFi slave is inactive] *********************
skipping: [127.0.0.1]

TASK [network : Restarting dnsmasq] ********************************************
changed: [127.0.0.1]

TASK [Select RPi firmware mode] ************************************************
included: firmware for 127.0.0.1

TASK [firmware : Install firmware (for RPi internal WiFi)] *********************
skipping: [127.0.0.1]

TASK [network : Create /etc/iiab/install-flags/iiab-network-complete on second pass of network role.] ***
changed: [127.0.0.1]

PLAY RECAP *********************************************************************
127.0.0.1 : ok=112 changed=33 unreachable=0 failed=0 skipped=74 rescued=0 ignored=1
```

jvonau, 2025-01-28:
>root@box:/opt/iiab/iiab# systemctl status hostapd
× hostapd.service - Hostapd IEEE 802.11 AP, IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator
     Loaded: loaded (/etc/systemd/system/hostapd.service; enabled; preset: enabled)
     Active: failed (Result: exit-code) since Tue 2025-01-28 16:44:34 GMT; 1min 4s ago
Jan 28 16:44:34 box hostapd[5693]: nl80211: Could not configure driver mode
Jan 28 16:44:34 box hostapd[5693]: nl80211: deinit ifname=wlan0 disabled_11b_rates=0
Jan 28 16:44:34 box hostapd[5693]: nl80211 driver initialization failed.
Jan 28 16:44:34 box hostapd[5693]: wlan0: interface state UNINITIALIZED->DISABLED
Jan 28 16:44:34 box hostapd[5693]: wlan0: AP-DISABLED
Jan 28 16:44:34 box hostapd[5693]: wlan0: CTRL-EVENT-TERMINATING
Jan 28 16:44:34 box hostapd[5693]: hostapd_free_hapd_data: Interface wlan0 wasn't started
Jan 28 16:44:34 box systemd[1]: hostapd.service: Main process exited, code=exited, status=1/FAILURE
Jan 28 16:44:34 box systemd[1]: hostapd.service: Failed with result 'exit-code'.
Jan 28 16:44:34 box systemd[1]: Stopped hostapd.service - Hostapd IEEE 802.11 AP, IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator.

The timestamps are close, that run is set with wifi_up_down False and wlan0 is shown above.
>changed: [127.0.0.1] => (item={'option': 'runtime_date', 'value': '2025-01-28T16:43:37Z'})
TASK [network : Copy manage.conf for NetworkManager when wifi_up_down False] ***
changed: [127.0.0.1]
TASK [network : Remove manage.conf for NetworkManager when wifi_up_down True] ***
skipping: [127.0.0.1]
TASK [network : Use custom 'hostapd' systemd service unit file for wlan0 when not wifi_up_down] ***
changed: [127.0.0.1]
TASK [network : Disable ap0 related services] **********************************
changed: [127.0.0.1] => (item=iiab-clone-wifi.service)
changed: [127.0.0.1] => (item=iiab-wifi-test.service)
TASK [network : Restart hostapd when WiFi is present but not when using WiFi as gateway] ***
changed: [127.0.0.1]

Should of failed here^^^
TASK [network : Waiting 10 seconds for network to stabilize (dnsmasq will fail if br0 isn't in a 'up' state!)] ***
changed: [127.0.0.1]
TASK [network : Checking if WiFi slave is active] ******************************
changed: [127.0.0.1]
TASK [network : Restart hostapd if WiFi slave is inactive] *********************
skipping: [127.0.0.1]

OR Failed here ^^^ Everything looks fine,

>TASK [network : Reload systemd] ************************************************
ok: [127.0.0.1]
TASK [network : Restart the NetworkManager service] ****************************
changed: [127.0.0.1]

That would be the pre-edit run. 
>root@box:~# systemctl status hostapd
× hostapd.service - Hostapd IEEE 802.11 AP, IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator
     Loaded: loaded (/etc/systemd/system/hostapd.service; enabled; preset: enabled)
     Active: failed (Result: exit-code) since Tue 2025-01-28 14:51:57 GMT; 5min ago
   Duration: 22h 5min 11.270s
    Process: 5759 ExecStart=/usr/sbin/hostapd -B -P /run/hostapd.pid /etc/hostapd/hostapd.conf (code=exited, status=1/FAILURE)
        CPU: 4ms
Jan 28 14:51:57 box hostapd[5759]: nl80211: deinit ifname=ap0 disabled_11b_rates=0
Jan 28 14:51:57 box hostapd[5759]: Could not read interface ap0 flags: No such device
Jan 28 14:51:57 box hostapd[5759]: nl80211 driver initialization failed.
Jan 28 14:51:57 box hostapd[5759]: ap0: interface state UNINITIALIZED->DISABLED
Jan 28 14:51:57 box hostapd[5759]: ap0: AP-DISABLED
Jan 28 14:51:57 box hostapd[5759]: ap0: CTRL-EVENT-TERMINATING
Jan 28 14:51:57 box hostapd[5759]: hostapd_free_hapd_data: Interface ap0 wasn't started
Jan 28 14:51:57 box systemd[1]: hostapd.service: Control process exited, code=exited, status=1/FAILURE
Jan 28 14:51:57 box systemd[1]: hostapd.service: Failed with result 'exit-code'.
Jan 28 14:51:57 box systemd[1]: Failed to start hostapd.service - Hostapd IEEE 802.11 AP, IEEE 802.1X/WPA/WPA2/EAP/RADIUS 

The run we are looking for is before 14:51 around 1450 would be the start time.

tim-moody, 2025-01-28:
I'll run again. We are looking for change from up down checked to unchecked. Expectation is that it will complete ansible but hostapd will not be running. systemctl restart hostapd will restart it.

tim-moody, 2025-01-28:
```
PLAY [all] *********************************************************************

TASK [Gathering Facts] *********************************************************
[WARNING]: Platform linux on host 127.0.0.1 is using the discovered Python
interpreter at /usr/bin/python3.11, but future installation of another Python
interpreter could change the meaning of that path. See
https://docs.ansible.com/ansible-
core/2.18/reference_appendices/interpreter_discovery.html for more information.
ok: [127.0.0.1]

TASK [0-init : ...IS BEGINNING ============================================] ***
ok: [127.0.0.1]

TASK [0-init : Ansible just ran /etc/ansible/facts.d/local_facts.fact to set 15 vars -- here we extract 6 of those -- iiab_stage: 9, rpi_model: Raspberry Pi 5 Model B Rev 1.0, devicetree_model: Raspberry Pi 5 Model B Rev 1.0, os_ver: raspbian-12, python_version: 3.11, php_version: 8.2] ***
ok: [127.0.0.1]

TASK [0-init : Create /etc/iiab/iiab.ini, if it doesn't exist] *****************
skipping: [127.0.0.1]

TASK [0-init : Set first_run flag] *********************************************
skipping: [127.0.0.1]

TASK [0-init : Copy iiab-update & iiab-summary & iiab-diagnostics & iiab-root-login from /opt/iiab/iiab/scripts/ to /usr/bin/] ***
ok: [127.0.0.1] => (item=iiab-update)
ok: [127.0.0.1] => (item=iiab-summary)
ok: [127.0.0.1] => (item=iiab-diagnostics)
ok: [127.0.0.1] => (item=iiab-root-login)

TASK [0-init : Symlink /usr/bin/iiab-upgrade -> /usr/bin/iiab-update] **********
ok: [127.0.0.1]

TASK [0-init : Create globally-writable directory /etc/iiab/diag (0777) so non-root users can run 'iiab-diagnostics'] ***
ok: [127.0.0.1]

TASK [0-init : Pre-check that IIAB's "XYZ_install" + "XYZ_enabled" vars (1) are defined, (2) are boolean-not-string variables, and (3) contain plausible values. Also checks that "XYZ_install" is True when "XYZ_installed" is defined.] ***
included: /opt/iiab/iiab/roles/0-init/tasks/validate_vars.yml for 127.0.0.1

TASK [0-init : Set vars_checklist for 45 + 45 + 40 vars ("XYZ_install" + "XYZ_enabled" + "XYZ_installed") to be checked] ***
ok: [127.0.0.1]

TASK [0-init : Assert that 45 "XYZ_install" vars are all... defined] ***********
ok: [127.0.0.1] => (item=hostapd)
ok: [127.0.0.1] => (item=dnsmasq)
ok: [127.0.0.1] => (item=bluetooth)
ok: [127.0.0.1] => (item=sshd)
ok: [127.0.0.1] => (item=tailscale)
ok: [127.0.0.1] => (item=remoteit)
ok: [127.0.0.1] => (item=admin_console)
ok: [127.0.0.1] => (item=squid)
ok: [127.0.0.1] => (item=cups)
ok: [127.0.0.1] => (item=samba)
ok: [127.0.0.1] => (item=usb_lib)
ok: [127.0.0.1] => (item=azuracast)
ok: [127.0.0.1] => (item=gitea)
ok: [127.0.0.1] => (item=jupyterhub)
ok: [127.0.0.1] => (item=lokole)
ok: [127.0.0.1] => (item=mysql)
ok: [127.0.0.1] => (item=mediawiki)
ok: [127.0.0.1] => (item=mosquitto)
ok: [127.0.0.1] => (item=nodejs)
ok: [127.0.0.1] => (item=nodered)
ok: [127.0.0.1] => (item=nextcloud)
ok: [127.0.0.1] => (item=wordpress)
ok: [127.0.0.1] => (item=kalite)
ok: [127.0.0.1] => (item=kolibri)
ok: [127.0.0.1] => (item=kiwix)
ok: [127.0.0.1] => (item=postgresql)
ok: [127.0.0.1] => (item=moodle)
ok: [127.0.0.1] => (item=mongodb)
[... 837 lines cut ...]

TASK [network : Run /usr/bin/iiab-gen-iptables] ********************************
changed: [127.0.0.1]

TASK [network : Checking if WiFi slave is active] ******************************
changed: [127.0.0.1]

TASK [network : Restart hostapd if WiFi slave is inactive] *********************
skipping: [127.0.0.1]

TASK [network : Restarting dnsmasq] ********************************************
changed: [127.0.0.1]

TASK [Select RPi firmware mode] ************************************************
included: firmware for 127.0.0.1

TASK [firmware : Install firmware (for RPi internal WiFi)] *********************
skipping: [127.0.0.1]

TASK [network : Create /etc/iiab/install-flags/iiab-network-complete on second pass of network role.] ***
changed: [127.0.0.1]

PLAY RECAP *********************************************************************
127.0.0.1 : ok=116 changed=37 unreachable=0 failed=0 skipped=70 rescued=0 ignored=1
```

tim-moody, 2025-01-28:
```
root@box:~# systemctl status hostapd
× hostapd.service - Hostapd IEEE 802.11 AP, IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator
     Loaded: loaded (/etc/systemd/system/hostapd.service; enabled; preset: enabled)
     Active: failed (Result: exit-code) since Tue 2025-01-28 21:32:02 GMT; 18s ago
   Duration: 6.323s
    Process: 5754 ExecStart=/usr/sbin/hostapd -P /run/hostapd.pid /etc/hostapd/hostapd.conf (code=exited, status=1/FAILURE)
   Main PID: 5754 (code=exited, status=1/FAILURE)
        CPU: 6ms

Jan 28 21:32:02 box hostapd[5754]: nl80211: Could not configure driver mode
Jan 28 21:32:02 box hostapd[5754]: nl80211: deinit ifname=wlan0 disabled_11b_rates=0
Jan 28 21:32:02 box hostapd[5754]: nl80211 driver initialization failed.
Jan 28 21:32:02 box hostapd[5754]: wlan0: interface state UNINITIALIZED->DISABLED
Jan 28 21:32:02 box hostapd[5754]: wlan0: AP-DISABLED
Jan 28 21:32:02 box hostapd[5754]: wlan0: CTRL-EVENT-TERMINATING
Jan 28 21:32:02 box hostapd[5754]: hostapd_free_hapd_data: Interface wlan0 wasn't started
Jan 28 21:32:02 box systemd[1]: hostapd.service: Main process exited, code=exited, status=1/FAILURE
Jan 28 21:32:02 box systemd[1]: hostapd.service: Failed with result 'exit-code'.
Jan 28 21:32:02 box systemd[1]: Stopped hostapd.service - Hostapd IEEE 802.11 AP, IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator.

root@box:~# systemctl restart hostapd

root@box:~# systemctl status hostapd
● hostapd.service - Hostapd IEEE 802.11 AP, IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator
     Loaded: loaded (/etc/systemd/system/hostapd.service; enabled; preset: enabled)
     Active: active (running) since Tue 2025-01-28 21:34:38 GMT; 2s ago
   Main PID: 6458 (hostapd)
      Tasks: 1 (limit: 9572)
        CPU: 6ms
     CGroup: /system.slice/hostapd.service
             └─6458 /usr/sbin/hostapd -P /run/hostapd.pid /etc/hostapd/hostapd.conf

Jan 28 21:34:38 box systemd[1]: Started hostapd.service - Hostapd IEEE 802.11 AP, IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator.
Jan 28 21:34:38 box hostapd[6458]: wlan0: interface state UNINITIALIZED->COUNTRY_UPDATE
Jan 28 21:34:38 box hostapd[6458]: wlan0: interface state COUNTRY_UPDATE->ENABLED
Jan 28 21:34:38 box hostapd[6458]: wlan0: AP-ENABLED

```

jvonau, 2025-01-28:
>TASK [network : Reload systemd] ************************************************
ok: [127.0.0.1]
TASK [network : Restart the NetworkManager service] ****************************
changed: [127.0.0.1]

I'm not sure if the hand edit is in place, can you do a `git diff` in /opt/iiab/iiab and have look? Should have the + line for that stanza
>TASK [network : Restart hostapd when WiFi is present but not when using WiFi as gateway] ***
changed: [127.0.0.1]

Wondering if iiab-network is subject to skip_role_on_error or not... Nothing like change that inverts what used to be normal. Kind off backed away from engaging with the project given how I'm treated.

If the change is not present we can try to validate 3925 with:
`git checkout master`
`git checkout -b 1520` just to make sure we don't have any unsaved changes, if there are then git will warn you
`git pull https://github.com/jvonau/iiab/ 1520`
`git log`  and check the last commit

Think at this point running `iiab-network` would be easier to post the logging from iiab-network.log but might be skewed from skip_role_on_error but other useful info is collected at the end.

jvonau, 2025-01-28:
>TASK [network : Static IP computing CIDR] **************************************
changed: [127.0.0.1]
TASK [network : Static IP setting CIDR] ****************************************
ok: [127.0.0.1]
TASK [network : Create uuid for NM's keyfile store static] *********************
changed: [127.0.0.1]
TASK [network : Put the uuid in place] *****************************************
ok: [127.0.0.1]
TASK [network : Copy static template for NetworkManager] ***********************
changed: [127.0.0.1]

Ok what else is going on in the background? 
This is not just testing wifi_up_down.

jvonau, 2025-01-29:
@holta does 'skip_role_on_error: True' alter the call to systemdctl adding --no-block to the into the mix?

jvonau, 2025-01-29:
Please add the same logic to iiab-network in #3925

jvonau, 2025-01-29:
> > does 'skip_role_on_error: True' alter the call to systemctl adding --no-block to the into the mix?
> 
> I don't think so? Interesting that this exists. I don't think I knew about `systemctl --no-block` — and Ansible module `systemd_service:` doing the equivalent using `no_block: True`.

https://docs.ansible.com/ansible/latest/collections/ansible/builtin/systemd_service_module.html#parameter-no_block

The default is 'false' but the docs might be stale or there maybe a regression. Update ansible recently?

holta, 2025-01-29:
> Please add the same logic to iiab-network in [#3925](https://github.com/iiab/iiab/pull/3925)

Done (https://github.com/iiab/iiab/pull/3925/commits/519daacc2af19bf3848c9f7926fec241400bbe85) on:

- PR #3925

jvonau, 2025-01-29:
I introduced an intentional bug to cause hostapd to fail
>diff --git a/roles/network/templates/hostapd/hostapd.conf.j2 b/roles/network/templates/hostapd/hostapd.conf.j2
index 171f18505..11f6b1d75 100644
--- a/roles/network/templates/hostapd/hostapd.conf.j2
+++ b/roles/network/templates/hostapd/hostapd.conf.j2
@@ -4,7 +4,7 @@ interface={{ iiab_wireless_lan_iface }}
 ctrl_interface=/var/run/hostapd
 ctrl_interface_group=0
 -ssid={{ host_ssid }}
+#ssid={{ host_ssid }}
 channel={{ host_channel }}
 {%if iiab_lan_iface == "br0" %}
 bridge=br0

Ran iiab-network without altering for --extra-vars
>TASK [network : Restart hostapd when WiFi is present but not when using WiFi as gateway] *************************
fatal: [127.0.0.1]: FAILED! => {"changed": false, "msg": "Unable to restart service hostapd: Job for hostapd.service failed because the control process exited with error code.\nSee \"systemctl status hostapd.service\" and \"journalctl -xeu hostapd.service\" for details.\n"}
PLAY RECAP *******************************************************************************************************
127.0.0.1                  : ok=107  changed=22   unreachable=0    failed=1    skipped=67   rescued=0    ignored=1

ansible --version
>ansible [core 2.18.1]
  config file = /opt/iiab/iiab/ansible.cfg
  configured module search path = ['/home/jvonau/.ansible/plugins/modules', '/usr/share/ansible/plugins/modules']
  ansible python module location = /usr/local/ansible/lib/python3.11/site-packages/ansible
  ansible collection location = /home/jvonau/.ansible/collections:/usr/share/ansible/collections
  executable location = /usr/local/bin/ansible
  python version = 3.11.2 (main, Nov 30 2024, 21:22:50) [GCC 12.2.0] (/usr/local/ansible/bin/python3)
  jinja version = 3.1.5
  libyaml = True

Install is a few days old
>install_date = 2025-01-23T17:43:47Z

jvonau, 2025-01-29:
#3922

jvonau, 2025-01-29:
To test the above intentional bug across Wifi I altered
` index 5902f5d50..bdd8bd59f 100644
--- a/roles/network/tasks/restart.yml
+++ b/roles/network/tasks/restart.yml
@@ -33,7 +33,7 @@
     name: hostapd
     state: restarted
     daemon_reload: yes
-  when: hostapd_enabled and not no_net_restart
+#when: hostapd_enabled and not no_net_restart`

`@@ -107,7 +107,7 @@
   systemd:
     name: hostapd
     state: restarted
-  when: hostapd_enabled and not no_net_restart and wifi_slave.stdout is defined and wifi_slave.stdout == 0
+ #when: hostapd_enabled and not no_net_restart and wifi_slave.stdout is defined and wifi_slave.stdout == 0`

To defeat the "not no_net_restart" for the upstream wifi

jvonau, 2025-01-29:
pulled 3925 in to a clean branch then altered for intentional bug and to defeat no_net_restart
ran iiab-network

>TASK [network : Restart hostapd when WiFi is present but not when using WiFi as gateway] *************************
fatal: [127.0.0.1]: FAILED! => {"changed": false, "msg": "Unable to start service hostapd: Job for hostapd.service failed because the control process exited with error code.\nSee \"systemctl status hostapd.service\" and \"journalctl -xeu hostapd.service\" for details.\n"}
PLAY RECAP *******************************************************************************************************
127.0.0.1                  : ok=107  changed=20   unreachable=0    failed=1    skipped=67   rescued=0    ignored=1

Think 3925 is good enough for now, --extra-vars didn't really change any behavior but didn't hurt anything either.

tim-moody, 2025-01-30:
>I'm not sure if the hand edit is in place
```
root@box:/opt/iiab/iiab# git diff
diff --git a/roles/network/tasks/NM-debian.yml b/roles/network/tasks/NM-debian.yml
index 9cb2cf1..a0af7ad 100644
--- a/roles/network/tasks/NM-debian.yml
+++ b/roles/network/tasks/NM-debian.yml
@@ -78,7 +78,7 @@
 - name: Reload systemd
   systemd:
     daemon_reload: yes
-  when: not iiab_lan_iface == "br0"
+  when: not no_net_restart or not iiab_lan_iface == "br0"

 - name: Restart the NetworkManager service
   systemd:

```
Not sure if this is still of interest.

tim-moody, 2025-01-30:
>This is not just testing wifi_up_down.

was the only parameter changed

jvonau, 2025-01-30:
> > This is not just testing wifi_up_down.
> 
> was the only parameter changed

Code:
https://github.com/iiab/iiab/blob/master/roles/network/tasks/NM-debian.yml#L40-L71

Default vars:
># See "How do I set a static IP address?" for Ethernet, in http://FAQ.IIAB.IO
wan_ip: dhcp       # wan_ip: 192.168.1.99
wan_netmask:       # wan_netmask: 255.255.255.0
wan_gateway:       # wan_gateway: 192.168.1.254

Can you post the above entries from your local_vars.yml file please. The memory got jogged, admin-console populates populates the latter two but keeps the former untouched.

tim-moody, 2025-01-30:
OK. You are right and I wondered about possible impact when I saw Gateway checked:

gui_desired_network_role: Gateway
wifi_up_down: False
gui_static_wan: True
gui_static_wan_ip: 192.168.3.60
gui_static_wan_netmask: 255.255.255.0
gui_static_wan_gateway: 192.168.3.1
gui_static_wan_nameserver: 192.168.3.1

But I didn't notice `Check to use a static WAN IP Address instead of DHCP` as being checked, which it is not at present.

I would have to repeat the test to see.

tim-moody, 2025-01-30:
This looks bogus too

[image]

jvonau, 2025-01-30:
> This looks bogus too
> 

I'm going to assume admin-console didn't find a gateway when booted. Maybe cmdsrv started before the network was available. Was the network cable plugged in at boot or plugged in post boot?

tim-moody, 2025-01-30:
and after reboot I can't even ssh in. feels like this install has had too many changes.

removed power and restarted. now adm cons agrees with local vars

>I'm going to assume admin-console didn't find a gateway when booted. Maybe cmdsrv started before the network was available. Was the network cable plugged in at boot or plugged in post boot?

seems likely, but cmdsrv doesn't set up the network. cable was plugged in at boot 

probably when I ran ICO before static lan was somehow checked and I missed it

[70 more replies]