iiab/iiab #1521: IIAB installer should auto-detect NOOBS (pre-installed on many RPi's) alerting you how to proceed

holta, 2019-02-25:
@mitra42 described the hours he lost and 2 recommendations:

> I'm a relative RPi newbie - the previous install was easy for me since NOOBS got me the OS and I only had to work on top of that, which I'm familiar with. For IIAB it was *far* from obvious to me. 
>
> My first problem is which to install, which might also be a problem of where to start.... 
>
> I started with http://d.iiab.io/install.txt ...
>
> The install.txt tells me its not compatible with NOOBS, fine - except that my RPi came out of the box with NOOBS, and your install script doesn't tell me how to get from having NOOBS to a suitable starting point to then run the IIAB installer. For example some pointer on how to get your preferred compatible distro would be useful.  Suggestion: Pointer or instruction for people whose RPi came with NOOBS (as both brands I bought do...)
>
> I eventually followed the link to the Unix manual at https://github.com/iiab/iiab/wiki/IIAB-Installation - this wasn't my obvious choice since I'm not trying to do anything on a different OS, I'm trying to use your prefered (Raspbian) and that led me to  downloading Stretch from https://www.raspberrypi.org/downloads/raspbian/ and somewhere I found https://www.balena.io/etcher/ which I could use to get it onto an SD and boot my RPi.  
>
> This was entirely non-obvious, I know Unix well, and have done a raw (NOOBS) install before, I doubt anyone just ordering a RPI online, and trying to get a IIAB box going will find this easy, and think most would fail.  Note in particular that some of the downloads are very large so that making mistakes can cause multi-hour waits to figure out what you did wrong.

He also recommends that all implementers learn about (the need to re-run) `sudo iiab` before they begin their install.  There are very loud warnings today, within the 1-line installer itself, but this makes sense &mdash; I need to make these warnings even louder &mdash; perhaps even forcing the implementer to acknowledge that they understand...before proceeding:

> [IIAB] install from the script worked ok - it failed twice and it probably wouldnt be obvious to someone less experienced. Suggestion stronger suggestion that people run `sudo iiab` repeatedly until they get a run with no error messages. (except that I got an error message even on my last run - something about being a member of a sharing network, so maybe that script needs tweaking to know its not an error)

holta, 2019-02-25:
@tim-moody suggests NOOBS might allow folks to easily "upgrade" to full Raspbian possibly?

If so we can try to document this &mdash; ideally as part of auto-alerting folks (in 1-line installer) that they're trying to install IIAB on an inappropriate OS.

holta, 2019-02-25:
@jvonau suggests we might put a LOUD BANNER in the implementers face every time they ssh in, similar to Raspbian & IIAB's security warnings about weak passwords.

If so emulating these 2 files:
https://github.com/iiab/iiab/blob/master/roles/iiab-admin/templates/lxde_ssh_warn.sh
https://github.com/iiab/iiab/blob/master/roles/iiab-admin/templates/profile_ssh_warn.sh

As installed by:
https://github.com/iiab/iiab/blob/master/roles/iiab-admin/tasks/main.yml#L24-L44

mitra42, 2019-02-25:
@holta - @tim-moody - Unless automatically going from NOOBS to Raspbian is possible, then I think the NOOBS thing is needed in two places.
a) in the script - when it runs - detect NOOBS and exit
b) in the text of install.txt this needs rewording. 
```
 To install Internet-in-a-Box (IIAB) 7.0 / pre-release onto Raspbian Stretch,
# Ubuntu 18.04 or Debian 9, run this 1-line installer: 
# curl d.iiab.io/install.txt | sudo bash
WARNING: NOOBS IS *NOT* SUPPORTED, as its partitioning is very different.
#    To attempt IIAB 7.0 on another Linux see the full/manual instructions:
#    https://github.com/iiab/iiab/wiki/IIAB-Installation#do-everything-from-scratch
```
Since I'm not trying to run `on another Linux`, I'm trying to get from NOOBS to Raspbian Stretch, which is ALMOST, but not quite, covered by the wiki.

mitra42, 2019-02-25:
Then I think the Wiki should tell me how to install Rasbian on a RPi. 

Note that you are three steps away from doing this ... 
a) find install.txt or home page
b) figure out that NOOBS isn't supported and then go to the do-everything-from-scratch wiki
c) figure out you need Rasbian and follow that link
d) Spot that you need the installation guide and follow that link.

There is enough geeky language on all those pages to reduce the chances of successfully getting there. I'd suggest a link in install.txt and home page direct to the installation guide that says something like: 

```
NOOBS is not supported, go to the [installation guide](https://www.raspberrypi.org/documentation/installation/installing-images/README.md) to install the latest version of Rasbian. 
```

holta, 2019-02-25:
@mitra42 I'm working with community volunteers in India on simplifying our most commonly-accessed docs, for non-technical librarians especially.  There will be major revisions in coming months.

Meantime thanks for the link (don't tell ppl what mistakes to avoid, rather show them the way ;-) which I've added here:

- http://d.iiab.io/6.7
- http://d.iiab.io/6.7/install.txt
- http://d.iiab.io/6.7/iiab

Also for IIAB 7.0/master:

- http://d.iiab.io (README auto-displayed)
- http://d.iiab.io/install.txt

holta, 2019-02-26:
Documentation blow further enhanced &mdash; alongside http://FAQ.IIAB.IO Number 8 ("[Is a quick installation possible?](http://wiki.laptop.org/go/IIAB/FAQ#Is_a_quick_installation_possible.3F)")

> * http://d.iiab.io/6.7 (README auto-displayed)
> * http://d.iiab.io/6.7/install.txt
> * http://d.iiab.io/6.7/iiab
> 
> Also for IIAB 7.0/master:
> 
> * http://d.iiab.io (README auto-displayed)
> * http://d.iiab.io/install.txt

mitra42, 2019-07-02:
@holta did you figure this out for 7.0 ? I'm just wondering if I need to modify directions for people who startoff on NOOBS. Do they still need to do the manual Rasbian image download etc ?

holta, 2019-07-02:
NOOBS is not supported for IIAB 7.0

(It's very different in it's partitioning from mainline "real" Raspbian, according to @floydianslips and others, so this is too difficult to support for now )

mitra42, 2019-07-02:
Yes - understood, the issue was more about how IIAB handles NOOBS from the perspective of guiding users to a successful install, since most RPI's I've seen come with NOOBS rather than Raspbian. 

I unfortunately don't have a high degree of confidence in most, even technical, users - who buy an off-the-shelf RPI - following the instructions at https://github.com/internetarchive/dweb-mirror/blob/master/INSTALLATION-raspberrypi.md all the way through to successful install of IIAB+IA