iiab/iiab #1527: RPi installs w/ Ethernet AND WiFi very flaky: disconnects ssh install session due to auto-creation of hotspot?

holta, 2019-02-26:
_CONTEXT: No matter what we tell people, it's an extremely common situation for a PC/RPi to be connected to both Ethernet and WiFi when its IIAB install begins.  Then things go a bit haywire, and we can do better!_

So @jvonau is helping try to resolve this on 10.8.0.58 = 192.168.0.142/lan = 192.168.0.100/wifi that has (should, until we run `iiab-hotspot-on`) have both Ethernet and WiFi gateways:
```
root@box:~# more /etc/iiab/openvpn_handle
142-rpi3bp-desktop-feb25-WIFI
```
Patterns...all of which are repeatable on 7.0/master:
- Every time you run `./iiab-network`, ssh output freezes at `TASK [network : Restart hostapd when WiFi is present]` even while it does in fact complete behind the scenes, according to /opt/iiab/iiab/iiab-network.log _(implementer should not be left high+dry like this)_
- ssh to BOTH 192.* local addresses (Ethernet & Wi-Fi) then fail _(let's fix this!)_
- ssh to OpenVPN address works about 30-50% of the time, freezing screen output every minute or less, and dropping the ssh connection every minute or so _(let's fix this!)_
- hotspot is forcibly re-created by `./iiab-network` even when the implementer has asked to turn it off using `iiab-hotspot-off` ...is this what we really want?? _(implementer's expressed desire should not be overruled)_
- ssh to 192.168.0.142/ethernet can be restored using a reboot (but this should not be nec!)
- ssh to 192.168.0.100/wifi can be restored using `iiab-hotspot-off` (but this should not be nec, right?)

Background...this is how you confirm the hotspot is on:
```
root@box:~# brctl show
bridge name     bridge id               STP enabled     interfaces
br0             8000.b827eb555ed2       no              wlan0
```
Background...this is how you confirm hotspot if off:
```
root@box:~# brctl show
bridge name     bridge id               STP enabled     interfaces
br0             8000.000000000000       no
```
_Might take a couple days to (re)solve these...such that output/instructions are crystal clear for new implementers as is a central goal here, however we work this out (all implemeters ask: "just tell me what is happening, so I know my options!")_

Refs: #1519 PRs #1453 #1523 #1525 #1526

holta, 2019-02-27:
> ssh to 192.168.0.142/ethernet can be restored using a reboot _(but this should not be nec)_

FYI this is exact same problem occurs when running `iiab-hotspot-on` on an RPi / Rasp Lite that did NOT have an Ethernet cord during its original IIAB installation (e.g. 10.8.0.38 = 192.168.0.143) &mdash; the ability to ssh in over Ethernet is blocked, until the next boot.

Whereas `iiab-hotspot-off` does _not_ clobber the ability to ssh over Ethernet on 10.8.0.38 here.

holta, 2019-02-28:
> > ssh to 192.168.0.142/ethernet can be restored using a reboot _(but this should not be nec)_
> 
> FYI this is exact same problem occurs when running `iiab-hotspot-on` on an RPi / Rasp Lite that did NOT have an Ethernet cord during its original IIAB installation (e.g. 10.8.0.38 = 192.168.0.143) — the ability to ssh in over Ethernet is blocked, until the next boot.

Turns out this is not always true on 143-rpi3bp-lite-feb26-MIN-WIFI.  Specifically, the Ethernet stays alive after `iiab-hotspot-on` *if* the machine was booted with the Ethernet inserted.  Thanks to @jvonau who suspected this, which I've now confirmed, just FYI.

(Anyway this ticket is principally about 142-rpi3bp-desk-feb25-WIFI+WIFI which continues to fail in the several ways described at the top of this ticket.)

jvonau, 2020-04-29:
resolved with either the latest master or #2337

closed 2020-04-29 by holta